raspiblitz/raspiblitz #2901: Refactor Seed+SCB recovery/rescan

rootzoll, 2022-01-18:
Recovering from seed (& SCB) needs a refactor. The idea is here to put the lnd & cl into a recovery mode where they are the started everytime with resetting on-chain transaction history and going into rescan when booted/restarted until they run trhu one time in a row and then get set back to normal mode.

So the scripts `lnd.backuo.sh` & `cl.backup.sh` with get an recovery option for mainnet/testnet/signet and the background process will check for the recovery to finish.

Notes for lnd:
```
lncli getrecoveryinfo 
lnd --reset-wallet-transactions
```

rootzoll, 2022-01-19:
Its now activated thru lnd config - with https://github.com/lightningnetwork/lnd/blob/431ef6c4c3dfb46ec886fa3450b19346a964a108/sample-lnd.conf#L273

rootzoll, 2022-01-19:
Ok .. so it works now this way:

First the wallet needs to be created from seed. Then `sudo ./config.scripts/lnd.backup.sh mainnet recoverymode on` switches the recoverymode on (editing the lnd conf). A restart of lnd or reboot of system is now needed. `./config.scripts/lnd.unlock.sh` is detecting the recoverymode and is unlocking with `--recovery_window=5000`. In between reboots are now not a problem - it would just restart the rescan. 

To detect the state of the recovery there are now two more vars that get outputted by `lnd.monitor.sh mainnet info` using the lncli getrecoveryinfo to conclude:
```
ln_lnd_recovery_mode='1'
ln_lnd_recovery_done='0'
``` 

the `_background.scan.sh` script is regularly running this monitor script (when wallet is unlocked) and then in chache these values are available:
```
ln_lnd_mainnet_recovery_mode="1"
ln_lnd_mainnet_recovery_done="0"
ln_default_recovery_mode="1"
ln_default_recovery_done="0"
```

Finally the `_background.sh` is monitoring the cache if recovery_mode was started & is sinaling recovery_done while also the blockchain is signaling no more initialblockdownload ... if all that is true it calls `sudo ./config.scripts/lnd.backup.sh mainnet recoverymode off` so that on the next reboot/restart of lnd the wallet keeps now its transactions.

@openoms so now we need all this also for c-lightning :)

openoms, 2022-01-19:
Tested the recovery from seed on testnet and mainnet from the menu. Both works with scanning starting correctly only the recovery mode does not make it in the cache once finished:
```
sudo config.scripts/lnd.monitor.sh mainnet info
...
ln_lnd_tor='1'
ln_lnd_sync_chain='1'
ln_lnd_sync_progress='100.00'
ln_lnd_sync_graph='1'
ln_lnd_channels_pending='0'
ln_lnd_channels_active='0'
ln_lnd_channels_inactive='0'
ln_lnd_channels_total='0'
ln_lnd_peers='4'
ln_lnd_recovery_mode='1'
ln_lnd_recovery_done='1'
```
```
/home/admin/_cache.sh get ln_lnd_mainnet_recovery_mode ln_lnd_mainnet_recovery_done btc_mainnet_sync_initialblockdownload
# _cache.sh get ln_lnd_mainnet_recovery_mode ln_lnd_mainnet_recovery_done btc_mainnet_sync_initialblockdownload
ln_lnd_mainnet_recovery_mode="1"
ln_lnd_mainnet_recovery_done="0"
btc_mainnet_sync_initialblockdownload="0"
```

The background.scan seems to run fine:
```
₿ sudo journalctl -f -u background.scan
_background.scan.sh[2448]: system_scan_all_temp found --> TRIGGER system_scan_all
_background.scan.sh[2448]: updating: /home/admin/config.scripts/lnd.monitor.sh mainnet wallet
```

rootzoll, 2022-01-20:
The problem is that the macaroon owner is bitcoin:bitcoin and root (doing the background scan) is not in group bitcoin. Will add root user to group bitcoin in beginning of background.scan script.

rootzoll, 2022-01-20:
Also it then needs a wallet unlock before data is available

rootzoll, 2022-01-20:
@openoms I get similar problem when trying to get info from `cl.monitor.sh`:
```
sudo /usr/local/bin/lightning-cli --lightning-dir=/home/bitcoin/.lightning --conf=/home/bitcoin/.lightning/config getinfo
lightning-cli: Connecting to 'lightning-rpc': Connection refused

/usr/local/bin/lightning-cli --lightning-dir=/home/bitcoin/.lightning --conf=/home/bitcoin/.lightning/config getinfo
lightning-cli: Moving into '/home/bitcoin/.lightning/bitcoin': Permission denied
```

Any idea?

openoms, 2022-01-20:
For CL:

`sudo ./config.scripts/cl.backup.sh mainnet recoverymode on` switches recovery on with editing the cl `config`.
This will start the rescan on every restart.

`cl.monitor.sh mainnet info` displays `ln_cl_recovery_mode` depending if `rescan=` is in the `config` and `ln_cl_recovery_done` if there if in `cl getinfo -H` there is no `warning_lightningd_sync=Still loading latest blocks from bitcoind.` .

`_background.scan.sh` running `cl.monitor.sh mainnet info`:
```
ln_cl_mainnet_recovery_mode="1"
ln_cl_mainnet_recovery_done="0"
ln_default_recovery_mode="1"
ln_default_recovery_done="0"
```

The `_background.sh` is monitoring the cache if `recovery_mode` was started & is signalling recovery_done while also the blockchain is signalling no more initialblockdownload ... if all that is true it calls `sudo ./config.scripts/cl.backup.sh mainnet recoverymode off` so that on the next reboot/restart so CL does not start a long rescan again on restart (the default is to scan the last 15 blocks).

openoms, 2022-01-20:
> sudo /usr/local/bin/lightning-cli --lightning-dir=/home/bitcoin/.lightning --conf=/home/bitcoin/.lightning/config getinfo

with `sudo` it should although it should be run with the `bitcoin` user.
```
lightning-cli: Connecting to 'lightning-rpc': Connection refused
```
just means that lightningd is not running or not ready yet.
Check: 
`cllog`
or 
`sudo journalctl -fu lightningd`

openoms, 2022-01-20:
TODO for CL:
- [ ] add `rescan` to the REPAIR-CL menu with choice of blockheight / depth
- [ ] add a manual `guesstoremote` option to get the funds from channels force closed by the peer

rootzoll, 2022-01-23:
When running the LND-REPAIR and REST with SEED+SCB I got the following error message after uploading the SCB:
```
WALLET --> UNLOCK WALLET - SCAN 0
WALLET --> SEED + SCB
StatusCode.UNKNOWN
/dev/fd/63: line 4: /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: Permission denied
lnd-wallet-seed+scb lnd.initwallet.py scb returned error /home/admin/config.scripts/lnd.initwallet.py scb mainnet ... --> RPCError RestoreChanBackupRequest + the RPC server is in the process of starting up, but not yet ready to accept calls
```

and as for first test ... the "rescan" afterthis does not finish - will keep runnnig over night to see results.

rootzoll, 2022-01-24:
OK simplified the Repair SEED/SEED+SCB/SCB and added restarts ... now tests works fine.
@openoms from my side this issue is now ready to merge .. are you still working on details?

openoms, 2022-01-24:
> OK simplified the Repair SEED/SEED+SCB/SCB and added restarts ... now tests works fine. @openoms from my side this issue is now ready to merge .. are you still working on details?

Glad you went to load the SCB through the cli, it is so much more reliable!
I am ready with the things here, ok to merge.

openoms, 2022-01-24:
Added the rescan function to recovering from seed and switching recovery mode on for CL in the REAIR-CL menu.
Will need testing, will try tomorrow.

rootzoll, 2022-01-25:
OK - I will merge the branch now into dev for final testing.

closed 2022-02-20 by rootzoll