raspiblitz/raspiblitz #2902: Stuck sitting at 100.00% with 12 peers, does not proceed to menu

nkinnan, 2022-01-19:
I copied the blockchain over from PC to save on sync time, went from 99.something percent to 100% but now it just sits there and does not give me the menu.  

SSD access light is blinking intermittently, no errors in logs that I can tell.

Accessing the http endpoint shows "unknown hostname: raspiblitz" (I named it simply raspiblitz which is also the .local name) and no other links or info.  

Tried a "sudo reboot now" but it just goes back to the same "100.00%" screen. 

"top" shows bitcoind using 1-2% CPU, about 6 minutes total over perhaps 3-4 hours.  "iotop" shows some minimal disk activity, mostly reads, from bitcoind.  As far as I can tell it's not really *doing* anything.

Debug output:
```
    ***************************************************************
    * RASPIBLITZ DEBUG LOGS
    ***************************************************************
    blitzversion: 1.7.1
    chainnetwork: bitcoin / main
     12:16:12 up 25 min,  3 users,  load average: 0.29, 0.45, 0.43

    *** SETUPPHASE / BOOTSTRAP***
    see logs: cat /home/admin/raspiblitz.log
    setupPhase--> done
    state--> ready

    *** BACKGROUNDSERVICE ***
    to monitor Background service call: sudo journalctl -f -u background

    *** BLOCKCHAIN(MAINNET) SYSTEMD STATUS ***
    * bitcoind.service - Bitcoin-Daemon
      Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
       Active: active(running) since Tue 2022-01-18 11:51:52 PST; 24min ago
     Process: 2932 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED(code= exited, status= 0 / SUCCESS)
      Process: 2934 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log(code=exited, status=0/SUCCESS)
     Main PID: 2985 (bitcoind)
        Tasks: 18 (limit: 4472)
       CGroup: /system.slice/bitcoind.service
               `-2985 /usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log

    Jan 18 11:51:51 raspiblitz systemd[1]: Starting Bitcoin-Daemon...
    Jan 18 11:51:52 raspiblitz systemd[1]: Started Bitcoin-Daemon.

    *** LAST BLOCKCHAIN(MAINNET) ERROR LOGS ***
    sudo journalctl -u bitcoind -b --no-pager -n8
    -- Logs begin at Tue 2022-01-18 11:50:14 PST, end at Tue 2022-01-18 12:16:12 PST. --
    Jan 18 11:51:51 raspiblitz systemd[1]: Starting Bitcoin-Daemon...
    Jan 18 11:51:52 raspiblitz systemd[1]: Started Bitcoin-Daemon.

    *** LAST BLOCKCHAIN(MAINNET) 20 INFO LOGS ***
    sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
    2022-01-18T19:53:05Z New outbound peer connected: version: 70016, blocks=719325, peer=0 (manual)
    2022-01-18T19:53:19Z New outbound peer connected: version: 70016, blocks=719325, peer=1 (outbound-full-relay)
    2022-01-18T19:53:30Z New outbound peer connected: version: 70016, blocks=719325, peer=3 (outbound-full-relay)
    2022-01-18T19:53:38Z New outbound peer connected: version: 70016, blocks=719325, peer=4 (outbound-full-relay)
    2022-01-18T19:53:45Z New outbound peer connected: version: 70016, blocks=719325, peer=5 (outbound-full-relay)
    2022-01-18T19:53:58Z New outbound peer connected: version: 70016, blocks=719325, peer=6 (outbound-full-relay)
    2022-01-18T19:54:26Z New outbound peer connected: version: 70016, blocks=719325, peer=7 (outbound-full-relay)
    2022-01-18T19:54:47Z New outbound peer connected: version: 70016, blocks=719325, peer=9 (outbound-full-relay)
    2022-01-18T19:54:54Z New outbound peer connected: version: 70016, blocks=719325, peer=10 (outbound-full-relay)
    2022-01-18T19:55:01Z New outbound peer connected: version: 70016, blocks=719325, peer=11 (outbound-full-relay)

    -- Logs begin at Tue 2022-01-18 11:50:14 PST, end at Tue 2022-01-18 12:16:12 PST. --
    Jan 18 11:51:51 raspiblitz systemd[1]: Starting Bitcoin-Daemon...
    Jan 18 11:51:52 raspiblitz systemd[1]: Started Bitcoin-Daemon.

    *** LAST BLOCKCHAIN(MAINNET) 20 INFO LOGS ***
    sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
    2022-01-18T19:53:05Z New outbound peer connected: version: 70016, blocks=719325, peer=0 (manual)
    2022-01-18T19:53:19Z New outbound peer connected: version: 70016, blocks=719325, peer=1 (outbound-full-relay)
    2022-01-18T19:53:30Z New outbound peer connected: version: 70016, blocks=719325, peer=3 (outbound-full-relay)
    2022-01-18T19:53:38Z New outbound peer connected: version: 70016, blocks=719325, peer=4 (outbound-full-relay)
[... 136 lines cut ...]
    *** SYSTEM STATUS(can take some seconds to gather) ***
    localIP = '192.168.1.101'
    tempCelsius='62.3'
    uptime=1550
    upsStatus='OFF'
    startcountBlockchain=1
    bitcoinActive=1
    blockchainPeers=12
    blockchainHeight=719327
    initialSync=0
    syncProgress=100.00
    startcountLightning=0
    lndActive=0
    clActive=0
    CLwalletLocked=0
    blitzTUIActive=0
    blitzTUIRestarts=0
    vagrant=0
    scriptRuntime=0

    *** OPTION: SHARE THIS DEBUG OUTPUT ***
    An easy way to share this debug output on GitHub or on a support chat
    use the following command and share the resulting link:
    debug | torsocks nc termbin.com 9999
```

nkinnan, 2022-01-19:
I think I have found the issue, but I'm new to this linux / bash stuff so please check me.  It appears to be an issue with the value of ```lightning``` being expected to equal ```""``` when disabled, but it is instead set to ```none```.  Probably a regression from changing what value signified "lighting not set up" from an empty string to the string "none" sometime in the past.  All I did to hit it is disable lightning in the initial setup screens after etching from the prebuilt image.

In ```/home/admin/00raspiblitz.sh```:
```
# CONFIGFILE - configuration of RaspiBlitz
configFile="/mnt/hdd/raspiblitz.conf"
source ${configFile} 2>/dev/null

# INFOFILE - state data from bootstrap
infoFile="/home/admin/raspiblitz.info"

<...snip...>

# get system state information raspiblitz.info
source ${infoFile}

<...snip...>

  #####################################
  # INITIAL BLOCKCHAIN SYNC (SUBLOOP)
  #####################################
  if [ "${lightning}" == "" ]; then syncedToChain=1; fi
  if [ "${setupPhase}" == "done" ] && [ "${state}" == "ready" ] && [ "${syncedToChain}" != "1" ]; then
    /home/admin/setup.scripts/eventBlockchainSync.sh ssh
    sleep 10
    continue
  fi
```

and in ```/mnt/hdd/raspiblitz.conf```:
```
lightning=none
```

and in ```/home/admin/raspiblitz.info```:
```
state=ready
setupPhase='done'
```

So because ```lightning``` is set to ```none``` instead of ```""```, this condition ```if [ "${lightning}" == "" ]; then syncedToChain=1; fi``` will evaluate to false, thus never setting ```syncedToChain``` true, invoking eventBlockchainSync.sh regardless of the value of ```setupPhase``` and ```state```.  And as you can see, those conditions do evaluate true, it is the ```syncedToChain``` variable only that is causing me to go into the sync script.

So now we're in ```eventBlockchainSync.sh```:
```
    # determine to loop or not
    loop=0
    if [ ${loopUntilSynced} -eq 1 ] && [ "${syncedToChain}" == "0" ]; then
        # loop until synced to chain
        loop=1
        sleep 3
    fi
```

I'm not quite sure how ```loopUntilSynced``` is set (it clearly must be) since I don't see that argument being passed in from ```00raspiblitz.sh``` but the ```syncedToChain``` variable appears to be set in ```/home/admin/config.scripts/blitz.statusscan.sh```.

Now this is where things get a bit sticky for me.  I've shown how and why the sync script gets invoked, but why it doesn't exit requires a deeper of this stuff than I'm able to demonstrate as a windows dev ;)  Hopefully you can take it from here.  Probably a similar regression to what caused the script to be invoked at all, and it expects a disabled lightning network to sync.

All you need to do to repro is deselect lightning on the initial setup screen.

nkinnan, 2022-01-19:
A quick fix I can do myself would be nice too - I now know that I can just ctrl-c and run "menu" and it should operate normally (I *think*) but whether I should patch the value of "lightning" in raspiblitz.conf or patch the script(s) (one or both of them), I don't know.

rootzoll, 2022-01-19:
Hi. @nkinnan thanks for already going so deep into it and isolating the bug. I think bes in your case as a quick fix is setting `lightning` in raspiblitz.conf manually to an empty string.

rootzoll, 2022-02-17:
Setup & syncing up with lightning disabled tested OK with RC2. Closing issue for release.

closed 2022-02-17 by rootzoll