raspiblitz/raspiblitz #2904: Failed to start btc-rpc-explorer (and no GUI on the LCD display)

nkinnan, 2022-01-20:
The two may or may not be related, but after installing a handful of services that looked useful, I get this output on the LCD on startup, and then it just continues displaying it without jumping into the GUI.

```
<snip>
[FAILED] Failed to start btc-rpc-explorer.
See 'systemctl status btc-rpc-explorer.service' for details.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Graphical Interface.
         Starting Update UTMP about System Runlevel Changes...
[  OK  ] Started Update UTMP about System Runlevel Changes.
```

running ```systemctl status btc-rpc-explorer.service``` gives this output:

```
admin@raspiblitz:~ $ systemctl status btc-rpc-explorer.service
* btc-rpc-explorer.service - btc-rpc-explorer
   Loaded: loaded (/etc/systemd/system/btc-rpc-explorer.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2022-01-19 19:21:55 PST; 6s ago
  Process: 5983 ExecStartPre=/home/admin/config.scripts/bonus.btc-rpc-explorer.sh prestart (code=exited, status=0/SUCCES
 Main PID: 6037 (npm start)
    Tasks: 23 (limit: 4472)
   CGroup: /system.slice/btc-rpc-explorer.service
           |-6037 npm start
           |-6277 sh -c node ./bin/www
           `-6278 node ./bin/www
lines 1-10/10 (END)
```

And then nothing else, no GUI, no additional messages on the LCD.  I can remote in and things appear to be working normally though I haven't attempted to actually use the rpc explorer yet but everything indicates it started and is running normally.

I'd note that all the processing and index/db building has completed from the services I installed as well (or at least there is minimal/no CPU or disk access is going on anymore).  

I can provide a list of services installed if needed, or inspect files, or carry out any debug steps if needed.

Last 20 btc-rpc-explorer logs from the ```debug``` command:
```
*** LAST 20 BTC-RPC-Explorer LOGS ***
sudo journalctl -u btc-rpc-explorer -b --no-pager -n20
-- Logs begin at Wed 2022-01-19 19:19:06 PST, end at Wed 2022-01-19 19:33:09 PST. --
Jan 19 19:21:56 raspiblitz npm[6037]: > node ./bin/www
Jan 19 19:21:56 raspiblitz npm[6037]: 2022-01-20T03:21:56.855Z btcexp:app Searching for config files...
Jan 19 19:21:56 raspiblitz npm[6037]: 2022-01-20T03:21:56.858Z btcexp:app Config file found at /home/btcrpcexplorer/.config/btc-rpc-explorer.env, loading...
Jan 19 19:21:56 raspiblitz npm[6037]: 2022-01-20T03:21:56.861Z btcexp:app Config file not found at /etc/btc-rpc-explorer/.env, continuing...
Jan 19 19:21:56 raspiblitz npm[6037]: 2022-01-20T03:21:56.861Z btcexp:app Config file not found at /home/btcrpcexplorer/btc-rpc-explorer/.env, continuing...
Jan 19 19:21:59 raspiblitz npm[6037]: 2022-01-20T03:21:59.938Z btcexp:app Default cacheId '3.2.0'
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.038Z btcexp:app Enabling view caching (performance will be improved but template edits will not be reflected)
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.063Z btcexp:app Environment(development) - Node: v14.17.6, Platform: linux, Versions: {"node":"14.17.6","v8":"8.4.371.23-node.76","uv":"1.41.0","zlib":"1.2.11","brotli":"1.0.9","ares":"1.17.2","modules":"83","nghttp2":"1.42.0","napi":"8","llhttp":"2.1.3","openssl":"1.1.1l","cldr":"39.0","icu":"69.1","tz":"2021a","unicode":"13.0"}
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.215Z btcexp:app Using sourcecode version as cacheId: '0e4ff45d75'
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.215Z btcexp:app Starting BTC RPC Explorer, v3.2.0 (commit: '0e4ff45d75', date: 2021-08-10) at http://0.0.0.0:3020/
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.216Z btcexp:app Connecting to RPC node at 127.0.0.1:8332
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.233Z btcexp:app Verifying RPC connection...
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.240Z btcexp:app Loading mining pools config
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.322Z btcexp:app RPC Connected: version=220000 subversion=/Satoshi:22.0.0/, parsedVersion(used for RPC versioning)=22.0.0, protocolversion=70016, chain=main, services=[NETWORK, BLOOM, WITNESS, NETWORK_LIMITED]
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.323Z btcexp:app Loading historical data for chain=main
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.324Z btcexp:app Skipping performance-intensive task: fetch UTXO set summary. This is skipped due to the flag 'slowDeviceMode' which defaults to 'true' to protect slow nodes. Set this flag to 'false' to enjoy UTXO set summary details.
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.324Z btcexp:app Skipping performance-intensive task: fetch last 24 hrs of blockstats to calculate transaction volume. This is skipped due to the flag 'slowDeviceMode' which defaults to 'true' to protect slow nodes. Set this flag to 'false' to enjoy UTXO set summary details.
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.325Z btcexp:app txindex check: trying getindexinfo
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.339Z btcexp:app txindex check: getindexinfo={"txindex":{"synced":true,"best_block_height":719548},"basic block filter index":{"synced":true,"best_block_height":719548}}
Jan 19 19:22:00 raspiblitz npm[6037]: 2022-01-20T03:22:00.339Z btcexp:app txindex check: available!
```

Contents of raspiblitz.log:
```
***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Wed 19 Jan 2022 07:20:35 PM PST
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Jan 18 20:15 .
drwxr-xr-x 119 root root   4096 Jan 19 09:21 ..
-rw-r--r--   1 root root 565189 Jan 18 20:15 moduli
-rw-r--r--   1 root root   1580 Jan 18 20:15 ssh_config
-rw-r--r--   1 root root   3271 Jan 19 08:53 sshd_config
-rw-------   1 root root   1381 Jan 18 20:15 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Jan 18 20:15 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Jan 18 20:15 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Jan 18 20:15 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Jan 18 20:15 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Jan 18 20:15 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Jan 18 20:15 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Jan 18 20:15 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Jan 18 20:15 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=1
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 1
Checking for UASP deactivation ...
HDD already part of system: 1
load configfile data
running LND users credentials update
###### SYNCING MACAROONS, RPC Password AND TLS Certificate ######
# make sure LND app-data directories exist

# copy macaroons to central app-data directory and ensure unix ownerships and permissions
# make sure admin has a symlink at ~/.lnd to /mnt/hdd/app-data/lnd/
# make sure LND conf is readable and symlinked
# make sure TLS certificate is readable and symlinked
Checking if additional backup device is configured .. (0BCE-EBD5)
Yes - Mounting BackupDrive: 0BCE-EBD5
### blitz.backupdevice.sh ###
# BACKUP DEVICE MOUNT
# OK BackupDrive mounted to: /mnt/backup
isMounted=1
CLEANING TEMP DRIVE/FOLDER
OK: Temp cleaned
DONE BOOTSTRAP
```

nkinnan, 2022-01-20:
I can get the display to show a QR code by going to the ELECTRS->CONNECT menu, but when exiting it goes back to the bootup text display.  Was hoping that'd kick something, but no.

The rpc-explorer at http://192.168.x.x:3020/ seems to be working fine.

nkinnan, 2022-01-20:
Disabling and re-enabling SETTINGS->Touchscreen seems to have resolved the screen issue (no idea, it was persistent across reboots though) and the rpc-explorer thing seems like a false alarm.  Not sure what's really going on here but it doesn't seem serious.

rootzoll, 2022-01-24:
The Touchscreen is up for refactor with v1.8.1 - its a bit of a construction site at the moment.

closed 2023-06-07 by cstenglein