raspiblitz/raspiblitz #2908: Given the UX is resolution locked, framebuffer size should be set in /boot/config.txt

nkinnan, 2022-01-24:
After unsuccessfully attempting to modify the scripts to handle different screen resolutions, I had success with the following settings which match the hard-coded values in your UX.  Without these settings, if touch UI is enabled, a significant portion of the screen to the right and bottom is left black.  Without touch UI, IIRC it just added extra rows and columns leaving the text very small but centered.

/boot/config.txt:
```
framebuffer_width=480
framebuffer_height=320
```

This resolves the issue completely.

openoms, 2022-01-24:
Great find! Which type of screen are you using?

nkinnan, 2022-01-24:
I'm using this one which is native 800x480: https://www.amazon.com/gp/product/B07XBVF1C9

Its based on the MPI4008 and uses this driver: https://github.com/goodtft/LCD-show

More details: http://www.lcdwiki.com/4inch_HDMI_Display-C

The framebuffer setting gives me some black bars on the left/right ("reverse letterbox") but this is expected due to the aspect ratio difference.  

The only minor issue is that the touchscreen active area includes those black bars on the left/right and is "horizontally scaled" onto the active screen area.  I don't find this to be a big concern though, and haven't bothered running the touchscreen calibration in the menu which would probably correct for it.

rootzoll, 2022-01-24:
Will add this for the milestone 1.8.1 where the refactor of the LCD/Touchscreen is one main focus .. but if anybody likes to make a PR & test it accross multiple setups ... we can add this sooner.

nkinnan, 2022-01-25:
This is sort of a "hack" to work around the resolution dependence of the current iteration of the UI.  IMO it would be better to put this in ASAP for those with different resolutions on the current release, and explicitly aim to *avoid* using this workaround on the next big refactor of the UI implementation.

rootzoll, 2022-01-25:
OK .. so I will add this to the `blitz.display.sh` for `hdmi` & `lcd` mode for the next v1.7.2 rc.

rootzoll, 2022-01-25:
@nkinnan I tested in HDMI mode with the framebuffer settings on ... but on my normal HDMI devices I dont get there any signal anymore. Could it be that this then only works for HDMI devices that somehow support this low-resolution?

Tending to not put that change in if it breaks output for normal HDMI devices at the moment

nkinnan, 2022-01-26:
Yeah, likely your monitors don't support that resolution.  This change would only make sense for small pi displays, not large monitors.  Perhaps it could be put in as an option under the "Node Settings & Options" menu alongside the rotate and touchscreen toggles?

nkinnan, 2022-01-26:
On second thought, I *think* this just changes the framebuffer size, not the resolution sent to the screen.  If that's so, I'm not sure why it would cause any issues with your monitor.  You can double-check the monitor is getting the correct settings in raspi-config.

I did have an issue with the screen going black in this configuration with touch UI enabled, but it disappeared over a couple reboots.