raspiblitz/raspiblitz #2909: LCD Display overlaps

22620303, 2022-01-24:
I am trying to set up raspiblitz, but when I insert the flashed SD card and power the pi on, the display seems to be overlapping and is unreadable and unusable. I have tried to rotate the display using "display_rotate=3", but it remains unreadable. 
[image]
[image]
I can't seem to find a solution, so any help would be appreciated.

openoms, 2022-01-24:
Never used this or seen else with this display, but try rotating with 90 degrees, ie. set `display_rotate=2`
Also check if this helps: https://github.com/rootzoll/raspiblitz/issues/2908

22620303, 2022-01-25:
I tried both suggestions and unfortunately neither one of them worked.

openoms, 2022-01-25:
I haven't got this display to test, but could help if you share the exact model if someone else tries it.

How does your display look now after rotating 90 degrees?