raspiblitz/raspiblitz #2913: [Security] Pwnkit - CVE-2021-4034

openoms, 2022-01-26:
https://www.zdnet.com/article/major-linux-policykit-security-vulnerability-uncovered-pwnkit/
https://www.bleepingcomputer.com/news/security/linux-system-service-bug-gives-root-on-all-major-distros-exploit-released/
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-4034

seems like a good idea to run:
`sudo chmod 0755 /usr/bin/pkexec`
until there is a patch available.

Will follow up, but please comment with new developments.

Example of running the temporary mitigation:
```
ls -la /usr/bin/pkexec
-rwsr-xr-x 1 root root 23440 Jun  3  2021 /usr/bin/pkexec

sudo chmod 0755 /usr/bin/pkexec

ls -la /usr/bin/pkexec
-rwxr-xr-x 1 root root 23440 Jun  3  2021 /usr/bin/pkexec
```

openoms, 2022-01-26:
The patch is available on the 64bit RapberryOS as well so run:
```
sudo apt update && sudo apt upgrade -y
```
ASAP
https://twitter.com/openoms/status/1486435300226310157

openoms, 2022-01-27:
As the update is out this can be the part of the next RC and v1.7.2 so can be closed then.

rootzoll, 2022-02-17:
Will be part of the v1.7.2 release .. closing issue.

closed 2022-02-17 by rootzoll