raspiblitz/raspiblitz #2917: [Errno 28] No space left on device

noobedo, 2022-01-29:
Hi my clightning node shut down. I tried to do a backup/rescue but i get an error saying [Errno 28] No space left on device. I am pretty shure i have alot of space on my hdd 1TB but micro sd 32G look full

How do i resolve this?
[image]
[image]

rootzoll, 2022-01-29:
The problem is your sd card - not the hdd. To locate what takes that space on the sd card please run the following command & post the results here: `sudo du -a /var | sort -n -r | head -n 10`

Basically to resolve: You can just turn the node off with command `off` and flash a new sd card and recover.

noobedo, 2022-01-29:
Ok thanks. I will turn node off, flash a new card and try again. Btw here is results:
[image]

openoms, 2022-01-29:
Solved on Telegram.
Related: https://github.com/rootzoll/raspiblitz/issues/2869

Solution:

To compact the CL backup run:
```
config.scripts/cl-plugin.backup.sh compact
```

to reinitialize the backup plugin in case of and error run:
```
config.scripts/cl-plugin.backup.sh on
```

will also need to delete the dated backup files from `/home/bitcoin`
```
sudo rm /home/bitcoin/lightningd.sqlite3.backup.*
```

@noobedo if these solved your issue, close please.

noobedo, 2022-01-29:
I have now run recovery with fresh sd card and looks like it still have problems 

sudo journalctl -u lightningd
[image]

openoms, 2022-01-30:
Did you encrypt your hsm_secret (c-lightning wallet)? 

Try decrypting it from the REPAIR-CL menu.

noobedo, 2022-01-30:
Yes i encypted it. Trying for a second time to recover from a newly flashed sd card atm. I will try to decrypt from the REPAIR-CL menu once its ready.

noobedo, 2022-01-30:
Ok, after second failed attempt at unlocking cl wallet during recovery i decrypted the wallet from the REPAIR-CL meny. Decryption of cl wallet seemed to work. Now waiting for Lightning progress. Screen blink every couple of minutes with some invalid numeric syntax error message.

noobedo, 2022-01-30:
Did a new fresh install with only blockchain data, and imported cl rescue file from FILERESTORE and now waiting again. Will report back tomorrow.

noobedo, 2022-01-31:
Looks like plugin-backup.py is working its way through history one state change at the time? I guess i will just have to wait.. 
[image]

noobedo, 2022-02-01:
Ok, looks like database and the rescuefile may have been corrupted by the cold restet when microsd became full. I will try with an older rescuefile. I got three resent rescuefiles from after the node shut down so they may all be corrupted but i will try to anyway.

[image]

noobedo, 2022-02-01:
Old Rescue file had wallet encrypted and would not open even after entering encryption password, and node fall back to open latest unencrypted corrupted database.

noobedo, 2022-02-01:
Latest try
[image]

[image]

openoms, 2022-02-01:
The latest try looks ok.

Check
`cl listpeers`
and
`cl listfunds`

noobedo, 2022-02-01:
[image]

openoms, 2022-02-01:
The wallet does not unlock because of the error pictured above.
If you can paste those lines as text will be eaiser to share with the c-lightning team.

noobedo, 2022-02-01:
I now unlocked with the decrypt option in menu

Enter hsm_secret password:
# Getting the password from /dev/shm/.cl.pw
Enter hsm_secret password:
Successfully decrypted hsm_secret, be careful now :-).
# Decrypted successfully

# Shredding the passwordFile

shred: /dev/shm/.cl.pw: pass 1/4 (random)...
shred: /dev/shm/.cl.pw: pass 2/4 (random)...
shred: /dev/shm/.cl.pw: pass 3/4 (random)...
shred: /dev/shm/.cl.pw: pass 4/4 (000000)...
shred: /dev/shm/.cl.pw: removing
shred: /dev/shm/.cl.pw: renamed to /dev/shm/000000
shred: /dev/shm/000000: renamed to /dev/shm/00000
shred: /dev/shm/00000: renamed to /dev/shm/0000
shred: /dev/shm/0000: renamed to /dev/shm/000
shred: /dev/shm/000: renamed to /dev/shm/00
shred: /dev/shm/00: renamed to /dev/shm/0
shred: /dev/shm/.cl.pw: removed
# Decrypted the hsm_secret for C-lightning mainnet
Removed /etc/systemd/system/multi-user.target.wants/lightningd.service.
# Create /etc/systemd/system/lightningd.service

Then i see this in log:
sudo journalctl -u lightningd
-- Logs begin at Tue 2022-02-01 22:20:23 GMT, end at Tue 2022-02-01 23:12:14 GMT. --
Feb 01 22:21:04 Supernode systemd[1]: Starting c-lightning daemon on mainnet...
Feb 01 22:21:04 Supernode systemd[1]: Started c-lightning daemon on mainnet.
Feb 01 22:21:04 Supernode sh[553]: cat: /dev/shm/.cl.pw: No such file or directory
Feb 01 22:21:04 Supernode sh[553]: cat: /dev/shm/.cl.pw: No such file or directory
Feb 01 22:21:09 Supernode sh[553]: /usr/local/bin/lightningd: --encrypted-hsm: Could not read pass from stdin.
Feb 01 22:21:09 Supernode systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Feb 01 22:21:09 Supernode systemd[1]: lightningd.service: Failed with result 'exit-code'.
Feb 01 22:21:39 Supernode systemd[1]: lightningd.service: Service RestartSec=30s expired, scheduling restart.
Feb 01 22:21:39 Supernode systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 1.
Feb 01 22:21:39 Supernode systemd[1]: Stopped c-lightning daemon on mainnet.
Feb 01 22:21:39 Supernode systemd[1]: Starting c-lightning daemon on mainnet...
Feb 01 22:21:40 Supernode systemd[1]: Started c-lightning daemon on mainnet.
Feb 01 22:21:40 Supernode sh[2972]: cat: /dev/shm/.cl.pw: No such file or directory
Feb 01 22:21:40 Supernode sh[2972]: cat: /dev/shm/.cl.pw: No such file or directory
Feb 01 22:21:41 Supernode sh[2972]: /usr/local/bin/lightningd: --encrypted-hsm: Could not read pass from stdin.
Feb 01 22:21:41 Supernode systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Feb 01 22:21:41 Supernode systemd[1]: lightningd.service: Failed with result 'exit-code'.
Feb 01 22:22:26 Supernode systemd[1]: lightningd.service: Service RestartSec=30s expired, scheduling restart.
Feb 01 22:22:26 Supernode systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2.
Feb 01 22:22:26 Supernode systemd[1]: Stopped c-lightning daemon on mainnet.
Feb 01 22:22:26 Supernode systemd[1]: Starting c-lightning daemon on mainnet...
Feb 01 22:22:26 Supernode systemd[1]: Started c-lightning daemon on mainnet.
Feb 01 22:22:26 Supernode sh[5169]: cat: /dev/shm/.cl.pw: No such file or directory
Feb 01 22:22:26 Supernode sh[5169]: cat: /dev/shm/.cl.pw: No such file or directory
Feb 01 22:22:28 Supernode sh[5169]: /usr/local/bin/lightningd: --encrypted-hsm: Could not read pass from stdin.
Feb 01 22:22:28 Supernode systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE

noobedo, 2022-02-01:
And calling cllog:

cllog

2022-02-01T23:17:39.250Z INFO    plugin-backup.py: root        : Comparing backup version 2406478 versus first write version 2406479
2022-02-01T23:17:39.250Z INFO    plugin-backup.py: Comparing backup version 2406478 versus first write version 2406479
2022-02-01T23:17:39.250Z INFO    plugin-backup.py: root        : Versions match up
2022-02-01T23:17:39.250Z INFO    plugin-backup.py: Versions match up
2022-02-01T23:17:40.910Z INFO    connectd: Static Tor service onion address: "2de3vrdvrmffsxg4pqbmiolkbdvtwfhsvmncxhjgyfgghpe3232gc6ad.onion:9736,127.0.0.1:9736" bound from extern port 9736 
2022-02-01T23:17:58.357Z INFO    plugin-bcli: bitcoin-cli initialized and connected to bitcoind.

2022-02-01T23:18:29.268Z INFO    plugin-backup.py: root        : Comparing backup version 2406480 versus first write version 2406481
2022-02-01T23:18:29.268Z INFO    plugin-backup.py: Comparing backup version 2406480 versus first write version 2406481
2022-02-01T23:18:29.268Z INFO    plugin-backup.py: root        : Versions match up
2022-02-01T23:18:29.269Z INFO    plugin-backup.py: Versions match up
2022-02-01T23:18:31.001Z INFO    connectd: Static Tor service onion address: "2de3vrdvrmffsxg4pqbmiolkbdvtwfhsvmncxhjgyfgghpe3232gc6ad.onion:9736,127.0.0.1:9736" bound from extern port 9736 
2022-02-01T23:18:48.444Z INFO    plugin-bcli: bitcoin-cli initialized and connected to bitcoind.

2022-02-01T23:19:19.510Z INFO    plugin-backup.py: root        : Comparing backup version 2406482 versus first write version 2406483
2022-02-01T23:19:19.510Z INFO    plugin-backup.py: Comparing backup version 2406482 versus first write version 2406483
2022-02-01T23:19:19.510Z INFO    plugin-backup.py: root        : Versions match up
2022-02-01T23:19:19.510Z INFO    plugin-backup.py: Versions match up
2022-02-01T23:19:21.184Z INFO    connectd: Static Tor service onion address: "2de3vrdvrmffsxg4pqbmiolkbdvtwfhsvmncxhjgyfgghpe3232gc6ad.onion:9736,127.0.0.1:9736" bound from extern port 9736 
2022-02-01T23:19:38.771Z INFO    plugin-bcli: bitcoin-cli initialized and connected to bitcoind.

noobedo, 2022-02-01:
Removed /etc/systemd/system/multi-user.target.wants/lightningd.service.
# Create /etc/systemd/system/lightningd.service

[Unit]
Description=c-lightning daemon on mainnet

[Service]
ExecStartPre=-/home/admin/config.scripts/cl.check.sh prestart mainnet
ExecStart=/bin/sh -c '(cat /dev/shm/.cl.pw;echo;cat /dev/shm/.cl.pw) | /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/config --encrypted-hsm'
User=bitcoin
Group=bitcoin
Type=simple
Restart=always
TimeoutSec=120
RestartSec=30
StandardOutput=null
StandardError=journal

# Hardening measures
PrivateTmp=true
ProtectSystem=full
NoNewPrivileges=true
PrivateDevices=true

[Install]
WantedBy=multi-user.target

Created symlink /etc/systemd/system/multi-user.target.wants/lightningd.service → /etc/systemd/system/lightningd.service.
# Enabled the lightningd.service
# Started the lightningd.service
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Waiting to unlock wallet ... 
# Failed to unlock the lightningd wallet - giving up after 1 minute
# Check: sudo journalctl -u lightningd
Removed /etc/systemd/system/multi-user.target.wants/lightningd.service.
# Create /etc/systemd/system/lightningd.service

[Unit]
Description=c-lightning daemon on mainnet

[Service]
ExecStartPre=-/home/admin/config.scripts/cl.check.sh prestart mainnet
ExecStart=/bin/sh -c '(cat /dev/shm/.cl.pw;echo;cat /dev/shm/.cl.pw) | /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/config --encrypted-hsm'
User=bitcoin
Group=bitcoin
Type=simple
Restart=always
TimeoutSec=120
RestartSec=30
StandardOutput=null
StandardError=journal

# Hardening measures
PrivateTmp=true
ProtectSystem=full
NoNewPrivileges=true
PrivateDevices=true

[Install]
WantedBy=multi-user.target

Created symlink /etc/systemd/system/multi-user.target.wants/lightningd.service → /etc/systemd/system/lightningd.service.
# Enabled the lightningd.service
# Started the lightningd.service
# Checking dependencies
WARNING: You are using pip version 21.2.4; however, version 22.0 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
# Stop the lightningd.service
# Backup the existing old backup on the SDcard
/home/bitcoin/.lightning/bitcoin/backup.lock
# Initialize the backup plugin
Found an existing database at /home/bitcoin/.lightning/bitcoin/lightningd.sqlite3, initializing the backup with a snapshot
Successfully written initial snapshot to file:///home/bitcoin/lightningd.sqlite3.backup
Initialized backup backend file:///home/bitcoin/lightningd.sqlite3.backup, you can now start c-lightning
# Started the lightningd.service

# DONE - lightningd is now starting
# Check that CL is starting up correctly and your old channels & funds are restored.
# Take into account that some channels might have been force closed in the meanwhile.

openoms, 2022-02-01:
The problem is not with the encryption.
The error message is misleading because it cannot unlock because it cannot start.

Now after decryption need to run
`config.scripts/cl.install.service.sh`
to fix the systemd service. This happens automatically in the next release.

noobedo, 2022-02-01:
admin@Supernode:~ $ config.scripts/cl.install.service.sh
-bash: config.scripts/cl.install.service.sh: No such file or directory
admin@Supernode:~ $

noobedo, 2022-02-01:
Ok so the problem with restoring the backup is because the wallet was encrypted, and there is some missing script to unlock on restore?

openoms, 2022-02-01:
> admin@Supernode:~ $ config.scripts/cl.install.service.sh
> -bash: config.scripts/cl.install.service.sh: No such file or directory
> admin@Supernode:~ $ 
> 
> 

it is 
`config.scripts/cl.install-service.sh`
corrected the typo

openoms, 2022-02-01:
> Ok so the problem with restoring the backup is because the wallet was encrypted, and there is some missing script to unlock on restore?
>  

it seems that the main issue is the
 `database disk image is malformed`
error, which I haven't seen before and would like to share with the C-lightning devs.

noobedo, 2022-02-01:
Ok I tried the corrected typo, seem to work but node is still not starting. 

sudo journalctl -u lightningd
-- Logs begin at Tue 2022-02-01 23:31:56 GMT, end at Tue 2022-02-01 23:41:33 GMT. --
Feb 01 23:32:36 Supernode systemd[1]: Starting c-lightning daemon on mainnet...
Feb 01 23:32:37 Supernode systemd[1]: Started c-lightning daemon on mainnet.
Feb 01 23:32:44 Supernode sh[622]: lightning_connectd: Connecting stream socket to Tor service: Connection refused
Feb 01 23:32:44 Supernode sh[622]: lightningd: connectd failed (exit status 1), exiting.
Feb 01 23:32:44 Supernode systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Feb 01 23:32:44 Supernode systemd[1]: lightningd.service: Failed with result 'exit-code'.
Feb 01 23:33:14 Supernode systemd[1]: lightningd.service: Service RestartSec=30s expired, scheduling restart.
Feb 01 23:33:14 Supernode systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 1.
Feb 01 23:33:14 Supernode systemd[1]: Stopped c-lightning daemon on mainnet.
Feb 01 23:33:14 Supernode systemd[1]: Starting c-lightning daemon on mainnet...
Feb 01 23:33:14 Supernode systemd[1]: Started c-lightning daemon on mainnet.
Feb 01 23:34:29 Supernode sh[2960]: Error executing statement: wallet/wallet.c:3575: DELETE FROM blocks WHERE height > ?: database
Feb 01 23:34:29 Supernode systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Feb 01 23:34:29 Supernode systemd[1]: lightningd.service: Failed with result 'exit-code'.
Feb 01 23:34:59 Supernode systemd[1]: lightningd.service: Service RestartSec=30s expired, scheduling restart.
Feb 01 23:34:59 Supernode systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2.
Feb 01 23:34:59 Supernode systemd[1]: Stopped c-lightning daemon on mainnet.
Feb 01 23:34:59 Supernode systemd[1]: Starting c-lightning daemon on mainnet...
Feb 01 23:35:00 Supernode systemd[1]: Started c-lightning daemon on mainnet.
Feb 01 23:35:27 Supernode sh[7083]: Error executing statement: wallet/wallet.c:3575: DELETE FROM blocks WHERE height > ?: database
Feb 01 23:35:27 Supernode systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Feb 01 23:35:27 Supernode systemd[1]: lightningd.service: Failed with result 'exit-code'.
Feb 01 23:35:57 Supernode systemd[1]: lightningd.service: Service RestartSec=30s expired, scheduling restart.
Feb 01 23:35:57 Supernode systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 3.
Feb 01 23:35:57 Supernode systemd[1]: Stopped c-lightning daemon on mainnet.
Feb 01 23:35:57 Supernode systemd[1]: Starting c-lightning daemon on mainnet...
lines 1-27

noobedo, 2022-02-01:
So we are getting somewhere with this debugging. I will have to go now. Checking inn again tomorrow. Thanks alot for your help!

openoms, 2022-02-02:
Answer from cdecker https://t.me/lightningd/10053

To confirm if the database is readable do a direct sqlite query like in https://github.com/rootzoll/raspiblitz/blob/dev/FAQ.cl.md#sqlite3-queries

```
sudo -u bitcoin sqlite3 /home/bitcoin/.lightning/bitcoin/lightningd.sqlite3 'select short_channel_id, timestamp, cause, message from channel_state_changes inner join channels on channel_id = id where new_state = 7 order by timestamp'
```

noobedo, 2022-02-02:
```sudo -u bitcoin sqlite3 /home/bitcoin/.lightning/bitcoin/lightningd.sqlite3 'select short_channel_id, timestamp, cause, message from channel_state_changes inner join channels on channel_id = id where new_state = 4 order by timestamp'
717791x763x1|1641738324237197070|2|User or plugin invoked close command
717856x323x1|1641904031827940160|2|User or plugin invoked close command
717925x301x1|1641904048263026456|2|User or plugin invoked close command
717828x1276x1|1641904071115513449|2|User or plugin invoked close command
718128x931x0|1641909614351405983|3|Peer closes channel
718208x2064x1|1641936542623232527|2|User or plugin invoked close command
717865x1769x0|1642019302037365531|2|User or plugin invoked close command
717856x323x0|1642756452866498673|2|User or plugin invoked close command
719323x1684x0|1642777367433712461|2|User or plugin invoked close command
717802x2282x0|1643371765948065823|3|Peer closes channel
718356x1521x0|1643398651811122368|2|User or plugin invoked close command```

noobedo, 2022-02-02:
Looks like there is not much more to do regarding originating issue, i will take my issue with recovery to the c-lighting project and see if they can help me. Closing this issue Found this guide and entering steep learning curve. https://github.com/mandelbit/bitcoin-tutorials/blob/master/CLightningRecoverFunds.md

closed 2022-02-02 by noobedo