raspiblitz/raspiblitz #2919: 1.7.2RC testing reports

openoms, 2022-01-31:
Missing number of min GB for the SDcard when using a 16GB card with v1.7.2rc1f: 
[image]

openoms, 2022-01-31:
Also found (running in a VM) that the copy of the blockchain is not offered when starting a new node. 

I think it as ok default behaviour, but the documentation needs to change to suggest to enter the menu - REPAIR - reset chain if one wants to copy over the network instead.

rootzoll, 2022-01-31:
Yeah the change to a 32GB sd card minimum is part of the CHANGES ... its not pressing. But its that 8GB sd cards (industrial quality) seem to fade out on the markets. So I plan to use a 16GB industrial grade sd card fro the master build card from now on. Which makes a 32GB sd card the best choice for users to ensure the master image will fit all user cards out there.

Also 32GB are getting cheap now .. old users that still use a 16GB sd card still should maybe upgrade/replace to a new card anyway .. and on my one test machine where I run ALL apps installed at the same time I now reached broke the 16GB limit.

So I think its a good time to change to a 32GB sd card as a minimum. What do you think?

rootzoll, 2022-01-31:
TODOs:
- [ ] check for the missing number in sd card info

openoms, 2022-01-31:
Agree to change it in the recommendations and I am using 32GB cards for a while as they are the bast value for price even in the endurance models.

This might not be the case everywhere so maybe shouldn't break if someone uses a 16GB card (against the recommendation). If manages to flash the image it can be a disappointment to break such arbitrarily.
On a 8GB one shouldn't be able to flash any more by start if built on 16GB.

rootzoll, 2022-01-31:
> This might not be the case everywhere so maybe shouldn't break if someone uses a 16GB card (against the recommendation). If manages to flash the image it can be a disappointment to break such arbitrarily.

I mean I can take that dialog away ... but still even when I build on 16GB - that image might not work anymore on other 16GB cards because they are some bytes smaller. Thats why when building on a 16GB, we should set 32 GB as a general minimum. So it might work for some 16GB cards, but not for all ... I find that a bit confusing and would prefer a clean switch to a 32GB standard instead.

If we want to support all 16GB cards I need to keep building on a 8GB OR we need to add a way to make the file system smaller on a 16GB (rteverse filesystem expand) before I make the image file.

openoms, 2022-01-31:
> I mean I can take that dialog away ... but still even when I build on 16GB - that image might not work anymore on other 16GB cards because they are some bytes smaller. Thats why when building on a [...]

I agree, just saying that if the flashing is successful (on most 16GB cards) the dialog shouldn't break the setup.
If the the flashing fails dues to the smaller size of the card it will be obvious before booting and 32GB should be hard recommendation in the shopping list.

vv01f, 2022-01-31:
It should be possible to `truncate` an image after resizing partitions via a loopback device if you'd prefer to keep the 16 GB requirement alive for a while longer. example see https://askubuntu.com/questions/1174487/re-size-the-img-for-smaller-sd-card-how-to-shrink-a-bootable-sd-card-image

openoms, 2022-01-31:
> It should be possible to `truncate` an image after resizing partitions via a loopback device if you'd prefer to keep the 16 GB requirement alive for a while longer. example see https://askubuntu.com/questions/1174487/re-size-the-img-for-smaller-sd-card-how-to-shrink-a-bootable-sd-card-image

yes, reducing the partition simply in Linux Disks does work and it will be expanded anyway. Saying that the free space does not affect the compressed image size and it can be a soft requirement to force the minimal SDcard size.

I think it is not really needed to force the SDcard size in any way since the minimal recommended size is clearly documented.

rootzoll, 2022-02-06:
OK I added to the build process (now beginning from RC2 on) the use of the pishrink script of:
https://github.com/Drewsif/PiShrink

So I can build now on my industrialgrade 16GB sd card and then optimize the image filesystem size afterwards and when zipped new even getting below 2GB :)

rootzoll, 2022-02-17:
Using a 16GB still works now - new process for shrinking the sd catd build image showed fine for RC2 image. So closing this issue for v1.7.2 release.

closed 2022-02-17 by rootzoll