raspiblitz/raspiblitz #2920: Location of channel.backup file in README is wrong?!

mineralf, 2022-01-31:
The README.md states 
> The channel.backup is stored on the HDD and updated by LND every time a new channel is opened or closed. The latest version of this file is needed to recover all your funds. In case your HDD gets damaged RaspiBlitz always keeps a copy of the latest version of the channel.backup file on the SD card within the sub-directories of: /home/admin/.lnd/data/chain/.

I can't find it in that location. Actually searching for it reveals those 2 locations.
```
/ $ find . -name "channel.backup" 2>/dev/null
./home/admin/backups/scb/channel.backup
./boot/channel.backup
```

Maybe the README.md should be updated with the correct locations?

openoms, 2022-01-31:
it is copied from the disk to those locations, try to search everywhere:

`sudo find / -name channel.backup`

openoms, 2022-01-31:
Thank you for double checking this, the readme should be updated.

rootzoll, 2022-02-06:
Readme updated. Thanks @mineralf

closed 2022-02-06 by rootzoll

mineralf, 2022-02-07:
Thanks @rootzoll for all your work and this quick update.

I just double checked the README von dev and I'm not sure this is correct - the path quoted there looks like the HDD path not the SD card:

> The channel.backup is stored on the HDD and updated by LND every time a new channel is opened or closed. The latest version of this file is needed to recover all your funds. In case your HDD gets damaged RaspiBlitz always keeps a copy of the latest version of the channel.backup file on the SD card: `/mnt/hdd/lnd/data/chain/bitcoin/mainnet/`.

```
# df /mnt/hdd/lnd/data/chain/bitcoin/mainnet
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/sda1      960322016 511290364 400180228  57% /mnt/hdd
```

Maybe update that paragraph to include both/all locations?
> The channel.backup is stored on the HDD (`/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup`) and updated by LND every time a new channel is opened or closed. The latest version of this file is needed to recover all your funds. In case your HDD gets damaged RaspiBlitz always keeps a copy of the latest version of the channel.backup file on the SD card: `/boot/channel.backup`.

and in `/home/admin/backups/scb/` is the whole history of backups as well -  but I guess only the `/boot` is mounted if inserted in another device by default ...

I know this is some details - but if my HDD crashes and I need the backup to recover my lnd funds I would love to know where to search without getting to much panic. (apart from the great optional external backup option with NextCloud which not everybody has configured probably)

rootzoll, 2022-02-07:
The path on the HDD is the most direct way to get the channel.backup at the source. But yeah it makes sense to mention also the automated backup location on the sd card. Ill add that.