raspiblitz/raspiblitz #2921: NextCloud Password with special characters fails

mineralf, 2022-01-31:
I just generated a password (not used anywhere so it's ok to have it public here) for a NextCloud Account and when setting it up via `Settings & Options -> LND StaticChannelBackup on NextCloud` I get the following error:

```
Nextcloud Setting changed ..
/mnt/hdd/raspiblitz.conf: line 30: syntax error near unexpected token `)'
/mnt/hdd/raspiblitz.conf: line 30: `nextcloudBackupPassword==eZZv3)Z)o§q%ez'
/mnt/hdd/raspiblitz.conf: line 30: syntax error near unexpected token `)'
/mnt/hdd/raspiblitz.conf: line 30: `nextcloudBackupPassword==eZZv3)Z)o§q%ez'
/mnt/hdd/raspiblitz.conf: line 30: syntax error near unexpected token `)'
/mnt/hdd/raspiblitz.conf: line 30: `nextcloudBackupPassword==eZZv3)Z)o§q%ez'
err='Nextcloud credentials are missing'
```

Can the password be encoded/escaped or at least mention in the README which special characters are not allowed?
Backups worked with another password omitting the `)` characters

nyxnor, 2022-02-04:
the password needs to be quoted, anyway, one character you must not use in the shell is `\`, it will need to be escaped with the same character.

https://github.com/rootzoll/raspiblitz/blob/d3f042f48ad1a50f3b3ad43f1479e6056f98a28b/home.admin/config.scripts/nextcloud.upload.sh#L47

```sh
echo "nextcloudBackupPassword=\"${password}\"" >> /mnt/hdd/raspiblitz.conf
```

change the script to enforce double quotes or open your /mnt/hdd/raspiblitz.conf and use doublequotes on the value. (example: key="value" -> `nextcloudBackupPassword="=eZZv3)Z)o§q%ez"`

rootzoll, 2022-02-06:
With upcomming v1.72. we use now a dedicated script to store config values that will quote the password value. Looked good on my test. Fix will be part of v1.7.2 release.

closed 2022-02-06 by rootzoll