raspiblitz/raspiblitz #2922: parse error: Invalid numeric literal at line 1, column 14

RemoRitzmann, 2022-01-31:
I just started my own node and run into the problem that shows me this message on monitor:

The only error I can think of is, that I started to configure LND through SSH even though the blockchain was not fully downloaded.

RaspiBlitz v1.7.1 Free Mem 2118M/3844M HDDuse 113G (27%)
CPU load 2.36, 2.16, 2.10, temp 74°C 165°F
bitcoin v22.0.0 mainnet   peers
Blocks - Sync
Wallet 0 sat
/0Channels 0 sat   peers
Fees collected: 

@:9736
parse error: Invalid numeric literal at line 1, column 14
(standard_in) 1: syntax error

RemoRitzmann, 2022-01-31:
Found something similar, but does not help since I'm a newbie:
https://stackoverflow.com/questions/63133465/jq-parse-error-invalid-numeric-literal-at-line-1-column-14

RemoRitzmann, 2022-01-31:
Actual setup state:
Blockchain Progress :   28.47% 10 peers
Lightning Progress: 93.41 %      # What is this?
Since I'm here asking questions, I have another one: How can I add emojis in the LN-Name? 
I use zram not to wear the sd-card down. Do you consider using zram in the standard image?

rootzoll, 2022-02-06:
>  now I see it 92.86%, why is it lower after a reboot?

Lightning is percent based on whats already synced by Bitcoin. So this can get lower if bitcoin syncy faster then LND is catching up.

> How can I add emojis in the LN-Name?

You can try UTF-8 coded chars fro the `alias` value in the `lnd.conf` (MAINMENU > SYSTEM > LNDCONF)

> Do you consider using zram in the standard image?

zram is a research topic for possible future versions. If you like you can contribute your experience here:
https://github.com/rootzoll/raspiblitz/issues/2905

> About the: Invalid numeric 

The whole display data part got a rewrite (separation of data & view) with the upcomming v1.7.2 version. Please recheck once version is out if this error continues for you.

RemoRitzmann, 2022-02-06:
Thanking you for your answer's. I'll try utf-8 as u mentioned. I have successfully running zram already in 80 RPi3 systems and 4 Rpi4 systems. Sure I'll add to #2905 when I set it up with raspiblitz. But for now syncing takes me to long. I now download the bitchain with bitcoin-qt on my desktop and later copy it on my Pi4 system. It's only 24h left there. Next I restart setting up raspiblitz with current (2022-01-28-raspios-bullseye-arm64.zip) and your build_sdcard.sh During the coming week you'll hear more on that later on. Regards, Remo

RemoRitzmann, 2022-02-07:
I run into python problems building SD-Card with build_sdcard.sh but found solution in #2899 (I now took build_sdcard.ch) from dev-branch ant it worked through. with one error I report (btrfs-tools no installation candidate) in #2905

rootzoll, 2022-02-07:
That `btrfs-tools` error giot addressed now here: https://github.com/rootzoll/raspiblitz/commit/d1cf7ee57ee382b327cc9a477f3dd7843b4488da - should be fixed

openoms, 2022-06-28:
closing as fixed

closed 2022-06-28 by openoms