raspiblitz/raspiblitz #2923: Can't create sd card image: !!! FAIL: Base Image cannot be detected or is not supported.

tifDev, 2022-02-01:
I've downloaded the last version of raspiblitz.

**Here's what I did:**

```
./build_sdcard.sh 

*****************************************
* RASPIBLITZ SD CARD IMAGE SETUP v1.7.1 *
*****************************************
For details on optional parameters - see build script source code:
1) will use NO-INTERACTION --> 'false'
2) will use FATPACK --> 'false'
3) will use GITHUB-USERNAME --> 'rootzoll'
4) will use GITHUB-BRANCH --> 'v1.7'
5) will use DISPLAY-CLASS --> 'lcd'
6) will use TWEAK-BOOTDRIVE --> 'true'
7) will use WIFI --> 'US'
X) will use CPU-ARCHITECTURE --> 'x86_64'
NAME="Zorin OS"
VERSION="15.3"
ID=zorin
ID_LIKE=ubuntu
PRETTY_NAME="Zorin OS 15.3"
VERSION_ID="15"
HOME_URL="https://www.zorinos.com"
SUPPORT_URL="https://www.zorinos.com/help"
BUG_REPORT_URL="https://bugs.launchpad.net/zorin-os"
PRIVACY_POLICY_URL="https://www.zorinos.com/legal/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
!!! FAIL: Base Image cannot be detected or is not supported.
```

Can you pls help?

Thanks

rootzoll, 2022-02-01:
We already did much work on the new build script in the dev branch .. can you try that one?
https://github.com/rootzoll/raspiblitz/blob/dev/build_sdcard.sh

tifDev, 2022-02-01:
Different error now...

./build_sdcard.sh
*****************************************
*     RASPIBLITZ SD CARD IMAGE SETUP    *
*****************************************
For details on optional parameters - call with '--help' or check source code.
interaction=true
fatpack=true
github_user=rootzoll
branch=v1.7
display=lcd
tweak_boot_drive=true
wifi_region=US
cpu=x86_64
architecture=amd64
**\n!!! FAIL: Base Image cannot be detected or is not supported.**
NAME="Zorin OS"
VERSION="15.3"
ID=zorin
ID_LIKE=ubuntu
PRETTY_NAME="Zorin OS 15.3"
VERSION_ID="15"
HOME_URL="https://www.zorinos.com"
SUPPORT_URL="https://www.zorinos.com/help"
BUG_REPORT_URL="https://bugs.launchpad.net/zorin-os"
PRIVACY_POLICY_URL="https://www.zorinos.com/legal/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
Linux az-pc 5.4.0-91-generic #102~18.04.1-Ubuntu SMP Thu Nov 11 14:46:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
[4]+  Done                    exo-open --launch FileManager "$1" > /dev/null 2>&1

openoms, 2022-02-01:
Try Debian or Ubuntu. Zorin OS is not supported (hence the message).

tifDev, 2022-02-02:
>

tifDev, 2022-02-02:
> Try Debian or Ubuntu. Zorin OS is not supported (hence the message).

I've flashed older versions of raspiblitz with no issues...
This distro is Debian based...

openoms, 2022-02-02:
We need to indicate somehow what distros were tested. There are hundreds of Debian based OSes, including Ubuntu itself. The possibility of possible edge cases are infinite.

What makes you Zorin OS over pure Debian? Also what kind of machine you are building.

Prebuilt images are only provided for the Raspberry Pi using the 64Bit Raspberry OS (also Debian based but modded for the RPi).

tifDev, 2022-02-02:
So that means if I do this in Ubuntu or Raspbian, I get more chances to succeed?

adevoss, 2022-02-02:
> I've flashed older versions of raspiblitz with no issues... This distro is Debian based...

I should not use an older version. There are many changes. My Debian setup only installs since a few weeks with the `dev` branch.

If it is Debian based then you could try to modify the build script from line 208 onwards
`AUTO-DETECTION: OPERATINGSYSTEM`
Beware: You are the first to test the Zorin distro