raspiblitz/raspiblitz #2925: CL: encrypt hsm_secret before backup to make sure it is not running again

openoms, 2022-02-01:
To guarantee that the CL node is not started accidentally after backup the `hsm_secret` should be encrypted and locked

rootzoll, 2022-02-06:
Do you want this still for v1.7.2 or do we push for later release?

openoms, 2022-02-06:
yes, would like to get it done as just force closed a few channels because of this

rootzoll, 2022-02-06:
OK .. you should be no problem to work on this while the RC2 is out and then have it part of RC3 for quick testing before final v1.7.2 release.

openoms, 2022-02-09:
tested
- [ ] cl  BACKUP and 
- [ ] MIGRATION

restoring at the start work too both with the 
- [ ] migration and
- [ ] rescue file

can test the menu once more
- [ ]  FILERESTORE
- [ ]  SEEDRESTORE

rootzoll, 2022-02-18:
@openoms can we close issue for release?

openoms, 2022-02-18:
will run those two more tests, but not expecting new issues

openoms, 2022-02-18:
Ok, testted succesfully and closing.

Found a non-showstopper issue, to be fixed later: #2979

closed 2022-02-18 by openoms