raspiblitz/raspiblitz #2926: Trying to develop pieces to add Helipad as a Raspiblitz service

SpencerPearson, 2022-02-02:
Hello!

I have been combing over some of the contributing guidelines and working on an install script with some colleagues for Podcast Index's app Helipad, which allows podcasters to read boosts and boostagrams that come into their node:

https://github.com/Podcastindex-org/helipad

So far we have written out an install script that has an "on" and "off" function and it works great for local access over http. This is the bonus.helipad.sh script we are currently using to run Helipad on our Raspiblitz nodes:

https://github.com/SpencerPearson/helipad-for-raspiblitz

I haven't quite cracked the https support and I'm wondering what else I might need to be thinking about for this config script to work up to inclusion as an installable service in a future Raspiblitz release. I hope this is the proper place to ask! I've been using Raspiblitz for around a year but I'm new to the contributions & issues side of github. Didn't see anything mentioned about Helipad I'm issues so far so I figured this was a good starting place.

adevoss, 2022-02-02:
Don't know how official the support for Debian is.
https://github.com/SpencerPearson/helipad-for-raspiblitz/pull/3

SpencerPearson, 2022-02-02:
> Don't know how official the support for Debian is. [SpencerPearson/helipad-for-raspiblitz#3](https://github.com/SpencerPearson/helipad-for-raspiblitz/pull/3)

Yeah we ended up reverting that as it fell outside the scope of Raspiblitz:
https://github.com/SpencerPearson/helipad-for-raspiblitz/pull/6

SpencerPearson, 2022-02-02:
> If Debian is officially supported then it has to be reverted again.
> Helipad runs on Debian.

The current script already works to install Helipad on Raspiblitz. The extra dependency for pkg-config in that Pull Request 3 was suggested for getting the script to work on a non-Raspiblitz install but in the Raspiblitz environment it is redundant to include.

SpencerPearson, 2022-02-03:
Looking over the pending PR for adding squeaknode (https://github.com/rootzoll/raspiblitz/pull/2732/files) I am getting a clearer picture...if I fork the dev branch, will I need to run that on new hardware? How exactly can I mess around/test with the dev branch without goofing my current node up?

This may be a question more suited for the telegram so I'll ask on there later tonight. Mostly this is just helping me keep my thoughts and process organized and tracked, so I will close this issue for now too. Sorry for the clutter 😅

adevoss, 2022-02-04:
> with the dev branch without goofing my current node up?

You can't. Upgrade is a one way operation. So you need a second Pi or a Debian (VM) setup for testing (see #2772)

SpencerPearson, 2022-02-04:
> You can't. Upgrade is a one way operation. So you need a second Pi or a Debian (VM) setup for testing (see #2772)

Dang... I have 2 pis now but they both are living nodes...did the VM setup ever get working smoothly? Any advice you might have for how to get the dev branch rocking within a vm?

adevoss, 2022-02-06:
Yes the VM setup is working for me. See #2772.
In short
- Backup the data drive (blockchain not needed obviously)
- Install Debian with sudo on VM with two disks.
- Execute the build script with the correct options
- Reboot
- Login as admin

It will start a recovery.

SpencerPearson, 2022-02-10:
Added a draft pr #2944, I think we just need to add an UPDATE call to the config script and add https configuration...I know where to start on update and should be able to knock that out tomorrow but I'll have to do some reading up to tackle https

SpencerPearson, 2022-02-18:
Update to this issue: I've flagged #2944 as ready for review! Should be good to go for testing

rootzoll, 2022-02-19:
OK now also tested for recovery - looks good. Will be part of v1.7.2 release. Closing issue.
Thanks @SpencerPearson good work :D

closed 2022-02-19 by rootzoll