raspiblitz/raspiblitz #2929: Blockchain Progress : waiting

stevewebb951, 2022-02-04:
Had a power failure. When I try to restart raspiblitz, the booting process gets hung on "Node is Syncing" "Blockchain Progress : waiting" "Lightning Progress : waiting" "Please wait - this can take some time" "Use CTRL+c to EXIT to Terminal"

It's been hung for 10 hours. No change.

What now? Thanks.

Running RaspiBlitz 1.7.1

konigra, 2022-02-05:
He stevewebb951,

1. a DEBUG report would help. Note that the report might have sensitive information like IPs and node IDs.

Here is described how to get one:

https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#how-do-i-generate-a-debug-report

2. The final sync can take some time indeed. It is even in the FAQ.
(do not delete or reset your HDD without consulsting somebody more experienced than me)

https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#why-is-my-final-sync-taking-so-long

3. What happens when you Exit to the terminal and then go into the menu with the "menu" command? Are you able to navigate there? Can I have a screenshot of your versions (Bitcoin Core, LND, C-lightning) ?

Regards,

Koni

stevewebb951, 2022-02-05:
I appreciate your help, Koni.

1. I would like to create a DEBUG report, but I can't get into rasbiblitz. The boot process hangs at the "Node is Syncing" window, with the messages I quoted in my original post. When I exit to the terminal, I get this message: "client_loop: send disconnect: Broken pipe" and then it goes to a normal Mac terminal prompt.

2. I did get to the final sync last week. Yeah, it took a LONG time. But that is not the problem this time. My raspiblitz is in a hung state.

3. When I CTRL-c to exit, I get this message: "client_loop: send disconnect: Broken pipe" and then it goes to a normal Mac terminal prompt. Would love to screenshot the versions, but since I can't get into the menus, not sure how to get them for you.

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.1
chainnetwork: bitcoin / main
 17:50:51 up 1 day, 11:54,  2 users,  load average: 0.25, 0.44, 0.38

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin-Daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: resources) since Sat 2022-02-05 17:50:26 GMT; 24s ago

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sat 2022-02-05 07:26:53 GMT, end at Sat 2022-02-05 17:50:51 GMT. --
Feb 05 17:49:56 raspberrypi systemd[1]: bitcoind.service: Failed with result 'resources'.
Feb 05 17:49:56 raspberrypi systemd[1]: Failed to start Bitcoin-Daemon.
Feb 05 17:50:26 raspberrypi systemd[1]: bitcoind.service: Service RestartSec=30s expired, scheduling restart.
Feb 05 17:50:26 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 4273.
Feb 05 17:50:26 raspberrypi systemd[1]: Stopped Bitcoin-Daemon.
Feb 05 17:50:26 raspberrypi systemd[1]: bitcoind.service: Failed to run 'start-pre' task: No such file or directory
Feb 05 17:50:26 raspberrypi systemd[1]: bitcoind.service: Failed with result 'resources'.
Feb 05 17:50:26 raspberrypi systemd[1]: Failed to start Bitcoin-Daemon.
1644083445 ERROR: Error found in Logs
1644083446 ERROR: Error found in Logs

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: resources) since Sat 2022-02-05 17:49:56 GMT; 54s ago
    Tasks: 0 (limit: 4532)
   CGroup: /system.slice/lnd.service

*** LAST LND (MAINNET) ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Sat 2022-02-05 07:26:53 GMT, end at Sat 2022-02-05 17:50:51 GMT. --
Feb 05 17:48:55 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Feb 05 17:48:55 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 2135.
Feb 05 17:48:55 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Feb 05 17:48:56 raspberrypi systemd[1]: lnd.service: Failed to run 'start-pre' task: No such file or directory
Feb 05 17:48:56 raspberrypi systemd[1]: lnd.service: Failed with result 'resources'.
Feb 05 17:48:56 raspberrypi systemd[1]: Failed to start LND Lightning Daemon.
Feb 05 17:49:56 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Feb 05 17:49:56 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 2136.
Feb 05 17:49:56 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Feb 05 17:49:56 raspberrypi systemd[1]: lnd.service: Failed to run 'start-pre' task: No such file or directory
Feb 05 17:49:56 raspberrypi systemd[1]: lnd.service: Failed with result 'resources'.
Feb 05 17:49:56 raspberrypi systemd[1]: Failed to start LND Lightning Daemon.

*** LAST 30 LND (MAINNET) INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
[... 190 lines cut ...]
network_device=eth0

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.##.##'
tempCelsius='44.3'
uptime=129244
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
clActive=0
CLwalletLocked=0
blitzTUIActive=0
blitzTUIRestarts=0
vagrant=0
scriptRuntime=0

Koni, let me know if you need any other info. I'm really new at all of this, so I apologize in advance for noob questions.

konigra, 2022-02-06:
Thank you for the logs and don´t worry about the questions, questions are good!

So, it seems to me that it is an issue with your Mac!

What might solve it:

1.Go to your ~/.ssh/config file on your Mac and look for 

"TCPKeepAlive no"

2. Change that "no" to a "yes".

3. This should fix the "client_loop: send disconnect: Broken pipe" problem.

Explanation:

TCPKeepAlive no means "do not send keepalive messages to the server". When the opposite, TCPKeepAlive yes, is set, then the client sends keepalive messages to the server and requires a response in order to maintain its end of the connection. This will detect if the server goes down, reboots, etc. The trouble with this is that if the connection between the client and server is broken for a brief period of time (due to flaky a network connection), this will cause the keepalive messages to fail, and the client will end the connection with "broken pipe".

Setting TCPKeepAlive no tells the client to just assume the connection is still good until proven otherwise by a user request, meaning that temporary connection breakages while your ssh term is sitting idle in the background won't kill the connection.

_**Source:

https://unix.stackexchange.com/questions/602518/ssh-connection-client-loop-send-disconnect-broken-pipe-or-connection-reset**_

Let me know if that fixed it.

Found it here:

https://raspberrypi.stackexchange.com/questions/111265/what-causes-raspberry-pi-to-reliably-drop-ssh-connections-client-loop-send-di
______________________________________________________________________________________________________________

Next thing:

You could think about to just update your Node by flashing your SD card with the latest RaspiBlitz update. Flashing your SD card won´t harm your funds as far as I know (they´re are saved on your SDD, as well as the bitcoin blockchain).
I am doing it right now, too, step by step.

stevewebb951, 2022-02-06:
Thanks for the instructions, Koni. I had to figure out how to get to the ~/.ssh/config file. Turns out it did not exist, so I created it.

Flashed the SD card. Now I'm getting "Connection refused" when I try to ssh into the pi. Thought maybe the flashing process didn't go right, so reflashed. Still no joy. 

So now I'm in worse shape than before. Can't even get into the pi. BTW, I can ping it.

Any suggestions?

stevewebb951, 2022-02-07:
Found a tool called Raspberry Pi Imager here: https://www.raspberrypi.com/software/

Was able to flash the SD card yet again with this tool. It has a nice UI for noobs like me. It's in the process of installing/recovering my previous raspiblitz account. 

I'll report back with results.

stevewebb951, 2022-02-07:
Got everything back and running. Raspberry Pi Imager really saved the day.

Koni, I appreciate your efforts. Thank you!

closed 2022-02-09 by openoms

mamar59, 2022-03-19:
Hello I found my node today with a problem that I don't understand because I haven't been affected for a few days, do you have an idea?
[image]

https://l.termbin.com/zk2f

mamar59, 2022-03-20:
[image]
hello after resetting my sd, I still have the same problem. I have to launch bitcoind and lnd on the command line to start my node, but that's not what happened

konigra, 2022-03-20:
hi mamar59,

please create a debug report and share it with me:

https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#how-do-i-generate-a-debug-report

Is your raspiblitz properly connected to the internet?
try to write in the command line:
ping www.google.com

to see if it is connected to the internet.

since it is not syncing the blockchain, I doubt that there is a stable internet connection.

mamar59, 2022-03-20:
I just changed my sd card with a new one and the problem seems to be fixed but I have the impression that it takes me time to synchronize with the blockchain thank you to you :)

JDarock, 2023-02-23:
Hi, I'm facing similar issue. I've just yesterday flashed again the SD card and from the begining my Raspiblitz is runing wothout any issues just with the LND installed. But then I started installing other apps as BTCpay Server, Electrum Rust and others and don't remember the exacts moment when after one of the resratrs it stoped syncing blockchain and the blochain progress got stuck and just continuously restarting without an result. Thank you

Here is my debug:
https://l.termbin.com/h4w8

mamar59, 2023-02-24:
hi, have you tried disabling the Electrum Rust? I've had a problem like this before.

JDarock, 2023-02-24:
> hi, have you tried disabling the Electrum Rust? I've had a problem like this before.

Hi, actually I was facing that issue even before I did install Electrum Rust. But thank you

fapmachine-eth, 2023-10-05:
Hi all,

I'm having the same issue.

Here is my debug report: https://l.termbin.com/pwgt

My node is continually stuck on this screen :
[image]

rootzoll, 2023-10-05:
Would need a longer tail of the bitcoin log - can you try & post: `sudo tail -n 500 /mnt/hdd/bitcoin/debug.log`

fapmachine-eth, 2023-10-06:
2023-10-06T02:46:29Z scheduler thread exit
2023-10-06T02:46:30Z Shutdown: done

2023-10-06T02:46:31Z Bitcoin Core version v25.0.0 (release build)
2023-10-06T02:46:31Z InitParameterInteraction: parameter interaction: -bind set -> setting -listen=1
2023-10-06T02:46:31Z InitParameterInteraction: parameter interaction: -proxy set -> setting -upnp=0
2023-10-06T02:46:31Z InitParameterInteraction: parameter interaction: -proxy set -> setting -natpmp=0
2023-10-06T02:46:31Z InitParameterInteraction: parameter interaction: -proxy set -> setting -discover=0
2023-10-06T02:46:31Z Using the 'standard' SHA256 implementation
2023-10-06T02:46:31Z Default data directory /home/bitcoin/.bitcoin
2023-10-06T02:46:31Z Using data directory /mnt/hdd/bitcoin
2023-10-06T02:46:31Z Config file: /mnt/hdd/bitcoin/bitcoin.conf
2023-10-06T02:46:31Z Config file arg: daemon="1"
2023-10-06T02:46:31Z Config file arg: datadir="/mnt/hdd/bitcoin"
2023-10-06T02:46:31Z Config file arg: dbcache="512"
2023-10-06T02:46:31Z Config file arg: deprecatedrpc="addresses"
2023-10-06T02:46:31Z Config file arg: disablewallet="0"
2023-10-06T02:46:31Z Config file arg: dns="0"
2023-10-06T02:46:31Z Config file arg: dnsseed="0"
2023-10-06T02:46:31Z Config file arg: maxconnections="40"
2023-10-06T02:46:31Z Config file arg: maxmempool="300"
2023-10-06T02:46:31Z Config file arg: maxorphantx="10"
2023-10-06T02:46:31Z Config file arg: maxuploadtarget="5000"
2023-10-06T02:46:31Z Config file arg: onlynet="onion"
2023-10-06T02:46:31Z Config file arg: peerbloomfilters="1"
2023-10-06T02:46:31Z Config file arg: proxy="127.0.0.1:9050"
2023-10-06T02:46:31Z Config file arg: rpcallowip="127.0.0.1"
2023-10-06T02:46:31Z Config file arg: rpcpassword=****
2023-10-06T02:46:31Z Config file arg: rpcuser=****
2023-10-06T02:46:31Z Config file arg: server="1"
2023-10-06T02:46:31Z Config file arg: testnet="0"
2023-10-06T02:46:31Z Config file arg: txindex="1"
2023-10-06T02:46:31Z Config file arg: whitelist="download@127.0.0.1"
2023-10-06T02:46:31Z Config file arg: [main] bind="127.0.0.1"
2023-10-06T02:46:31Z Config file arg: [main] debuglogfile="/mnt/hdd/bitcoin/debug.log"
2023-10-06T02:46:31Z Config file arg: [main] rpcbind="127.0.0.1:8332"
2023-10-06T02:46:31Z Config file arg: [main] rpcport="8332"
2023-10-06T02:46:31Z Config file arg: [main] wallet="wallet.dat"
2023-10-06T02:46:31Z Config file arg: [main] zmqpubrawblock="tcp://127.0.0.1:28332"
2023-10-06T02:46:31Z Config file arg: [main] zmqpubrawtx="tcp://127.0.0.1:28333"
2023-10-06T02:46:31Z Config file arg: [test] bind="127.0.0.1"
2023-10-06T02:46:31Z Config file arg: [test] rpcbind="127.0.0.1:18332"
2023-10-06T02:46:31Z Config file arg: [test] rpcport="18332"
2023-10-06T02:46:31Z Command-line arg: conf="/mnt/hdd/bitcoin/bitcoin.conf"
2023-10-06T02:46:31Z Command-line arg: daemonwait=""
2023-10-06T02:46:31Z Command-line arg: datadir="/mnt/hdd/bitcoin"
2023-10-06T02:46:31Z Using at most 40 automatic connections (1024 file descriptors available)
2023-10-06T02:46:31Z Using 16 MiB out of 16 MiB requested for signature cache, able to store 524288 elements
2023-10-06T02:46:31Z Using 16 MiB out of 16 MiB requested for script execution cache, able to store 524288 elements
2023-10-06T02:46:31Z Script verification uses 3 additional threads
2023-10-06T02:46:31Z scheduler thread start
2023-10-06T02:46:31Z Binding RPC on address 127.0.0.1 port 8332
2023-10-06T02:46:31Z [http] creating work queue of depth 16
2023-10-06T02:46:31Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2023-10-06T02:46:31Z [http] starting 4 worker threads
2023-10-06T02:46:31Z Using wallet directory /mnt/hdd/bitcoin
2023-10-06T02:46:31Z init message: Verifying wallet(s)…
2023-10-06T02:46:31Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2023-10-06T02:46:31Z Using wallet /mnt/hdd/bitcoin/wallet.dat/wallet.dat
2023-10-06T02:46:31Z BerkeleyEnvironment::Open: LogDir=/mnt/hdd/bitcoin/wallet.dat/database ErrorFile=/mnt/hdd/bitcoin/wallet.dat/db.log
2023-10-06T02:46:31Z Using /16 prefix for IP bucketing
2023-10-06T02:46:31Z init message: Loading P2P addresses…
2023-10-06T02:46:31Z Loaded 15445 addresses from peers.dat  225ms
2023-10-06T02:46:31Z init message: Loading banlist…
2023-10-06T02:46:31Z SetNetworkActive: true
2023-10-06T02:46:31Z Cache configuration:
2023-10-06T02:46:31Z * Using 2.0 MiB for block index database
2023-10-06T02:46:31Z * Using 63.8 MiB for transaction index database
2023-10-06T02:46:31Z * Using 8.0 MiB for chain state database
2023-10-06T02:46:31Z * Using 438.2 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
[... 385 lines cut ...]
2023-10-06T02:47:46Z Binding RPC on address 127.0.0.1 port 8332
2023-10-06T02:47:46Z [http] creating work queue of depth 16
2023-10-06T02:47:46Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2023-10-06T02:47:46Z [http] starting 4 worker threads
2023-10-06T02:47:46Z Using wallet directory /mnt/hdd/bitcoin
2023-10-06T02:47:46Z init message: Verifying wallet(s)…
2023-10-06T02:47:46Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2023-10-06T02:47:46Z Using wallet /mnt/hdd/bitcoin/wallet.dat/wallet.dat
[... cut ...]
2023-10-06T02:47:47Z init message: Loading block index…
2023-10-06T02:47:47Z Assuming ancestors of block 000000000000000000035c3f0d31e71a5ee24c5aaf3354689f65bd7b07dee632 have valid signatures.
2023-10-06T02:47:47Z Setting nMinimumChainWork=000000000000000000000000000000000000000044a50fe819c39ad624021859
2023-10-06T02:47:47Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2023-10-06T02:47:47Z Opened LevelDB successfully
2023-10-06T02:47:47Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000

fapmachine-eth, 2023-10-06:
2023-10-06T07:05:22Z init message: Loading block index…
2023-10-06T07:05:22Z Assuming ancestors of block 00000000000000000009c97098b5295f7e5f183ac811fb5d1534040adb93cabd have valid signatures.
2023-10-06T07:05:22Z Setting nMinimumChainWork=00000000000000000000000000000000000000003404ba0801921119f903495e
2023-10-06T07:05:22Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2023-10-06T07:05:22Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2023-10-06T07:05:22Z Opened LevelDB successfully
2023-10-06T07:05:22Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2023-10-06T07:05:33Z LoadBlockIndexDB: last block file = 3849
2023-10-06T07:05:33Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=43, size=83036004, heights=809664...809706, time=2023-09-28...2023-09-28)
2023-10-06T07:05:33Z Checking all blk files are present...
2023-10-06T07:05:36Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2023-10-06T07:05:36Z Opened LevelDB successfully
2023-10-06T07:05:36Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: f7ea420faa39a63e
2023-10-06T07:05:37Z Loaded best chain: hashBestChain=000000000000000000018f8e2ebc08b2c6c8b5eb120a388707b186ed909f8bf8 height=809701 date=2023-09-28T07:56:27Z progress=0.997770
2023-10-06T07:05:37Z init message: Verifying blocks…
2023-10-06T07:05:37Z Verifying last 6 blocks at level 3
2023-10-06T07:05:37Z [0%]...LevelDB read failure: Corruption: block checksum mismatch: /mnt/hdd/bitcoin/chainstate/355260.ldb
2023-10-06T07:05:38Z Fatal LevelDB error: Corruption: block checksum mismatch: /mnt/hdd/bitcoin/chainstate/355260.ldb
2023-10-06T07:05:38Z You can use -debug=leveldb to get more complete diagnostic messages
2023-10-06T07:05:38Z Fatal LevelDB error: Corruption: block checksum mismatch: /mnt/hdd/bitcoin/chainstate/355260.ldb
2023-10-06T07:05:38Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2023-10-06T07:05:38Z Aborted block database rebuild. Exiting.
2023-10-06T07:05:38Z Shutdown: In progress...
2023-10-06T07:05:38Z scheduler thread exit
2023-10-06T07:05:38Z Shutdown: done

2023-10-06T07:05:39Z Bitcoin Core version v24.0.1 (release build)
2023-10-06T07:05:39Z InitParameterInteraction: parameter interaction: -bind set -> setting -listen=1
2023-10-06T07:05:39Z InitParameterInteraction: parameter interaction: -proxy set -> setting -upnp=0
2023-10-06T07:05:39Z InitParameterInteraction: parameter interaction: -proxy set -> setting -natpmp=0
2023-10-06T07:05:39Z InitParameterInteraction: parameter interaction: -proxy set -> setting -discover=0
2023-10-06T07:05:39Z Using the 'standard' SHA256 implementation
2023-10-06T07:05:39Z Default data directory /home/bitcoin/.bitcoin
2023-10-06T07:05:39Z Using data directory /mnt/hdd/bitcoin
2023-10-06T07:05:39Z Config file: /mnt/hdd/bitcoin/bitcoin.conf
2023-10-06T07:05:39Z Config file arg: daemon="1"
2023-10-06T07:05:39Z Config file arg: datadir="/mnt/hdd/bitcoin"
2023-10-06T07:05:39Z Config file arg: dbcache="512"
2023-10-06T07:05:39Z Config file arg: deprecatedrpc="addresses"
2023-10-06T07:05:39Z Config file arg: disablewallet="0"
2023-10-06T07:05:39Z Config file arg: dns="0"
2023-10-06T07:05:39Z Config file arg: dnsseed="0"
2023-10-06T07:05:39Z Config file arg: maxconnections="40"
2023-10-06T07:05:39Z Config file arg: maxmempool="300"
2023-10-06T07:05:39Z Config file arg: maxorphantx="10"
2023-10-06T07:05:39Z Config file arg: maxuploadtarget="5000"
2023-10-06T07:05:39Z Config file arg: onlynet="onion"
2023-10-06T07:05:39Z Config file arg: peerbloomfilters="1"
2023-10-06T07:05:39Z Config file arg: proxy="127.0.0.1:9050"
2023-10-06T07:05:39Z Config file arg: rpcallowip="127.0.0.1"
2023-10-06T07:05:39Z Config file arg: rpcpassword=****
2023-10-06T07:05:39Z Config file arg: rpcuser=****
2023-10-06T07:05:39Z Config file arg: server="1"
2023-10-06T07:05:39Z Config file arg: testnet="0"
2023-10-06T07:05:39Z Config file arg: txindex="1"
2023-10-06T07:05:39Z Config file arg: whitelist="download@127.0.0.1"
2023-10-06T07:05:39Z Config file arg: [main] bind="127.0.0.1"
2023-10-06T07:05:39Z Config file arg: [main] debuglogfile="/mnt/hdd/bitcoin/debug.log"
2023-10-06T07:05:39Z Config file arg: [main] rpcbind=****
2023-10-06T07:05:39Z Config file arg: [main] rpcport="8332"
2023-10-06T07:05:39Z Config file arg: [main] wallet="wallet.dat"
2023-10-06T07:05:39Z Config file arg: [main] zmqpubrawblock="tcp://127.0.0.1:28332"
2023-10-06T07:05:39Z Config file arg: [main] zmqpubrawtx="tcp://127.0.0.1:28333"
2023-10-06T07:05:39Z Config file arg: [test] bind="127.0.0.1"
2023-10-06T07:05:39Z Config file arg: [test] rpcbind=****
2023-10-06T07:05:39Z Config file arg: [test] rpcport="18332"
2023-10-06T07:05:39Z Command-line arg: conf="/mnt/hdd/bitcoin/bitcoin.conf"
2023-10-06T07:05:39Z Command-line arg: daemonwait=""
2023-10-06T07:05:39Z Command-line arg: datadir="/mnt/hdd/bitcoin"
[... 385 lines cut ...]
2023-10-06T07:06:50Z Using wallet directory /mnt/hdd/bitcoin
2023-10-06T07:06:50Z init message: Verifying wallet(s)…
2023-10-06T07:06:50Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2023-10-06T07:06:50Z Using wallet /mnt/hdd/bitcoin/wallet.dat/wallet.dat
2023-10-06T07:06:50Z BerkeleyEnvironment::Open: LogDir=/mnt/hdd/bitcoin/wallet.dat/database ErrorFile=/mnt/hdd/bitcoin/wallet.dat/db.log
2023-10-06T07:06:50Z Using /16 prefix for IP bucketing
2023-10-06T07:06:50Z init message: Loading P2P addresses…
[... cut ...]
2023-10-06T07:06:50Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2023-10-06T07:06:50Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2023-10-06T07:06:50Z Opened LevelDB successfully
2023-10-06T07:06:50Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2023-10-06T07:07:02Z LoadBlockIndexDB: last block file = 3849
2023-10-06T07:07:02Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=43, size=83036004, heights=809664...809706, time=2023-09-28...2023-09-28)
2023-10-06T07:07:02Z Checking all blk files are present...

jorge-at-v, 2023-11-16:
Hello,
The issue with my setup is the LTN node not progressing (it's been around 92% for 2 weeks). BTC is complete. YHere's my debug report:
https://l.termbin.com/xgnk

Access to Menu is good but I haven't want to do any of the actions without some guidance :)
I appreciate any help you can offer !

rootzoll, 2023-11-22:
Hmmm .. while your bitcoind seems runnign fine your CLN says`lightningd: The Bitcoin backend died.`
@openoms any idea? should this handled in a seperate new issue?

```
2023-11-09T13:50:49.426Z UNUSUAL plugin-bcli: bitcoin-cli: finished bitcoin-cli getmempoolinfo (900136 ms)
2023-11-09T13:50:49.628Z UNUSUAL lightningd: Unable to estimate any fees
2023-11-09T14:04:44.298Z UNUSUAL plugin-bcli: bitcoin-cli: finished bitcoin-cli getblock 0000000000000000000720ba059104ac05e1a63e00f5c133efd764ba161c532c 0 (900132 ms)
2023-11-09T14:04:44.594Z **BROKEN** plugin-bcli: bitcoin-cli getblock 0000000000000000000720ba059104ac05e1a63e00f5c133efd764ba161c532c 0 exited 1 (after 2 other errors) 'error: timeout on transient error: Could not connect to the server *.*.*.*:8332 (error code 0 - \"timeout reached\")\n\nMake sure the bitcoind server is running and that you are connecting to the correct RPC port.\n'; we have been retrying command for --bitcoin-retry-timeout=60 seconds; bitcoind setup or our --bitcoin-* configs broken?
2023-11-09T14:04:44.603Z INFO    plugin-bcli: Killing plugin: exited during normal operation
2023-11-09T14:04:44.631Z **BROKEN** lightningd: The Bitcoin backend died.
2023-11-09T14:04:45.917Z **BROKEN** lightningd: FATAL SIGNAL 6 (version v23.08.1)
```

jorge-at-v, 2023-12-02:
Thanks @rootzoll 
Unfortunately, I'm still at the same state. I'll create a new issue in a couple of days if necessary.

@openoms

openoms, 2023-12-02:
It is likely data corruption on your disk.
The block https://mempool.space/block/0000000000000000000720ba059104ac05e1a63e00f5c133efd764ba161c532c lightningd tries to get is from March 2022.
The containing block file is likely corrupted so bitcoind doesn't serve it.

Could test this by connecting CLN to another bitcoin node.

The likely solution is to redownload the blockchain - reset the blockchain in the REPAIR menu.

jorge-at-v, 2023-12-02:
alrighty. I'm on it !
I'll see if it returns an error, else I'll let you know how the repair goes around Xmas :)
Thanks !