raspiblitz/raspiblitz #2934: Dirty update to dev or v1.7.2

openoms, 2022-02-07:
Having questions about errors after patching from v1.7.1 to dev (or v1.7.2 later when it is merged to the v1.7 branch)

This continues to be not the intended update method and if running into issues the quickest solution will be to reflash the SDcard with the latest image.

Some setups are salvageable so will make notes here:

To fix redis in `admin`:
```
sudo apt install redis-server
cd
sudo ./_background.scan.sh install
sudo ./_background.scan.sh
restart
```

openoms, 2022-02-22:
Also a good time to run a full:
```
sudo apt update
sudo apt upgrade -y
```

openoms, 2022-03-16:
To bring everything up-to-date (summarizing together with the above):

```
sudo apt update
sudo apt upgrade -y

python_libs="grpcio==1.42.0 googleapis-common-protos==1.53.0 toml==0.10.2 j2cli==0.3.10 requests[socks]==2.21.0"
torbox_libs="pytesseract mechanize PySocks urwid Pillow requests setuptools"
sudo -H python3 -m pip install --upgrade pip
sudo -H python3 -m pip install ${python_libs} ${torbox_libs}

sudo apt install redis-server
cd
sudo ./_background.scan.sh install
sudo ./_background.scan.sh
restart
```

cstenglein, 2023-06-07:
We are on 1.9.0 already => closing

closed 2023-06-07 by cstenglein