raspiblitz/raspiblitz #2939: lnd[26720]: unable to create server: alias too large: max is 32, got 35

mineralf, 2022-02-08:
It is easily possible to create a node name in the raspiblitz menu which caused my blitz to end up in a endless LND restart loop.

**Proposed fix: Restricting the input field in the "UI" to those 32 characters would be great**

This is how to reproduce and fix it then manually (for anybody stupid enough to try long names like me):

`Main Menu -> LND Wallet Options -> NAME : SomeLongNameWith35CharactersCausedCrash
`
After the restart the status screen hangs in an endless restart of the LND service

Investigating a bit: 

```
cat systemd.lightning.log

1644316926 STARTED
1644316937 ERROR: [lncli] rpc error: code = Unknown desc = the RPC server is in the process of starting up, but not yet ready to accept calls
1644316938 ERROR: LND RPC is not responding. LND may have problems starting up. Check logs, config files and systemd service. Org-Error: [lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
```

```
sudo journalctl -u lnd
Feb 08 10:05:57 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Feb 08 10:05:57 raspberrypi systemd[1]: Started LND Lightning Daemon.
Feb 08 10:06:12 raspberrypi **lnd[26720]: unable to create server: alias too large: max is 32, got 35**
Feb 08 10:06:12 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Feb 08 10:06:12 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Feb 08 10:07:12 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Feb 08 10:07:12 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 6.
Feb 08 10:07:12 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
```

The "LND Wallet Options" Menu does not work anymore once LND fails to start - so you need to find the config file and shorten the name there:

```
sudo vi /home/bitcoin/.lnd/lnd.conf
<insert mode>
alias=<NewShorterName>
:wq
```

Node is up and running again 😅

rootzoll, 2022-02-08:
Good idea. Will push this to the 1.8 release, because thats also an reminder to then limit the name length for the WebUI that targeted for v1.8.

rootzoll, 2022-05-09:
@cstenglein do we have this max 32 chars limit for node name already implemented in the WebUI?

cstenglein, 2022-05-11:
I have added it in https://github.com/cstenglein/raspiblitz-web/pull/284

rootzoll, 2022-05-17:
OK part of v1.8 release. Closing issue.

closed 2022-05-17 by rootzoll