raspiblitz/raspiblitz #2940: [1.7.2rc2] apt source to torproject.org:

GoetzGoerisch, 2022-02-08:
On `1.7.2rc2` image with patch to latest `dev` with commit 4b7f8892f18d02f432950e1bd377752d97b81e8b:

```
The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
Reading package lists... Done
W: GPG error: tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
E: The repository 'tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

**Probable cause:**

`/etc/apt/trusted.gpg.d/torproject.gpg` is empty

GoetzGoerisch, 2022-02-08:
from my initial debugging:

This would might not run probably: https://github.com/rootzoll/raspiblitz/blob/4b7f8892f18d02f432950e1bd377752d97b81e8b/home.admin/config.scripts/tor.install.sh#L144

Manually running `sudo ./tor.install.sh install` fixes it.

openoms, 2022-02-09:
@GoetzGoerisch where do you have this exaclty?

I can't reproduce so far in v1.7.2rc2:
```
₿ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease                              
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease       
Get:4 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]            
Hit:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease
Fetched 39.4 kB in 3s (12.1 kB/s)                          
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

GoetzGoerisch, 2022-02-09:
This was image `v1.7.2.rc2` not the new `v1.7.2rc2b` from yesterday.

Setup a new node, wihile keeping the blockchain from a former myNode test on this drive.
After it finished with the setup and rebooted and started syncing the blockchain, I ran `sudo apt update.`

After looking through `raspiblitz.setup.log` the error occured already there:

```text
###################################
# _provision_.sh
###################################
### BASIC SYSTEM SETTINGS ###
# Optimizing log files: rotate daily, keep 2 weeks & compress old days 
Copy HDD content for user admin
mkdir: cannot create directory ‘/home/admin/.bitcoin’: File exists
cp: cannot stat '/mnt/hdd/lnd/lnd.conf': No such file or directory
cp: cannot stat '/mnt/hdd/lnd/tls.cert': No such file or directory
cp: cannot stat '/mnt/hdd/lnd/data/chain': No such file or directory

### RUNNING PROVISIONING SERVICES ###
Provisioning BLITZ WEB SERVICE - run config script
Turning ON: Web
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:4 http://deb.debian.org/debian bullseye-updates InRelease
Get:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease [3,526 B]
Err:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
Reading package lists...
W: GPG error: tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
E: The repository 'tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease' is not signed.
Reading package lists...
Building dependency tree...
Reading state information...
--8< ---
```

GoetzGoerisch, 2022-02-09:
As written above, no.
> Manually running `sudo ./tor.install.sh` install fixes it.

Just flashed `v1.7.2rc2b` and will now run the migration.