raspiblitz/raspiblitz #2946: Impossible to compact channel.db for possible corruption

CommanderPoe, 2022-02-10:
Background: This afternoon, I tried compacting my `channel.db` which at the moment is nearly **14GB** . Followed the steps of this new [script](https://github.com/rootzoll/raspiblitz/pull/2761/commits/447a7419cd39ec0520a0c772a2aaaee3e647fea9) but I constantly started getting this error:

```
sudo -u bitcoin /usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf --db.bolt.auto-compact
2022-02-10 19:50:14.664 [INF] LTND: Version: 0.13.3-beta commit=v0.13.3-beta, build=production, logging=default, debuglevel=debug
2022-02-10 19:50:14.665 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2022-02-10 19:50:14.706 [DBG] VRPC: Versioner RPC server successfully registered with root gRPC server
2022-02-10 19:50:14.706 [DBG] WLKT: WalletKit RPC server successfully registered with root gRPC server
2022-02-10 19:50:14.706 [DBG] IRPC: Invoices RPC server successfully registered with root gRPC server
2022-02-10 19:50:14.706 [DBG] RRPC: Router RPC server successfully register with root gRPC server
2022-02-10 19:50:14.706 [DBG] SGNR: Signer RPC server successfully register with root gRPC server
2022-02-10 19:50:14.706 [DBG] ARPC: Autopilot RPC server successfully register with root gRPC server
2022-02-10 19:50:14.706 [DBG] NTFR: ChainNotifier RPC server successfully register with root gRPC server
2022-02-10 19:50:14.707 [INF] RPCS: RPC server listening on 0.0.0.0:10009
2022-02-10 19:50:14.712 [DBG] VRPC: Versioner REST server successfully registered with root REST server
2022-02-10 19:50:14.713 [DBG] WLKT: WalletKit REST server successfully registered with root REST server
2022-02-10 19:50:14.715 [DBG] IRPC: Invoices REST server successfully registered with root REST server
2022-02-10 19:50:14.715 [DBG] RRPC: Router REST server successfully registered with root REST server
2022-02-10 19:50:14.716 [DBG] SGNR: Signer REST server successfully registered with root REST server
2022-02-10 19:50:14.716 [DBG] ARPC: Autopilot REST server successfully registered with root REST server
2022-02-10 19:50:14.716 [DBG] NTFR: ChainNotifier REST server successfully registered with root REST server
2022-02-10 19:50:14.716 [INF] RPCS: gRPC proxy started at 0.0.0.0:8080
2022-02-10 19:50:14.717 [INF] LTND: Opening the main database, this might take a few minutes...
2022-02-10 19:50:14.717 [INF] LTND: Opening bbolt database, sync_freelist=true, auto_compact=true
2022-02-10 19:50:14.724 [INF] CHDB: Not compacting database file at /home/bitcoin/.lnd/data/graph/mainnet/channel.db, it was last compacted at 2022-02-10 11:45:07.300762972 +0000 GMT (8h5m7s ago), min age is set to 168h0m0s
2022-02-10 19:50:14.745 [INF] CHDB: Checking for schema update: latest_version=22, db_version=22
2022-02-10 19:50:14.745 [INF] LTND: Database now open (time_to_open=28.577768ms)!
2022-02-10 19:50:14.745 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2022-02-10 19:50:54.462 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-02-10 19:50:54.685 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-02-10 19:50:59.687 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-02-10 19:51:04.693 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
...
^C2022-02-10 19:52:04.508 [INF] LTND: Received interrupt
2022-02-10 19:52:04.508 [INF] LTND: Shutting down...
2022-02-10 19:52:04.508 [INF] LTND: Gracefully shutting down.
2022-02-10 19:52:04.508 [ERR] LTND: unable to set up wallet password listeners: shutting down
2022-02-10 19:52:04.515 [INF] LTND: Shutdown complete

unable to set up wallet password listeners: shutting down
```

I then tried to unlock the wallet with `lncli unlock` and got this error which I have been having since I first started with this node some years ago and never let me unlock the wallet unless I reboot the node and start from scratch:

```
[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
```

I started searching among previous issues and this [one](https://github.com/rootzoll/raspiblitz/issues/2532) seems particularly similar and Im afraid that I have been running this node in `panic mode` all these years. Can anyone confirm if that's the reason and if there's something different than just a start a fresh new node?

openoms, 2022-02-11:
Can you share your lnd.conf (minus passwords)?

CommanderPoe, 2022-02-13:
Hey open, thanks for that. Will add that line. Isnt that line supposed to be in the `.conf` by default? 

```
# bitcoind configuration
# mainnet/testnet
testnet=0
# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=0
# Connection settings
rpcuser=XXXXXXX
rpcpassword=XXXXXX
main.rpcport=8332
rpcallowip=127.0.0.1
main.rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
# Raspberry Pi optimizations
dbcache=1024
maxorphantx=10
maxmempool=300
maxconnections=40
maxuploadtarget=5000
datadir=/mnt/hdd/bitcoin
test.rpcport=18332
test.rpcbind=127.0.0.1:18332
peerbloomfilters=1
main.wallet=wallet.dat
deprecatedrpc=addresses
onlynet=onion
proxy=127.0.0.1:9050
main.bind=127.0.0.1
test.bind=127.0.0.1
dnsseed=0
dns=0
db.bolt.auto-compact=true
```

openoms, 2022-02-13:
Yes, see the default with the Tor settings:
https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/assets/lnd.bitcoin.conf

The other problem is that your headers are missing, they need to be specified like:
```
[Application Options]

[Bitcoin]
```
etc.

Use 
https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf
and
https://www.lightningnode.info/advanced-tools/lnd.conf
as reference.

CommanderPoe, 2022-02-17:
Hey @openoms , sorry for getting back to this same 'issue' 

I was gonna close it but i realized i shared the wrong `.conf`  .... its the bitcoin.conf 

Here's the LND, I keep having the same error when trying to unlock wallet and I do have the line you told me about. have a look please.

```
# lnd configuration

[Application Options]
listen=0.0.0.0:9735
tlsdisableautofill=1
tlsautorefresh=1
stagger-initial-reconnect=1
sync-freelist=1
ignore-historical-gossip-filters=1
debuglevel=debug
maxpendingchannels=5
alias=XXXXXXX
color=#FED0D0
nat=false

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# REST open to all connections on Port 8080
restlisten=0.0.0.0:8080
# Domain, could use https://freedns.afraid.org
tlsextradomain=XXXXXXXXX

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
# bitcoin.testnet=1
bitcoin.mainnet=1

[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.6

# zerotier configs
# tlsextraip=172.X

[bitcoind]
bitcoind.rpchost=127.0.0.1:8332
bitcoind.rpcuser=XXXXXXXX
bitcoind.rpcpass=XXXXXXX
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333

[tor]
tor.active=true
tor.v3=true
tor.streamisolation=true
tor.privatekeypath=/mnt/hdd/lnd/v3_onion_private_key
tor.socks=9050
tor.control=9051
```

Any ideas, thanks & sorry

rootzoll, 2022-02-17:
@openoms is this still an issue regarding the v1.7.2 release?

openoms, 2022-02-17:
your conf seems to be correct 

see the lnd logs is SYSTEM - LNDLOGS
and
`sudo journalctl -u lnd -n 50`

for errors'

openoms, 2022-02-17:
@rootzoll I suspect this is a unique problem not to be solved in v1.7.2.

@CommanderPoe wha tversion are you running on?