raspiblitz/raspiblitz #2949: 1.7.2rc2 double slash used with sed

cstenglein, 2022-02-11:
Log when upgrading to 1.7.2rc2:

```
Provisioning C-Lightning Binary - not active
Provisioning CL Mainnet - not active as secondary option
Provisioning CL Testnet - not active
Provisioning CL Signet - not active
Provisioning TOR - run config script
# switching Tor ON
*** Changing bitcoin Config ***
Adding Tor config to the the bitcoin.conf ...
onlynet=onion
proxy=127.0.0.1:9050
main.bind=127.0.0.1
test.bind=127.0.0.1
dnsseed=0
dns=0
sed: can't read /mnt/hdd/app-data/tor//torrc.d/services: No such file or directory

# Hidden Service for web80
HiddenServiceDir /mnt/hdd/tor/web80
HiddenServiceVersion 3
```

Note the "double slashes" between tor and torrc.d

rootzoll, 2022-02-12:
That error is a bit mysterious to me ... because it seems to come from this call:
https://github.com/rootzoll/raspiblitz/blob/fb6f1a243c81674f026a1bedb2a8bfd73a392568/home.admin/config.scripts/tor.network.sh#L115 of `tor.onion-service.sh` ...

... where the path`/mnt/hdd/app-data/tor//torrc.d/services` gets consztructed here: https://github.com/rootzoll/raspiblitz/blob/fb6f1a243c81674f026a1bedb2a8bfd73a392568/home.admin/config.scripts/tor.onion-service.sh#L16 and called with sed later.

But I cannot see how that double slash is slipping in there - this just looks fine. 

How can this:
```
tor_conf_dir="/mnt/hdd/app-data/tor"
torrc_services="${tor_conf_dir}/torrc.d/services"
```

result in this: `/mnt/hdd/app-data/tor//torrc.d/services`

@openoms or @nyxnor any thoughts?

openoms, 2022-02-12:
Fixed here https://github.com/rootzoll/raspiblitz/commit/940b108567f751be10acdd23ed4e0d9283789197

Check if still present in a new install.

closed 2022-02-17 by rootzoll