raspiblitz/raspiblitz #2952: Look into SSD trim again

rootzoll, 2022-02-12:
see https://github.com/rootzoll/raspiblitz/issues/2488#issuecomment-1036473136

cricktor, 2022-02-12:
As far as I understand the topic during my research for a solution, automatic TRIM support might not be activated automatically even if the USB bridge adapter used does have proper UASP support AND does support TRIM properly. So the device provisioning mode the OS chooses may be more on the conservative side. Frankly, I'm no expert here.
So if UASP works properly, TRIM is supported by the USB adapter bridge AND the storage device it is worth a try to activate it. But as Jeff Geerling reported (see full story [https://www.jeffgeerling.com/blog/2020/enabling-trim-on-external-ssd-on-raspberry-pi](https://www.jeffgeerling.com/blog/2020/enabling-trim-on-external-ssd-on-raspberry-pi)) in rare cases things can still go horribly wrong.
I am not sure if there is a reliable automatic scripting way to do this; should be possible but I have no idea how it turns out if a USB bridge adapter misbehaves. SATA and NVMe SSDs (the bare storage devices) should all support TRIM, it's usually the USB adapter bridge that causes trouble if it's firmware doesn't support stuff properly. And the Raspberry Pi and the USB adapter bridge have to "understand" each other and speak the "same language".

cricktor, 2022-02-24:
I can confirm two additional USB3 enclosures to work properly with UASP and TRIM support enabled (based on own testing and procedures as in [my issue #2488 comment](https://github.com/rootzoll/raspiblitz/issues/2488#issuecomment-1036473136)).
The enclosures were purchased mid 2021, current incarnations may have different controllers built-in.

[ICY BOX IB-247-C31 Type-C Enclosure for 2.5inch SATA Drives](https://www.amazon.de/gp/product/B07J2HYF8C/)
Mine has vendor and device ID 2109:0715

[Cable Matters USB 3.1 Type-C Gen2 External SATA SSD Enclosure](https://www.amazon.de/gp/product/B07PBD1K4M/)
Mine has vendor and device ID 174c:235c

I tested also a third enclosure [Inateck Enclosure 2.5inch USB 3.0 metal mesh black SA01002](https://www.amazon.de/gp/product/B07NPGR4CX/): the item title on Amazon claims to have UASP support for this one, but it failed me to get it properly working. With UASP enabled on this one, IO performance is non-existant and maybe errors show up in a scary frequency in the logs. I quickly abandoned further UASP tests to not screw up the content of the used Samsung EVO 860 1TB SSD.
This (very likely) UASP malfunctioning enclosure (with current RaspiOS 64bit) has vendor and device ID 152d:0578. (It might be possible to get the controller working with a firmware update, but I haven't tried.)

I haven't updated yet to RaspiBlitz v1.7.2, still on v1.7.1, need first to see what recent issues have been reported with the new version... ;-)

With properly working UASP enclosures I enforce UASP mode by creating the empty file `/boot/uasp.force`, particular whitelisting in file `/home/admin/config.scripts/blitz.datadrive.sh` might not be necessary at all because IIRC the uasp.force file flag takes precedance.
To make active TRIM mode reboot resiliant, create file `/etc/udev/rules.d/10-trim.rules` with content
```
ACTION=="add|change", ATTRS{idVendor}=="174c", ATTRS{idProduct}=="1153", SUBSYSTEM=="scsi_disk", ATTR{provisioning_mode}="unmap"
ACTION=="add|change", ATTRS{idVendor}=="2109", ATTRS{idProduct}=="0715", SUBSYSTEM=="scsi_disk", ATTR{provisioning_mode}="unmap"
ACTION=="add|change", ATTRS{idVendor}=="174c", ATTRS{idProduct}=="235c", SUBSYSTEM=="scsi_disk", ATTR{provisioning_mode}="unmap"
```
Reboot and confirm UASP and enabled TRIM by following commands, the output of `lsusb -t` should show something like
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
...
```
and the command (as root user) `find /sys/ -name provisioning_mode -exec grep -H . {} + | sort` should spit out something like
```
/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode:unmap
```
If you see `.../provisioning_mode:full` instead of `.../provisioning_mode:unmap` then something is wrong with your udev rules.
 
To enable weekly automatic trimming, execute the command once after enabling UASP for your USB SSD enclosure.
```
$ sudo systemctl enable fstrim.timer
```
Enjoy more efficient UASP protocol and happy TRIMming!

rootzoll, 2022-03-08:
For v1.8 add quickly the vendor ids (see above) and the push to v1.8.1 to take a look at Trim again.

NilByte, 2022-07-15:
For inspiration:

https://github.com/muhkuh-sys/raspberry_enable_trim

I have forced UASP and enabled TRIM for this combination without experiencing any trouble:

https://github.com/rootzoll/raspiblitz/issues/1805#issuecomment-734417492

rootzoll, 2025-05-12:
With v1.12.0 a complete new HDD/SSD/NVMe setup is introduced ... keeping open for further trim reports.