raspiblitz/raspiblitz #2953: Sparko Problem

rootzoll, 2022-02-12:
On my testmachine c-lightning startup us failing in a loop when I have the sparko plugin installed:

```
Feb 11 21:33:01 raspberrypi sh[297834]: lightningd: /home/bitcoin/.lightning/config line 23: sparko-host=0.0.0.0: unrecognized option
```

This also dont fixes it selft after a restart. @openoms

openoms, 2022-02-12:
Couldn't reproduce yet:

Is your lightningd.service the same?

`sudo -u bitcoin /home/admin/config.scripts/cl.check.sh prestart mainnet`
(part of lightningd.service), should clear unless 
sparko=on
is in your `raspiblitz.conf`?

openoms, 2022-02-12:
Actually there is an  error abou the bitcoin  user not being permitted to edit the `raspiblitz.conf`, but that should not stop it from to clear the
/home/bitcoin/.lightning/config from the sparko entries.

openoms, 2022-02-12:
Likely your sparko plugin didi not installor download correctly.

Verify if it's running with:
`cl plugin list`

To install it's newer version run:

```
config.scripts/cl-plugin.sparko.sh off purge
patch
config.scripts/cl-plugin.sparko.sh on
```

rootzoll, 2022-02-13:
Was not able to call `cl plugin list` because c-lightning is not starting up with the broken plugin config. I purged, patched, reinstalled & restarted. But the problem persists.

rootzoll, 2022-02-13:
OK .. I deactivated all `sparko-` lines in the cl config and restarted cl to check on the plugins:
```
admin@192.168.178.61:~ ₿ cl plugin list
{
   "command": "list",
   "plugins": [
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/autoclean",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/bcli",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/fetchinvoice",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/funder",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/topology",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/keysend",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/offers",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/pay",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/txprepare",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/spenderp",
         "active": true
      }
   ]
}
```

So it seems the sparko install fails and thats why cl failes to start not recognizing the new config entries.

rootzoll, 2022-02-13:
So I purged and installed again .. here are the logs - I cannot see any obvious fails:
```
admin@192.168.178.61:~ ₿ config.scripts/cl-plugin.sparko.sh on
# Detect CPU architecture ...
mkdir: cannot create directory ‘/home/bitcoin/cl-plugins-available’: File exists
--2022-02-13 15:21:09--  https://github.com/fiatjaf/sparko/releases/download/v2.8/sparko_linux_arm
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/178970766/89d177f2-066a-4c6c-a14d-c37c3967825a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220213%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220213T152109Z&X-Amz-Expires=300&X-Amz-Signature=6489b9b2128dca0884bdea83064fea70beb7a763a0389cc89d9784f19bcb74e3&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=178970766&response-content-disposition=attachment%3B%20filename%3Dsparko_linux_arm&response-content-type=application%2Foctet-stream [following]
--2022-02-13 15:21:10--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/178970766/89d177f2-066a-4c6c-a14d-c37c3967825a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220213%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220213T152109Z&X-Amz-Expires=300&X-Amz-Signature=6489b9b2128dca0884bdea83064fea70beb7a763a0389cc89d9784f19bcb74e3&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=178970766&response-content-disposition=attachment%3B%20filename%3Dsparko_linux_arm&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.111.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 18350080 (18M) [application/octet-stream]
Saving to: ‘/home/bitcoin/cl-plugins-available/sparko’

/home/bitcoin/cl-plugins-av 100%[==========================================>]  17.50M  8.33MB/s    in 2.1s

2022-02-13 15:21:12 (8.33 MB/s) - ‘/home/bitcoin/cl-plugins-available/sparko’ saved [18350080/18350080]

# Editing /home/bitcoin/.lightning/config
# See: https://github.com/fiatjaf/sparko#how-to-use

sparko-host=0.0.0.0
sparko-port=9000
sparko-tls-path=/home/bitcoin/.lightning/sparko-tls
sparko-login=blitz:12345678
sparko-keys=uWsuQWcTfcrSFICCgt3z; IePHjDYK1fpyzfLfhVhI: getinfo, listchannels, listnodes; KdtO53SzF8JavgNTfoj7: pay; uMjIUjVAYLcD9W6FpF5k: stream

# Allowing port 9000 through the firewall
Rule updated
Rule updated (v6)

# Hidden Service for sparko
HiddenServiceDir /mnt/hdd/tor/sparko
HiddenServiceVersion 3
HiddenServicePort 443 127.0.0.1:9000

Reloading Tor to activate the Hidden Service...

The Tor Hidden Service address for sparko is:
juwansfawcyagmmo7r6s5t4dolhsb2dlw7p7i7phqo5xw2xjdwacbnqd.onion
use with the port: 443
# Restart the lightningd.service to activate Sparko
# Sparko was installed
# Monitor with:
sudo journalctl | grep sparko | tail -n5
sudo tail -n 100 -f /home/bitcoin/.lightning/bitcoin/cl.log | grep sparko
admin@192.168.178.61:~ ₿
```

openoms, 2022-02-13:
it looks, ok. So your CL is running with Sparko now?

rootzoll, 2022-02-13:
No it still runs into the same problem

```
Feb 13 18:24:33 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Feb 13 18:24:33 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Feb 13 18:24:34 raspberrypi sh[1818402]: lightningd: /home/bitcoin/.lightning/config line 20: sparko-host=0.0.0.0: unrecognized option
Feb 13 18:24:34 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Feb 13 18:24:34 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
```
```
cl plugin list
lightning-cli: Connecting to 'lightning-rpc': Connection refused
```

and when I remove the `sparko-` linies manually again from the config and restart - there is no sparko in the list:
```
admin@192.168.178.61:~ ₿ cl plugin list
{
   "command": "list",
   "plugins": [
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/autoclean",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/bcli",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/fetchinvoice",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/funder",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/topology",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/keysend",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/offers",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/pay",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/txprepare",
         "active": true
      },
      {
         "name": "/usr/local/bin/../libexec/c-lightning/plugins/spenderp",
         "active": true
      }
   ]
}
```

openoms, 2022-02-13:
So despite it downloaded to:
`/home/bitcoin/cl-plugins-available/sparko`
and should have been symlinked to:
`/home/bitcoin/cl-plugins-enabled/sparko`
it is not loading. You don't have the backup plugin either.

Can you check if your you cl config 
`sudo cat /home/bitcoin/.lightning/config`
has the line:
```
plugin-dir=/home/bitcoin/cl-plugins-enabled
```

If not that it is is probably from one of the earliest iterations.

Could check for this line with `cl.check.sh prestart`.

openoms, 2022-02-13:
try to patch, [f6aeeb0](https://github.com/rootzoll/raspiblitz/commit/f6aeeb0fa7507325431c08c2297beef4124184f9) should solve it

rootzoll, 2022-02-14:
Installed it on my other testmachine - Sparko is working. Maybe it has something todo with the BTRFS thats on the other testmachine were its failing. Seems to be an edge case, should not be a blocker for 1.7.2 - pushing this issue to recheck for v1.8

closed 2023-11-14 by rootzoll