raspiblitz/raspiblitz #2954: can't ssh in after updating policykit and reboot

JessieRaspi, 2022-02-12:
Hi,
I did the update and install of the policykit with openoms guide.
The updating process went smoothly, I could reboot and ssh in, but my screen went into the same loop as I had a few months ago. Tried to rescue it the same way as before, but failed. Now my screen only shows an arrow in the middle, which I can't move.  
I now wonder what to do, now that I can't SSH in anymore. Hope you can advise me!

openoms, 2022-02-12:
Not sure what happened as not aware of your previous issue.

In any case the best option is to reflash the SDcard. You could also test the v1.7.2rc2c image from the dev branch for the latest updates (including policykit).

JessieRaspi, 2022-02-13:
My previous issue was the hanging touchscreen (#2637), but this is different. 

Yes, I will reflash the SDcard. Thanks for your advice.

closed 2022-02-13 by JessieRaspi