raspiblitz/raspiblitz #2959: Awaiting Unilateral Close - RTL try to close channels that never was funded.

moskusminer, 2022-02-13:
Looks like CL-Boss tried to open channels that never was funded, now my node is stuck with multiple channels.. i tried "close all" "turned and turned CL-Boss off.
Now im not able to send onchain sats to a difrent onchain wallet, and start over with a new node. 

Next i tried to start CL-boss again and was able to send half of my sats over lightning. 

is there any way to delete those channels thats not funded and says "Awaiting Unilateral Close"?

openoms, 2022-02-13:
`lightning-cli listfunds`
tells you where your sats are.

CLBOSS is opening channels automatically after you have sats available onchain so if you want to send onchain to a different wallet keep CLBOSS off.

moskusminer, 2022-02-13:
i know.. but cl-boss off does not change anything.. even if my channels are closes and funds are onchain. i get error when trying to send. 
like my UTXOs are reserved or something. 
[image]

openoms, 2022-02-13:
`AWAITING_UNILATERAL` means that the channels were force closed (when you run CLOSE ALL) and waiting for the timelock to expire. Come back to it in a few days and check 
`ligthning-cli listfunds`
again.

moskusminer, 2022-02-13:
I have been waiting a week now.. think I have to run "lightning-cli unreserveinputs" but don't know how. 

Looks like CL-Boss tried to one channels without enough balance onchain. Mempool says "waiting on transaction"

openoms, 2022-02-13:
As this is more of C-lightning specific issue (not caused by the raspiblitz implementation) best to look for support channels with C-lightning devs:
https://github.com/ElementsProject/lightning
https://t.me/lightningd
https://matrix.to/#/#c-lightning:libera.chat or same on IRC
https://discord.gg/YGdpyj2aXj

Could help looking into it if you share txid-s in text.
Also are you week since you have run "close all" or  since when CLBOSS opened the channels?

moskusminer, 2022-02-13:
Yes about a week ago since "close all" 

found this on a c-lightning issue:
can use unreservepsbt.

You'll need to identify the specific utxo's that you want to unreserve, pass them to `utxopsbt` with `reserve=false` to build a psbt that has them as an input, and then pass the resulting psbt to `unreservepsbt`, which will unreserve them for you.

i did not understad how, but anyway here is some of the TXIDs

TXID-S:
(ada812079231e853f064f86be9bf1cf5e6ca4674097a1c4f64996bd92d6786cf)
(b9fa4692ede69041b6049141ad797e85ab16a2d6b58c9d73333cae09f79173d9)
(a8ad7a3a83b074a8ad27d0d0204d693b98a0d9e95032a40f315bf9d3c5a25ea5)
(d52e105c289ec692b3dbe89f3bde469ee71b3d200950cc9f4b9629ca3d13dbeb)
(12e9f03ce0e12cbc8cfdf4491a2606c8650af7d0d29c84063feb155b105f8364)
(022a96e4fba329a8c79e0058eb373bfc0a554bfc87556d6549d05dad7861f61e)
(022a96e4fba329a8c79e0058eb373bfc0a554bfc87556d6549d05dad7861f61e)

moskusminer, 2022-02-13:
[image]

moskusminer, 2022-02-14:
MPOOL   Mempool Space                            x x           "outputs": [   SPECTER   Specter Desktop                          x x              { x         SETTINGS  Node Settings & Options                  x x                 "txid": "be6b1f712f98a741ee6071d8b1aaf07eaa3100059648f8eed6f965b2115aa889",  x x         SYSTEM    Monitoring & Configuration               x x                 "output": 1,NECT   Connect Apps & Show Credentials          x x                 "value": 9845,ORD  Change Passwords                         x x                 "amount_msat": "9845000msat",n Calibration                  x x                 "scriptpubkey": "0014eb2f61f01355c89092551d64e6c01499499d17a1",                 "address": "bc1qavhkruqn2hyfpyj4r4jwdsq5n9ye69ap3ytpw6",    x x                 "status": "unconfirmed",t RaspiBlitz                        x x                 "reserved": false  PowerOff RaspiBlitz                      x x              },mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj x              {qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqu                 "txid": "cb0b05fc16794891bc92c6d9bb1f9b3577761351933764eaacf8f211c8fcef0c",  mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj                 "output": 1,                                                                    "value": 26026,                                                                 "amount_msat": "26026000msat",                                                  "scriptpubkey": "0020d79c64fbaf8e6211718343f4ddcd90faceef84baadcad3700cf22fff618613f1",                                                                         "address": "bc1q67wxf7a03e3pzuvrg06dmnvslt8wlp964h9dxuqv7ghl7cvxz0csvdqgtf",                                                                                    "status": "confirmed",                                                          "blockheight": 721552,                                                          "reserved": true,
         "reserved_to_block": 723330
      },
      {
         "txid": "eb665bee29328be6bac1d19263ab1b49840bb79ce128f39b742ba867c3b46491",
         "output": 0,
         "value": 4101,
         "amount_msat": "4101000msat",
         "scriptpubkey": "0014dd963c865b8532acc4b2632df3f306e2e99d222c",
         "address": "bc1qmktrepjms5e2e39jvvkl8ucxut5e6g3vfh4853",
         "status": "confirmed",
         "blockheight": 722806,
         "reserved": true,
         "reserved_to_block": 723330
      },
      {
         "txid": "8bed864e5ef4b29ada4779e2120098ea99b66fe51618522e0f2c1199d889ccdd",
         "output": 1,
         "value": 731047,
         "amount_msat": "731047000msat",
         "scriptpubkey": "001457101794d60c7baafbe723480cb401939f2cc878",
         "address": "bc1q2ugp09xkp3a647l8ydyqedqpjw0jejrch4c9f0",
         "status": "confirmed",
         "blockheight": 723161,
         "reserved": true,
         "reserved_to_block": 723330
      }
   ],
   "channels": [
      {
         "peer_id": "03c2abfa93eacec04721c019644584424aab2ba4dff3ac9bdab4e9c97007491dda",
         "connected": false,
         "state": "AWAITING_UNILATERAL",
         "short_channel_id": "721541x1318x0",
         "channel_sat": 390000,
         "our_amount_msat": "390000000msat",
         "channel_total_sat": 390000,
         "amount_msat": "390000000msat",
         "funding_txid": "be6b1f712f98a741ee6071d8b1aaf07eaa3100059648f8eed6f965b2115aa889",
         "funding_output": 0
      },
      {
         "peer_id": "0242a4ae0c5bef18048fbecf995094b74bfb0f7391418d71ed394784373f41e4f3",
         "connected": false,
         "state": "AWAITING_UNILATERAL",
         "channel_sat": 1147052,
         "our_amount_msat": "1147052000msat",
         "channel_total_sat": 1147052,
         "amount_msat": "1147052000msat",
         "funding_txid": "6426bdc7220d06a99c9c8a306cbc31b1684b7fa8e6f9084b2c8c48eb7b8c5fa7",
         "funding_output": 2
      },
      {
         "peer_id": "03bc9337c7a28bb784d67742ebedd30a93bacdf7e4ca16436ef3798000242b2251",
         "connected": false,
         "state": "AWAITING_UNILATERAL",
         "channel_sat": 1147052,
         "our_amount_msat": "1147052000msat",
         "channel_total_sat": 1147052,
         "amount_msat": "1147052000msat",
         "funding_txid": "6426bdc7220d06a99c9c8a306cbc31b1684b7fa8e6f9084b2c8c48eb7b8c5fa7",
         "funding_output": 1
      },
      {
         "peer_id": "03391210b7fd47678df727ed7c39ea2f3c8dd43d6fb747582b1c13d4c1376f4714",
         "connected": false,
         "state": "AWAITING_UNILATERAL",
         "channel_sat": 2294105,
         "our_amount_msat": "2294105000msat",
         "channel_total_sat": 2294105,
[... 201 lines cut ...]
      },
      {
         "peer_id": "0267586e510d80fab314fbfa091f383b775646e7a87bc9ed8cd59bcded50405d6a",
         "connected": false,
         "state": "AWAITING_UNILATERAL",
         "channel_sat": 1641666,
         "our_amount_msat": "1641666000msat",
         "channel_total_sat": 1641666,
         "amount_msat": "1641666000msat",
         "funding_txid": "b9fa4692ede69041b6049141ad797e85ab16a2d6b58c9d73333cae09f79173d9",
         "funding_output": 1
      },
      {
         "peer_id": "030c3f19d742ca294a55c00376b3b355c3c90d61c6b6b39554dbc7ac19b141c14f",
         "connected": false,
         "state": "AWAITING_UNILATERAL",
         "channel_sat": 1182094,
         "our_amount_msat": "1182094000msat",
         "channel_total_sat": 1182094,
         "amount_msat": "1182094000msat",
         "funding_txid": "ada812079231e853f064f86be9bf1cf5e6ca4674097a1c4f64996bd92d6786cf",
         "funding_output": 0
      }
   ]
}

openoms, 2022-02-14:
Ok so need to find out the state of those closes.
Another usefuk command is:
`lightning-cli listpeers`

(you can use `clear` or CTRL+L to refresh the terminal screen)

moskusminer, 2022-02-14:
"state_changes": [
                  {
                     "timestamp": "2022-02-13T11:04:56.837Z",
                     "old_state": "CHANNELD_AWAITING_LOCKIN",
                     "new_state": "CHANNELD_SHUTTING_DOWN",
                     "cause": "user",
                     "message": "User or plugin invoked close command"
                  },
                  {
                     "timestamp": "2022-02-13T11:05:26.843Z",
                     "old_state": "CHANNELD_SHUTTING_DOWN",
                     "new_state": "AWAITING_UNILATERAL",
                     "cause": "user",
                     "message": "Forcibly closed by `close` command timeout"
                  }
               ],
               "status": [
                  "AWAITING_UNILATERAL:Will attempt reconnect in 9 seconds"
               ],
               "in_payments_offered": 0,
               "in_msatoshi_offered": 0,
               "in_offered_msat": "0msat",
               "in_payments_fulfilled": 0,
               "in_msatoshi_fulfilled": 0,
               "in_fulfilled_msat": "0msat",
               "out_payments_offered": 0,
               "out_msatoshi_offered": 0,
               "out_offered_msat": "0msat",
               "out_payments_fulfilled": 0,
               "out_msatoshi_fulfilled": 0,
               "out_fulfilled_msat": "0msat",
               "htlcs": []
            }
         ]
      },
      {
         "id": "024a8228d764091fce2ed67e1a7404f83e38ea3c7cb42030a2789e73cf3b341365",
         "connected": false,
         "channels": [
            {
               "state": "AWAITING_UNILATERAL",
               "scratch_txid": "86f987652c8e85a56aa45429b73e3982a8bd1aaacbb3eefd2a326c4c44bd1833",
               "last_tx_fee": "183000msat",
               "last_tx_fee_msat": "183000msat",
               "feerate": {
                  "perkw": 1001,
                  "perkb": 4004
               },
               "channel_id": "1ef66178ad5dd049656d5587fc4b550afc3b37eb58009ec7a829a3fbe4962a03",
               "funding_txid": "022a96e4fba329a8c79e0058eb373bfc0a554bfc87556d6549d05dad7861f61e",
               "close_to_addr": "bc1q0fyx3k6t0heka39pjaw47zaf22rr4nz2z387vd",
               "close_to": "00147a4868db4b7df36ec4a1975d5f0ba952863acc4a",
               "private": false,
               "opener": "local",
               "closer": "local",
               "features": [
                  "option_static_remotekey"
               ],
               "funding_allocation_msat": {
                  "024a8228d764091fce2ed67e1a7404f83e38ea3c7cb42030a2789e73cf3b341365": 0,
                  "0246edb5a101688147e844e7ade511ae10a1a065ce3cd84a871ce35fb277dc5e15": 1947953000
               },
               "funding_msat": {
                  "024a8228d764091fce2ed67e1a7404f83e38ea3c7cb42030a2789e73cf3b341365": "0msat",
                  "0246edb5a101688147e844e7ade511ae10a1a065ce3cd84a871ce35fb277dc5e15": "1947953000msat"
               },
               "funding": {
                  "local_msat": "1947953000msat",
                  "remote_msat": "0msat"
               },
[... 831 lines cut ...]
                     "message": "Forcibly closed by `close` command timeout"
                  }
               ],
               "status": [
                  "AWAITING_UNILATERAL:Will attempt reconnect in 18 seconds"
               ],
               "in_payments_offered": 0,
               "in_msatoshi_offered": 0,
               "in_offered_msat": "0msat",
               "in_payments_fulfilled": 0,
               "in_msatoshi_fulfilled": 0,
               "in_fulfilled_msat": "0msat",
               "out_payments_offered": 0,
               "out_msatoshi_offered": 0,
               "out_offered_msat": "0msat",
               "out_payments_fulfilled": 0,
               "out_msatoshi_fulfilled": 0,
               "out_fulfilled_msat": "0msat",
               "htlcs": []
            }
         ]
      }
   ]
}
admin@debianblitz:~ $

openoms, 2022-02-15:
You need to wait for the timelock to expire.
from your data:
```
cat paste.txt | grep "our_to_self_delay"
               "our_to_self_delay": 144,
               "our_to_self_delay": 144,
               "our_to_self_delay": 144,
               "our_to_self_delay": 275,
               "our_to_self_delay": 275,
               "our_to_self_delay": 144,
               "our_to_self_delay": 432,
               "our_to_self_delay": 400,
               "our_to_self_delay": 400,
               "our_to_self_delay": 299,
               "our_to_self_delay": 400,
               "our_to_self_delay": 144,
               "our_to_self_delay": 162,
               "our_to_self_delay": 144,
               "our_to_self_delay": 311,
               "our_to_self_delay": 144,
               "our_to_self_delay": 165,
               "our_to_self_delay": 157,
               "our_to_self_delay": 399,
               "our_to_self_delay": 2000,
               "our_to_self_delay": 144,
               "our_to_self_delay": 144,
               "our_to_self_delay": 197,
               "our_to_self_delay": 144,
```
From https://lightning.readthedocs.io/lightning-listpeers.7.html
```
our_to_self_delay (u32, optional): the number of blocks before we can take our funds if we unilateral close
```

moskusminer, 2022-02-15:
ok. thanks! really appreciate the help! i might have been a little reckless! but lightning is so much fun! 
So nothing i can do to speed them up?