raspiblitz/raspiblitz #2961: 1.7.2rc2 - Fresh Sync Wrong Progress Indication

mrsieb, 2022-02-14:
On a fresh sync the progress Number looks strange, and counts down starting from 100000%

[image]
[image]

mrsieb, 2022-02-16:
After sync is complete everything looks good, its just  during sync.

rootzoll, 2022-02-16:
Thanks for the report - will check if I can reproduce.

rootzoll, 2022-02-16:
I did a setup of a fresh sync and number look OK - so I was not able to reproduce this. I leave this issue open as unconfirmed until the v1.8 release for anybody to report this seeing the same.

rootzoll, 2022-05-09:
Did not got further reports on this. Closing for now.

closed 2022-05-09 by rootzoll