raspiblitz/raspiblitz #2963: [v1.7.2rc2c]  LCD screen behaviour

JessieRaspi, 2022-02-15:
It seems, that the LCD screen is not as stable as in [1.7.1].  The screen turns black regularly with only a cursor and some info above  (see photo). After a while the information comes back, but sometimes only half the info.
Also temperature is higher than before, constantly above 60 `C. I use the recommended hardware with heatsinkcase and I have the Raspiblitz running constantly since june `21.
I don't have much services activated and running at the moment (only Electrs, Mempool, BTC explorer, LND-RTL), don't have channels yet.
SSh works, services are working all well, too and logs seems good as far as I can judge. 
I did a reboot again, but the screen behaviour didn't change.

[image]
.

openoms, 2022-02-16:
DIdn't reproduce this yet.
Please try syncing the latest commits with
`patch`
and restart.

Also try to click `Off` then `Cancel`

openoms, 2022-02-16:
Found that refreshing the status screen takes at least 3 seconds and repeated every 5 seconds. 
I presume if the refresh takes longer occasionally it restarts and ends up with the black screen.

Can extend the refresh time in:
`nano 00infoLCD.sh`
line 138 on the end
change `sleep 5` to `sleep 30` for example to refresh only twice a minute.
This seems to help, but still getting a blank screen sometimes.

openoms, 2022-02-16:
Checking  
`time bash -x 00infoBlitz.sh `
there is also a 1 sec sleep in there ending up with:
```
real    0m3.173s
user    0m1.155s
sys     0m0.888s
```

openoms, 2022-02-16:
playing with the sleep values does not seem to help so likely will need to tweak:
https://github.com/rootzoll/raspiblitz/blob/2daaff60be62121d1a3aa8c1979657deeb342507/home.admin/BlitzTUI/blitztui/main.py#L157

openoms, 2022-02-16:
It is likely some QT issue as cannot repruduce the blank screen with:
```
# test by connecting to the raspiblitz 'ssh -X admin@LAN_IP' and run:
uxterm -fa Terminus -fs 9 -fn fixed +sb -hold -e 'bash -c "sudo -u pi /home/admin/00infoLCD.sh --pause 0"'
```

rootzoll, 2022-02-16:
I am also not sure about this. Basically for the v1.7.2 the LCD script is not gathering the system information itself anymore and mostly is just getting values from the new cache - which should be faster and should not result in long loading times.

For the v1.8.1 the touchscreen is up for a possible rewrite anyways - so if not a quick fix for this is found we might live with this until then.

JessieRaspi, 2022-02-17:
If a quick fix is not possible, my suggestion is to mention this somewhere in the release notes with the coming version 1.7.2.  So, that people without any experience in system development are able to understand, that the LCD script is different from the version 1.7.1 and that it can cause some delay in showing the information, but that this is not a big problem and that it will be fixed in a later release.

JessieRaspi, 2022-02-17:
My screen was freezing again (as in photo above) as I mentioned in issue #2960 and the Raspiblitz lost connection, too. Couldn't SSH in anymore and the buttons were freezed. So I had to do a hard reset and unplugged the LAN cable first, after that the power, waited for a moment and then plugged everything in again. Now, everything looks normal again, but it seems to me, that it is more serious than I thought.
It all started after I tried to update the policykit in version 1.7.1 which didn't succeed and then I did a reflash with version 1.7.2rc2c. 

It's not a mega thing and I can live with it, but may be better to fix this earlier and not wait till 1.8? Because it's a bit scary.....at least it was for me in the beginning!

rootzoll, 2022-02-17:
OK I added info on touchscreen situation to CHANGES & README.

openoms, 2022-02-17:
@rootzoll I doubt that it is the touchscreen making the system unstable. Two separate issues probably so that wording might be a bit strong (also has a typo).

@JessieRaspi are the stability problems related only the tpuckscreen? is your node not freezing if you deactivate it?

rootzoll, 2022-02-17:
@openoms OK marked touchscreen as "experimental" again until 1.8.1

JessieRaspi, 2022-02-18:
@openoms , yes, the stability problems are touchscreen related. 
I deactivated the touchscreen yesterday and since then, I have no freezing problems.

I tried to connect Zeus again, however the display still didn't show the QR code. Then I tried Console QR to display it on my pc, but a gigantic QR code appeared (I'm on Windows 10), which I couldn't get smaller, so I couldn't scan.

openoms, 2022-02-18:
I couldn't reproduce the stability problem with my node running with the touchscreen on.

For the QRs to appear on the LCD run:
`patch`
and try again.
There is a way to shrink the size on Windows as well, just need to look it up in the FAQ or issues.

closed 2022-05-02 by openoms