raspiblitz/raspiblitz #2964: Health Check Issue under v1.7.2.rc2

konigra, 2022-02-15:
Hi guys,

I experienced lots of rebooting activity from raspiblitz since I updated from 1.7.1. to 1.7.2rc2.
The reason seems to be the following:

I monitored the LND daemon closely and at some point (about 5 minutes into starting up, starting bitcoind and the LND client) this happens:

2022-02-15 14:21:29.896 [INF] PEER: unable to read message from 0298f6074a454a1f5345cb2a7c6f9fce206cd0bf675d177cdbf0ca7508dd28852f@73.114.40.246:9735: read tcp 10.0.0.173:57124->73.114.40.246:9735: use of closed network connection
2022-02-15 14:21:29.896 [INF] HLCK: Health monitor shutting down
2022-02-15 14:21:29.896 [DBG] HLCK: Health check: monitor quit
2022-02-15 14:21:30.093 [DBG] SRVR: Peer 030c3f19d742ca294a55c00376b3b355c3c90d61c6b6b39554dbc7ac19b141c14f@52.50.244.44:9735 has been disconnected
2022-02-15 14:21:30.095 [DBG] SRVR: Peer 03271338633d2d37b285dae4df40b413d8c6c791fbee7797bc5dc70812196d7d5c@3.95.117.200:9735 has been disconnected
2022-02-15 14:21:30.096 [DBG] SRVR: Peer 03bc9337c7a28bb784d67742ebedd30a93bacdf7e4ca16436ef3798000242b2251@46.229.165.141:9735 has been disconnected
2022-02-15 14:21:30.097 [DBG] SRVR: Peer 03f3297397c8f5f685a562847611e20d15f56d6aaabc4d808a6e04e631dea6e612@188.34.182.180:9735 has been disconnected
2022-02-15 14:21:30.097 [DBG] SRVR: Peer 0298f6074a454a1f5345cb2a7c6f9fce206cd0bf675d177cdbf0ca7508dd28852f@73.114.40.246:9735 has been disconnected
2022-02-15 14:21:30.113 [INF] RPCS: Stopping RPC Server
2022-02-15 14:21:30.116 [INF] RPCS: Stopping SignRPC Sub-RPC Server
2022-02-15 14:21:30.119 [INF] RPCS: Stopping RouterRPC Sub-RPC Server
2022-02-15 14:21:30.119 [INF] RPCS: Stopping WatchtowerClientRPC Sub-RPC Server
2022-02-15 14:21:30.119 [INF] RPCS: Stopping InvoicesRPC Sub-RPC Server
2022-02-15 14:21:30.119 [INF] RPCS: Stopping VersionRPC Sub-RPC Server
2022-02-15 14:21:30.119 [INF] RPCS: Stopping WatchtowerRPC Sub-RPC Server
2022-02-15 14:21:30.122 [INF] RPCS: Stopping WalletKitRPC Sub-RPC Server
2022-02-15 14:21:30.123 [INF] RPCS: Stopping AutopilotRPC Sub-RPC Server
2022-02-15 14:21:30.123 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
2022-02-15 14:21:30.181 [INF] LTND: Shutdown complete

I cannot find anything about the "health monitor" in LND client.
I checked the powersupply (its working).
I checked the internet connection (its working).

It must be the software for some reason telling my raspiblitz to reboot.

Any suggestions what I can do next to solve the problem?

Regards,

Koni

openoms, 2022-02-16:
it might be you block data corrupted. Please monitorthe bitcoin debug.log for errors and resync the chain if neccesary.

konigra, 2022-02-16:
Hello openoms, thank you for your response.

I directly went for the resyncing of the blockchain through the raspiblitz menu. 

Now it is resyncing, but rebooting nonetheless every 10 minutes.

Will see what it does when it is synced.

konigra, 2022-02-16:
The whole node is rebooting.
(I have the original raspiblitz setup, so hardwarewise shouldn't be an issue.)
(I checked if the receiveing too less power, but it isn´t the case. CPU is not "throttled", everything is fine).

openoms, 2022-02-16:
This is weird. Can you try to reflash the SDcard? 
Can also try to exit to cli with ctrl+c and run: `patch` before recovery to sync the latest commits.

konigra, 2022-02-18:
Hello opennoms,

I patched it, and the blockchain is resyncing since 2 days.

What I observed: When I connect to the raspiblitz via LAN-cable (laptop -> Raspi), it tends to shut down.

Now when I cut the connection via LAN-cable, it seems to work fine.

Any idea why this could be an issue?
Will oberserve and report back.

Thank you for your time and thoughts openoms.

konigra, 2022-02-22:
Somehow connecting via LAN-cable from laptop to Raspberry Pi sends my node rebooting. IDK why but got the issue circled enough to avoid id. Thanks for help openoms. Will close the issue now.

closed 2022-02-22 by konigra