raspiblitz/raspiblitz #2972: dev 1.7.2.rc2 build_sdcard.sh on Debian 11 VM

Firewalker99, 2022-02-16:
Hey, want to install Blitz in a Virtual Machine with Debian11.
HDD is connectet.
wget --no-cache https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh

sudo bash ./build_sdcard.sh --interaction 1 --fatpack 1 --github-user rootzoll --branch dev --display headless --tweak-boot-drive 0 --wifi-region off

Error:
`
*** Install Tor ***
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  tor-geoipdb torsocks
Vorgeschlagene Pakete:
  mixmaster torbrowser-launcher tor-arm apparmor-utils
Die folgenden NEUEN Pakete werden installiert:
  tor tor-geoipdb torsocks
0 aktualisiert, 3 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 3.498 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 15,8 MB Plattenplatz zusätzlich benutzt.
Holen:1 http://deb.debian.org/debian bullseye/main amd64 tor amd64 0.4.5.10-1~deb11u1 [2.019 kB]
Holen:2 http://deb.debian.org/debian bullseye/main amd64 tor-geoipdb all 0.4.5.10-1~deb11u1 [1.402 kB]
Holen:3 http://deb.debian.org/debian bullseye/main amd64 torsocks amd64 2.3.0-3 [76,6 kB]
Es wurden 3.498 kB in 1 s geholt (5.291 kB/s).
Vormals nicht ausgewähltes Paket tor wird gewählt.
(Lese Datenbank ... 158315 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../tor_0.4.5.10-1~deb11u1_amd64.deb ...
Entpacken von tor (0.4.5.10-1~deb11u1) ...
Vormals nicht ausgewähltes Paket tor-geoipdb wird gewählt.
Vorbereitung zum Entpacken von .../tor-geoipdb_0.4.5.10-1~deb11u1_all.deb ...
Entpacken von tor-geoipdb (0.4.5.10-1~deb11u1) ...
Vormals nicht ausgewähltes Paket torsocks wird gewählt.
Vorbereitung zum Entpacken von .../torsocks_2.3.0-3_amd64.deb ...
Entpacken von torsocks (2.3.0-3) ...
tor (0.4.5.10-1~deb11u1) wird eingerichtet ...
Something or somebody made /var/lib/tor disappear.
Creating one for you again.
Something or somebody made /var/log/tor disappear.
Creating one for you again.
Created symlink /etc/systemd/system/multi-user.target.wants/tor.service → /lib/systemd/system/tor.service.
torsocks (2.3.0-3) wird eingerichtet ...
tor-geoipdb (0.4.5.10-1~deb11u1) wird eingerichtet ...
Trigger für man-db (2.9.4-2) werden verarbeitet ...
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
curl ist schon die neueste Version (7.74.0-1.3+deb11u1).
gpg ist schon die neueste Version (2.2.27-2).
gpg wurde als manuell installiert festgelegt.
obfs4proxy ist schon die neueste Version (0.0.8-1+b6).
torsocks ist schon die neueste Version (2.3.0-3).
torsocks wurde als manuell installiert festgelegt.
Die folgenden NEUEN Pakete werden installiert:
  apt-transport-tor nyx python3-stem
0 aktualisiert, 3 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 385 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 1.943 kB Plattenplatz zusätzlich benutzt.
Holen:1 http://deb.debian.org/debian bullseye/main amd64 apt-transport-tor all 0.5 [7.612 B]
Holen:2 http://deb.debian.org/debian bullseye/main amd64 python3-stem all 1.8.0-3 [310 kB]
Holen:3 http://deb.debian.org/debian bullseye/main amd64 nyx all 2.1.0-2.1 [67,8 kB]
Es wurden 385 kB in 0 s geholt (1.666 kB/s).
Vormals nicht ausgewähltes Paket apt-transport-tor wird gewählt.
(Lese Datenbank ... 158393 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../apt-transport-tor_0.5_all.deb ...
Entpacken von apt-transport-tor (0.5) ...
Vormals nicht ausgewähltes Paket python3-stem wird gewählt.
Vorbereitung zum Entpacken von .../python3-stem_1.8.0-3_all.deb ...
Entpacken von python3-stem (1.8.0-3) ...
Vormals nicht ausgewähltes Paket nyx wird gewählt.
Vorbereitung zum Entpacken von .../archives/nyx_2.1.0-2.1_all.deb ...
Entpacken von nyx (2.1.0-2.1) ...
python3-stem (1.8.0-3) wird eingerichtet ...
update-alternatives: /usr/bin/python3-tor-prompt wird verwendet, um /usr/bin/tor-prompt (tor-prompt) im automatischen Modus bereitzustellen
apt-transport-tor (0.5) wird eingerichtet ...
nyx (2.1.0-2.1) wird eingerichtet ...
Trigger für man-db (2.9.4-2) werden verarbeitet ...

*** Adding deb.torproject.org keyring ***
gpg: Keine gültigen OpenPGP-Daten gefunden.
- OK key added

*** Adding Tor Sources ***

deb [arch=amd64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye main
deb-src [arch=amd64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org  bullseye main

- OK sources added

*** Reinstall ***
OK:1 http://security.debian.org/debian-security bullseye-security InRelease
OK:2 http://deb.debian.org/debian bullseye InRelease
OK:3 http://deb.debian.org/debian bullseye-updates InRelease
OK:4 https://deb.nodesource.com/node_12.x bullseye InRelease
Holen:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease [3.526 B]
Fehl:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease
  Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 74A941BA219EC810
Paketlisten werden gelesen… Fertig
W: GPG-Fehler: tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 74A941BA219EC810
E: Das Depot »tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease« ist nicht signiert.
N: Eine Aktualisierung von solch einem Depot kann nicht auf eine sichere Art durchgeführt werden, daher ist es standardmäßig deaktiviert.
N: Weitere Details zur Erzeugung von Paketdepots sowie zu deren Benutzerkonfiguration finden Sie in der Handbuchseite apt-secure(8).
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
tor ist schon die neueste Version (0.4.5.10-1~deb11u1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
apt-transport-tor ist schon die neueste Version (0.5).
curl ist schon die neueste Version (7.74.0-1.3+deb11u1).
gpg ist schon die neueste Version (2.2.27-2).
nyx ist schon die neueste Version (2.1.0-2.1).
obfs4proxy ist schon die neueste Version (0.0.8-1+b6).
python3-stem ist schon die neueste Version (1.8.0-3).
torsocks ist schon die neueste Version (2.3.0-3).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

*** PREPARING BITCOIN ***
gpg: Verzeichnis `/root/.gnupg' erzeugt
gpg: Die "Keybox" `/root/.gnupg/pubring.kbx' wurde erstellt
gpg: /root/.gnupg/trustdb.gpg: trust-db erzeugt
gpg: Schlüssel 74810B012346C9A6: Öffentlicher Schlüssel "Wladimir J. van der Laan <laanwj@visucore.com>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:                              importiert: 1
--2022-02-16 19:17:56--  https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS
Auflösen des Hostnamens bitcoincore.org (bitcoincore.org)… 198.251.83.116, 107.191.99.5
Verbindungsaufbau zu bitcoincore.org (bitcoincore.org)|198.251.83.116|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 2353 (2,3K) [application/octet-stream]
Wird in »SHA256SUMS« gespeichert.

SHA256SUMS                                     100%[=================================================================================================>]   2,30K  --.-KB/s    in 0s

2022-02-16 19:17:57 (34,6 MB/s) - »SHA256SUMS« gespeichert [2353/2353]

--2022-02-16 19:17:57--  https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS.asc
Auflösen des Hostnamens bitcoincore.org (bitcoincore.org)… 198.251.83.116, 107.191.99.5
Verbindungsaufbau zu bitcoincore.org (bitcoincore.org)|198.251.83.116|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 9870 (9,6K) [application/octet-stream]
Wird in »SHA256SUMS.asc« gespeichert.

SHA256SUMS.asc                                 100%[=================================================================================================>]   9,64K  --.-KB/s    in 0s

2022-02-16 19:17:58 (97,9 MB/s) - »SHA256SUMS.asc« gespeichert [9870/9870]

goodSignature(0)
correctKey(1)

!!! BUILD FAILED --> PGP Verify not OK / signature(0) verify(1)
`

openoms, 2022-02-16:
can you try to verify the SHA256SUMS.asc amanually?

Firewalker99, 2022-02-16:
download₿ gpg --verify SHA256SUMS.asc
gpg: die unterzeichneten Daten sind wohl in 'SHA256SUMS'
gpg: Signatur vom Fr 10 Sep 2021 13:29:17 CEST
gpg:                mittels RSA-Schlüssel 0CCBAAFD76A2ECE2CCD3141DE2FFD5B1D88CA97D
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Do 09 Sep 2021 22:09:04 CEST
gpg:                mittels RSA-Schlüssel 152812300785C96444D3334D17565732E08E5E41
gpg:                Aussteller "achow101@gmail.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Do 09 Sep 2021 22:16:18 CEST
gpg:                mittels RSA-Schlüssel 0AD83877C1F0CD1EE9BD660AD7CC770B81FD22A8
gpg:                Aussteller "benthecarman@live.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Fr 10 Sep 2021 15:00:35 CEST
gpg:                mittels RSA-Schlüssel 590B7292695AFFA5B672CBB2E13FC145CD3F4304
gpg:                Aussteller "darosior@protonmail.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Do 09 Sep 2021 22:54:01 CEST
gpg:                mittels RSA-Schlüssel 28F5900B1BB5D1A4B6B6D1A9ED357015286A333D
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Fr 10 Sep 2021 16:26:03 CEST
gpg:                mittels RSA-Schlüssel 637DB1E23370F84AFF88CCE03152347D07DA627C
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Fr 10 Sep 2021 03:04:14 CEST
gpg:                mittels RSA-Schlüssel CFB16E21C950F67FA95E558F2EEB9F5CC09526C1
gpg:                Aussteller "fanquake@gmail.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Fr 10 Sep 2021 10:03:16 CEST
gpg:                mittels RSA-Schlüssel 6E01EEC9656903B0542B8F1003DB6322267C373B
gpg:                Aussteller "gugger@gmail.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Do 09 Sep 2021 22:07:53 CEST
gpg:                mittels RSA-Schlüssel D1DBF2C4B96F2DEBF4C16654410108112E7EA81F
gpg:                Aussteller "hebasto@gmail.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Fr 10 Sep 2021 09:14:14 CEST
gpg:                mittels RSA-Schlüssel 82921A4B88FD454B7EB8CE3C796C4109063D4EAF
gpg:                Aussteller "jon@atack.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Fr 10 Sep 2021 19:33:30 CEST
gpg:                mittels RSA-Schlüssel 9DEAE0DC7063249FB05474681E4AED62986CD25D
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Do 09 Sep 2021 22:22:36 CEST
gpg:                mittels RSA-Schlüssel 9D3CC86A72F8494342EA5FD10A41BDC3F4FAFF1C
gpg:                Aussteller "aaron@sipsorcery.com"
gpg: Signatur kann nicht geprüft werden: No public key
gpg: Signatur vom Fr 10 Sep 2021 11:59:33 CEST
gpg:                mittels RSA-Schlüssel 74E2DEF5D77260B98BC19438099BAD163C70FBFA
gpg:                Aussteller "will8clark@gmail.com"
gpg: Signatur kann nicht geprüft werden: No public key

Firewalker99, 2022-02-16:
`sudo /home/admin/config.scripts/bitcoin.install.sh install*** PREPARING BITCOIN ***
gpg: Schlüssel 74810B012346C9A6: "Wladimir J. van der Laan <laanwj@visucore.com>" nicht geändert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:                             unverändert: 1
--2022-02-16 19:54:45--  https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS
Auflösen des Hostnamens bitcoincore.org (bitcoincore.org)… 198.251.83.116, 107.191.99.5
Verbindungsaufbau zu bitcoincore.org (bitcoincore.org)|198.251.83.116|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 2353 (2,3K) [application/octet-stream]
Wird in »SHA256SUMS« gespeichert.

SHA256SUMS             100%[===========================>]   2,30K  --.-KB/s    in 0s

2022-02-16 19:54:46 (22,6 MB/s) - »SHA256SUMS« gespeichert [2353/2353]

--2022-02-16 19:54:46--  https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS.asc
Auflösen des Hostnamens bitcoincore.org (bitcoincore.org)… 198.251.83.116, 107.191.99.5
Verbindungsaufbau zu bitcoincore.org (bitcoincore.org)|198.251.83.116|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 9870 (9,6K) [application/octet-stream]
Wird in »SHA256SUMS.asc« gespeichert.

SHA256SUMS.asc         100%[===========================>]   9,64K  --.-KB/s    in 0s

2022-02-16 19:54:46 (82,7 MB/s) - »SHA256SUMS.asc« gespeichert [9870/9870]

goodSignature(0)
correctKey(1)

!!! BUILD FAILED --> PGP Verify not OK / signature(0) verify(1)
`

openoms, 2022-02-16:
Found that this is because of your language setting:
https://github.com/rootzoll/raspiblitz/blob/aca20317c17c87dd79df76370ce339ad488981b0/home.admin/config.scripts/bitcoin.install.sh#L42

Set to English for now to avoid errors like this (might be other places too).

Not a priority for the v1.7.2 release, but could be refactored later to be language agnostic.

closed 2022-02-16 by Firewalker99