raspiblitz/raspiblitz #2973: Finally able to upgrade to 1.71 - SLOW and Inactive Channels

mantiev, 2022-02-17:
I'm not really sure what is going on with my node.  I had no issues since I launched the node and then in the last month on 1.70 channels kept going inactive for no reason and I couldn't do a backup without boot looping the node.  I upgraded to 1.71 through the menu.  Now it's painfully slow, menu's take minutes to go into and make any changes.  My channels are still doing the same thing with them being randomly inactive.  I have to reboot the node constantly to get my 15 channels to all be online.  Then random channels go inactive.  Why is the node having so many issues when no changes have been made?

openoms, 2022-02-17:
check the load in 
`htop`
type:
`debug`
 for general logs (on the end it shows how to generate a link which can be shared here).
Compact your channel.db (will be automatic in next release)
also might be a dying SDcard.
blBest to wait for the new release and reflash to a new SDcard, keep the old as backup.

mantiev, 2022-02-17:
Thanks for the reply.  Looks like TOR is pegging at 100% every now and then.  I'm going to make sure it's updated.

Here is the link from the log.

https://termbin.com/0vhb

I will also compact the channel.db

I do have a new SD card I could use as well that is already flashed at 1.7.1

mantiev, 2022-02-18:
I'm unable to any backup of the node.

tar: Removing leading `/' from member names
/mnt/hdd/lnd/
/mnt/hdd/lnd/tls.key
/mnt/hdd/lnd/v3_onion_private_key
/mnt/hdd/lnd/logs/
/mnt/hdd/lnd/logs/bitcoin/
/mnt/hdd/lnd/logs/bitcoin/mainnet/
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.6075.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.6077.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.6076.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
/mnt/hdd/lnd/version.info
/mnt/hdd/lnd/tls.cert
/mnt/hdd/lnd/lnd.conf
/mnt/hdd/lnd/data/
/mnt/hdd/lnd/data/graph/
/mnt/hdd/lnd/data/graph/mainnet/
/mnt/hdd/lnd/data/graph/mainnet/sphinxreplay.db
/mnt/hdd/lnd/data/graph/mainnet/channel.db

It hangs on the last item and goes into a boot loop

openoms, 2022-02-19:
Possibly your channel.db is too big to fit on the SDcard. Compact it: https://github.com/openoms/lightning-node-management/blob/master/technicals/lnddatabasecompaction.md#auto-compact-on-restart

mantiev, 2022-02-19:
It appears that did the trick.  Somehow it had gotten up to 31GB.  Once the db was compacted the channels that were showing inactive immediately went active.  Thank you for the assistance.

closed 2022-02-19 by mantiev