raspiblitz/raspiblitz #2975: LND won't sync after using a master in COPY-SOURCE

chrisBonGit, 2022-02-17:
Hi, 
I *had* succesfully set up my first LND, 'LND#1'. Raspberry Pi 4, Raspiblitz 1.7.1
I'd set up a few services (RTL, Sphinx Server, etc.), opened a few channels (small amounts of funding) etc and even followed the Plebnet wiki suggestions for making the node robust (SSH keys from HWW, virtual LAN to control local access etc.). This all worked well, happy days. 
At this point I was feeling confident, thought I'd set up another raspiblitz, LND#2, as an experimental node. I decided to save time and use the running LND#1's bitcoin/blocks and bitcoin/chainstate data.
To do this, I tried following the instructions on raspiblitz github for running a workshop, (i.e. https://github.com/rootzoll/raspiblitz/blob/v1.7/WORKSHOP.md). That gives instructions on how to turn a RaspiBlitz into a copy station. LND#2 could not be accessed from LND#1 (possibly due to some step I took in making LND#1 robust). So I tried to be clever…
I shutdown LND#1 and used a linux laptop to rsync the bitcoin/blocks & chainstate data from the HDD of LND#1 to the HDD of LND#2 (i.e. unplugged both HDDs from their raspiblitz device, connected both to linux laptop, used a terminal to rsync over the data - which was very fast). The LND#2 did not work when powered up (using a freshly burned SD card, keeping the blockchain data I'd copied across to the HDD - that was ok, it's a learning experience. HOWEVER, my original LND (LND#1) was also unable to sync with either bitcoin network or lightning network. Seems like bitcoin-daemon was in a start/stop infinite loop. I didn't use XXcleanHDD.sh on LND#1. I did use a menu option on raspiblitz GUI (I this I used REPAIR -? COPY-SOURCE).
 
So, The current plan is to set up LND#2 as a clean node, then restore LND#1 data, which I have been able to export (AFAI can tell) using lnd.backup.sh as per rootzoll's guide / FAQ at https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#i-have-the-full-blockchain-on-another-storage-how-do-i-copy-it-to-the-raspiblitz
So far, I flashed a new SD card. Started LND#2 from the beginning (sync blockchain itself). This is progressing for bitcoin BUT I still see an error for LND which keeps restarting.
So my questions are;
Is there a clever way to bring LND#1 back on-line (I read about problems if you don't unmount a device cleanly when you exit, on linux. But I'm sure I did 'eject' the drive. Also I have remounted that HD to linux and definitely ejected it again)?
If the answer to 1) is "no clever fix, do it the slow way", should I be worried that lightning sync is not making progress (keeps restarting)? Is the blockchain sync going to be useful to me, or it is a waste of time? i.e. should I continue so that I can make a copy of the bitcoin/blocks and bitcoin/chainstate data on my home comptuer to save time when I have to do this again.
Is there a way to force a restart for LND while leaving the bitcoin sync process running? I've been attempting to fix it by myself but the long lead-time to get a clean, successfully synced hdd of bitcoin data makes that approach very slow and frustrating.
Thanks for any suggestions.

rootzoll, 2022-02-23:
marked for v1.8 to recheck COPY-SOURCE feature

chrisBonGit, 2022-02-24:
> Hi first of all as a note: The COPYSTATION mode is just just for special setups, where you need to presync a lot of HDDs. To copy the blockchain from one running node to a second new node there is the "COPY-SOURCE" option: https://github.com/rootzoll/raspiblitz/blob/v1.7/README.md#copy-source-offer-your-blockchain-to-another-raspiblitz-for-setup <https://github.com/rootzoll/raspiblitz/blob/v1.7/README.md#copy-source-offer-your-blockchain-to-another-raspiblitz-for-setup>
> On your situation: Let the self sync run thru - if LND has problems, you should still be able to use that synced blockchain again during next "FRESHSETUP" situations.

Hi,

Thanks for you reply.

I’m still resyncing a fresh / new LND, ‘LND01' (it’s at 86%, but it is much slower than my first LND sync, although this rasp only has less RAM, so maybe that contributes).

To keep busy and learning (and distract myself from global conflict), I’m working with the original LND - LND00- which no longer starts a bitcoin deamon. This is still accessible (via SSH) and I can access the HDD, so I made a recovery file, as described on GitHub.
I stored this on my mac’s drive and then tried to use it to restore the old LND.
But when I try to use the /home/admin/config.scripts/lnd.backup.sh lnd-import-gui I get the error message that file is not found.
I inspect the .sh file and found that it expects me to pass it the Filename if this is not a fresh setup.
So I tried to do this as part of set up (new SD flash), but using the menu option ‘From Migration Backup’ doesn’t work as it expects the migration file to be called ‘raspiblitz…tar.gz.’ Whereas the LND backup file is ‘lnd-rescue…tar.gz’.
I tried just renaming but this didn’t work.

So then I tried again (new SD flash), escaped during the initial set up and used the cli as described in GitHub.
But, I can’t guess the correct syntax.

Please could you give me an example of how to call the restore command from the cli.
I have the lnd-restore file on the LND (I’ve learned how to use scp)
I just can’t figure out how to get /home/admin/config.scripts/lnd.backup.sh -lnd-import-gui to pick up the file
I tried it with no additional arguments, with the filename (running from the same location as the lid-restore file).
I tried making the filename a string (both single and double quotes).

Any advice would be gratefully received.

Many thanks,
Chris.

chrisBonGit, 2022-02-25:
Hi,

Update - I made some progress but now am blocked by the lack of lightning wallet password in the recovery file.

The problem below was (probably) that the blockchain had not re-synced.
So I set up a fresh LND (keeping the blockchain data only) and syncs that up.
Then I re-flashed the SD card and used the menu option to restore an old LND.
This process started ok, but has now failed several times with this error

############## Setup LND
FAIL: missing passwordC in (/var/cache/raspiblitz/temp/raspiblitz.setup)!
***********************************************************
ERROR - please report to development team
***********************************************************

When I vi that file, I confirm that it contains both passwordA & passwordB, but there is no entry for passwordC.
I attempted to add this to the file and reboot - but the LND start up menu stated I must use a new flashed SD card.
When I restarted with a fresh flashed SD card, the problem above repeated.

Is there any way to modify the recovery file so that passwordC will be written into raspiblitz.setup?
I unpacked that archive file (‘lnd-rescue-…..tar.gz’) but it does not contain raspiblitz.setup.

Thanks,
Chris.

chrisBonGit, 2022-03-02:
Update on 02/03/2022
I spent the last week trying to recover my LND on & off-chain balance, and I've hit several road-blocks which gives me some concern about the recovery tools (I hope it is just down to user error...)

Here's what I've tried.

Fresh build of a RaspiBlitz, so that both bitcoin and lightning are synchronised 100%. Then flash another SD card and reboot the raspiblitz with this new card. This is my 'clean start' point in the text below. From this I tried (separately) 2 approaches;
1) restore from the LND-recovery...tar.gz file  (clean start -> FRESHSETUP -> LND -> OLD -> LND RESCUE).
2) restore from the seed words + SCB (clean start -> FRESHSETUP -> LND -> OLD -> SEED + SCB).
Both of these approaches fail.
Method 1) results in the error msg 'EXIT _provision.setup.sh BECAUSE OF ERROR STATE (4)'
Method 2) results in the error msg below;
'FAIL: missing passwordA in (/var/cache/raspiblitz/temp/raspiblitz.setup)!
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(config: missing passwordA)'

Note during method 2), I was only asked for password A (not for B or C).

So, now I'm a bit worried that I can't recover my on- or off-chain balances.
I will start to read about the command line options, but I would like to ask if there is experience of using these recovery methods successfully with RaspiBlitz v1.7.1?

chrisBonGit, 2022-03-02:
Just downloaded and flashed RaspiBlitz v1.7.2.
This worked first time, both on-chain (wallet) and off-chain (channels) restored from the lnd-restore....tar.gz file.
Thanks, this issue can now be closed.

closed 2022-03-02 by chrisBonGit