raspiblitz/raspiblitz #2982: loop command fails on 1.7.2

bfeeny, 2022-02-21:
Upgraded from 1.7.1 to 1.7.2
Created a few Loop Outs in Lightning Terminal
From command line try to execute some loop commands such as "loop monitor", "loop suggestswaps", etc.
```
admin@172.16.5.244:~‚Çø loop setparams
[loop] unable to read macaroon path : open /home/admin/.loop/mainnet/loop.macaroon: no such file or directory 
```

Error always returned is` [loop] unable to read macaroon path : open /home/admin/.loop/mainnet/loop.macaroon: no such file or directory `

openoms, 2022-02-21:
see the LIT menu for details about switching to the `lit` user and commands to use

openoms, 2022-02-22:
Is this solved with typing:
`lit`
then:
`lit-pool -h`
for the options?

bfeeny, 2022-02-23:
Yes, that works, or just doing lit-loop, I was typing loop

openoms, 2022-02-23:
ok, thanks for the update. Closing as solved.

closed 2022-02-23 by openoms