raspiblitz/raspiblitz #2983: C-Lightning: The best way to save backups

moskusminer, 2022-02-21:
What is the best and safest way to do backups (C-lightning) im currently running on a Debian Intel PC, with two SSDs, one for OS and one 1TB SSD for the Raspiblitz. 

- USB stick (like in LND) (USB stick wears out?)
- Nextcloud to a private server?

And how to change the location in the CL-PLUGIN.BACKUP.SH script

openoms, 2022-02-22:
For now you need to edit the path here:
https://github.com/rootzoll/raspiblitz/blob/320015dc6086fb2bf238527d67581eaa574ac638/home.admin/config.scripts/cl-plugin.backup.sh#L75

Skip the `${netprefix}` (this is for testnet and signet backups), use an absolute path and make sure that it is writeable by the `bitcoin` user.
Test the accessibility of the destination directory with:
`sudo -u bitcoin touch /test/path/testfile`

Once edited and saved the script run:
`config.scripts/cl-plugin.backup.sh on`
to initialize the backup to the new location.

For the backup-compact command and cronjob to work edit the path in:
`https://github.com/rootzoll/raspiblitz/blob/320015dc6086fb2bf238527d67581eaa574ac638/home.admin/config.scripts/cl-plugin.backup.sh#L143`
And save the script.

Will change the script to need only one place to edit the backup location and will expose it in the menu.

openoms, 2022-02-22:
Regarding the a cheap USB key being unsuitable for the backup location first need to understand that the backup plugin replicated the `lightningd.sqllite3` (channel database file) on-the-fly on every channel update.
(vs the LND SCB is an encrypted list of LN nodeIDs to be pinged for force close in case of loss of the channel database - that only changes on a channel open or close).

Currently the backup is done on the SDcard so the USB drive need to match at least the quality of an Endurance SDcard model. See some recommendations here: https://github.com/rootzoll/raspiblitz/issues/924

Backing up to a Nextcloud instance is a good idea for security - this can be achieved already if the nextcloud share is mounted in the linux filesystem. The drawback is that it introduces latency as the write of a backup location needs to be confirmed after all channel state changes.

More about CL backups here: https://lightning.readthedocs.io/BACKUP.html

moskusminer, 2022-02-23:
> so you mounted a disk as `/mnt/cl.backup` ? check the mounts with `df -h` Would be best to create at least another directory at the root of the disk and maybe avoid using a dot in a directory name as it makes ti look like a file (for a human). `sudo chown -R bitcoin:bitcoin /mnt/your/path` would be correct to set the owner, but need to make sure that `/mnt` is readable by bitcoin also. For example use `sudo chmod 755 /mnt` to make it readable for all users and executable so they can `cd` in to it. As above test readability with a simple `ls` command with the bitcoin user and create the test file to check if the files inside the directory are writable.
The problem was solved when I formated and mounted again :)

moskusminer, 2022-02-23:
I ended up formatting and mounted it again. 
I also added UUID and mountpoint in fstab file.
Now everything works! Thanks for great help again! 
Now I got a Ubuntu server pc, SSD for OS, SSD for chain, SSD for backup. And it is lightning ⚡ fast! And i love it! 🥇

moskusminer, 2022-02-24:
```
/config.scripts₿ ./cl-plugin.backup.sh on
# Checking dependencies
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
# Stop the lightningd.service
ls: cannot access '/home/bitcoin/.lightning/bitcoin/backup.lock': No such file or directory
# Initialize the backup plugin
Traceback (most recent call last):
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backup-cli", line 100, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backup-cli", line 27, in init
    backend = get_backend(destination, create=True)
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backends.py", line 27, in get_backend
    backend = backend_cl(destination, create=create)
  File "/home/bitcoin/cl-plugins-available/plugins/backup/filebackend.py", line 19, in __init__
    raise ValueError("Attempted to create a FileBackend, but file already exists.")
ValueError: Attempted to create a FileBackend, but file already exists.
# The crontab for admin now is:
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact mainnet

# Started the lightningd.service
```

openoms, 2022-05-05:
> @openoms I meant if with the next version it will be possible to have the real time replication written to an external media like a USB stick with the backup script from Raspiblitz?

A new issue or PR would be welcome, but I feel switching the replication from the SDcard to a some likely less reliable USB stick has no benefit.

openoms, 2022-05-05:
Alternatively you could switch on the sqlite replication feature to a third destination, but that has a higher wear to the media so would rather suitable for a second SSD.

Since that is not equipped with the RPis a manual guide to do it would do and would be much welcome.

openoms, 2022-07-21:
For now:
* store your seed (or the `hsm_secret` HEX) as text
* the channeldb (`lightningd.sqlite3`) is replicated to the SDcard real-time
* can make a rescue file from time-to-time so you have backup of the onchain wallet (=seed) and the channel database (can be restored as a last resort like the SCBs - will trigger force closes with the peers)

The next CLN version will have SCB like functionality, but will be stored automatically with the peers (encrypted over LN). See: https://github.com/ElementsProject/lightning/pull/5361

0xtrr, 2022-07-21:
If we change the cl-plugin.backup.sh file now, will it just be overwritten when we upgrade to the next version?

openoms, 2022-07-21:
> If we change the cl-plugin.backup.sh file now, will it just be overwritten when we upgrade to the next version?

it is overwritten every time you run `patch` or change the SDcard.

closed 2022-07-26 by rootzoll