raspiblitz/raspiblitz #2991: Touchscreen not booting into status screen (stops at pi shell)

JoKohono, 2022-02-23:
With the 1.7.2 update I'm observing that the system boot occasionally stops at the shell (user pi).

All services are running properly though and since it's only approx 1 in 10 times, it's not a serious issue at all. 

However: what is the command(sequence) to manually start the LCD status screen?

NilByte, 2022-02-24:
`menu` and then select INFO.

JoKohono, 2022-02-24:
@NilByte : 
Menu>Info works from an ssh-shell and then displays the status screen in the ssh terminal window. 
I was looking for manually starting the status screen on the Touchscreen-UI (with the Info/Node/Invoice/Off buttons). 

But I think I found it: 
As user pi, execute /home/admin/00infoLCD.sh