raspiblitz/raspiblitz #2994: System keeps disabling ssh password after every update

w-canvas, 2022-02-24:
After disabling ssh password one time, system keeps turning it off at every raspiblitz update, while restoring persistent configuration at boot.

Fixed it by placing the `ssh.reset` file on boot partition root and manually changing the `PasswordAuthentication` field to `yes` on the following files inside the ssd:

```
/mnt/hdd/app-data/sshd/ssh/sshd_config
/mnt/hdd/app-storage/ssh-old-backup/sshd_config
```

rootzoll, 2022-03-08:
Does is now survive updates?

w-canvas, 2022-03-08:
>Does is now survive updates?

Don't know. Haven't reflashed since.

It did correctly accept my password ssh connection after changing the `ssh_config` files in the ssd.