raspiblitz/raspiblitz #2995: armbian-config non-exsitance breaks armbian SDcard build

harvhat, 2022-02-25:
line 293 of  build_sdcard.sh

'''
[ "${baseimage}" = "armbian" ] && armbian_dependencies="armbian-config" # add armbian-config
'''

However there is no package called "armbian-config" so the whole 'apt install' line fails while the build script continues on anyway, unsuccessfully of course.

I'm thinking armbian-config should be a variable that points to a list of packages, perhaps?

nyxnor, 2022-02-25:
We haven't tested armbian for a while. That was an old line if you follow the blames.

Can you check any package with a similar name on armbian? Look at their source repositories.

harvhat, 2022-02-26:
I din't realize ambian was a OS distro. The comment in the code "# experimental: fallback for all debian on arm" had me thinking it was just what a moniker for any debian on arm, hence my confusing. There is a ambian-config package according to my researc (once I realized it was a distro) however was not trying to install raspiblitz that, I have raspios_arm64 but it didn;t get picked up because the hostname was not "raspberrypi" (I already have other raspberrypi's)

This, however, is a different issue. (I added an option to force a particular baseimage which defaults to "auto" to solve it)

More related to this issue though is the build process breaking when even 1 of the items on the apt install list isn't found, for then _none_ of them are installed, yet the build continues full steam ahead instead if exiting.

I solved this by writing a apt_install function, which will exit if it fails, and calling that everywhere ```apt install -y``` is called.

Though maybe just a || exit 1 is all you really need.

harvhat, 2022-02-28:
Ok @nyxnor I made a pull request for the apt_install() function bit (#3002) and the "added an option to force a particular baseimage" bits are in the setting-base-os branch of my fork. I don't plan on making a PR for that as I think OS detection needs a more major overhaul anyways. But it's there if you want to see.

nyxnor, 2022-02-28:
> However there is no package called "armbian-config" so the whole 'apt install' line fails while the build script continues on anyway, unsuccessfully of course.

When using armbian, there is no package called armbian-config, maybe we should update that package name to the new one.

harvhat, 2022-02-28:
I agree, can't say I know how though. What I did for myself was add a ```-o base_os``` option to the command line args, I could make PR for this but I think improving the os detection a better idea. (Maybe do both though?).  I've found Armbian so I can download and test but the other one referred to in the code as "a clean alternative image of debian for RaspberryPi". I don't know where to find that (or which) that is. If you could inform me the I could get that and then look into the detection issue more. This will take me longer then a few days (at least) though.

nyxnor, 2022-02-28:
> Maybe do both though?

I don't see a problem with adding a `-o` option, I just think it is better if detected, because if it is not detected, it is not supported and if the detection if wrong, then it must be corrected.

Some marker files can be found on neofetch code for os and distro detection.

https://github.com/dylanaraps/neofetch/blob/ccd5d9f52609bbdcd5d8fa78c4fdb0f12954125f/neofetch#L986

harvhat, 2022-02-28:
That link is helpful for Armbian at least.
Do you know where I can find this "a clean alternative image of debian for RaspberryPi" that sets  ```baseimage="debian_rpi64"```
Does it have a more official name?

closed 2023-06-07 by cstenglein