raspiblitz/raspiblitz #2997: [1.7.2] "lnbalance" from LND Lightning Options menu has an issue

cricktor, 2022-02-27:
Yesterday I updated from v1.7.1 to v1.7.2 which as far as I recall went through smoothly.
My node's Settings before update: +run behind TOR; +LND; +Accept Keysend
My node's Services before update: +BTC-RPC-Explorer +Mempool Space +JoinMarket +LND RTL +LND LIT +BOS
Of course node's blockchain and LND are synced before testing.

Calling the option "lnbalance" from LND Lightning Options menu prints the following where it didn't while on v1.7.1:
```
*** YOUR SATOSHI BALANCES ***
[lncli] flag provided but not defined: -1
parse error: Invalid numeric literal at line 1, column 10

LN BALANCE
LOCAL             REMOTE            TOTAL
...
```
Tracing what's going on in script `config.scripts/lnd.balance.sh` the following call to lncli produces the quirk:
```
++ lncli fwdinghistory 0 --max_events -1
[lncli] flag provided but not defined: -1
parse error: Invalid numeric literal at line 1, column 10
```
The value `-1` for `--max_events` seems to be misinterpreted as an unknown flag for the command `lncli fwdingshistory` in the code line https://github.com/rootzoll/raspiblitz/blob/45d7dc702613009b8d039ec686752776740c8038/home.admin/config.scripts/lnd.balance.sh#L27

harvhat, 2022-02-27:
Looks like it needs to be 
```
 LN_EARNED_FEES_IN_MSATS=$(lncli fwdinghistory 0 --max_events=-1 | jq -r '.forwarding_events[] | .fee_msat' | awk '{s+=$1} END {print s}')
```
The ```=``` sign is important

harvhat, 2022-02-27:
Setting --max_events to -1 isn't documented though, I assume it means get it ALL, but I'm not a fan of using undocumented API's.

cricktor, 2022-02-27:
The fun fact is that as far as I see it the script hasn't changed from v1.7.1 to v1.7.2 and I believe I used LND v0.14.2 already when I was still on RaspiBlitz 1.7.1 where this error message didn't occur. I can only check later if your suggestion with the equal sign does work.
Default for option --max_events is `0` and doesn't that mean allready "no limit", not sure though.

harvhat, 2022-02-27:
Probably -1 worked in earlier versions of LND, it's LND that changed. --max_events is 0 will get the default max_events which is 100. The *max* max_events is 50000 (according to the docs), so you could do --max_events 50000, and for me that returns the same answer as --max_events=-1, but I have less then 50000 events so can't test what happens if you have more the 50k and use -1.

rootzoll, 2022-03-08:
Thanks. @harvhat based on your proposed solution I created a commit to the dev branch to be part of the next v1.8 release. Up for final testing.

rootzoll, 2022-06-05:
Tested for v1.8.0 release ... looks good so far. Closing for release.

closed 2022-06-05 by rootzoll