raspiblitz/raspiblitz #3003: letsencrypt script syntax error

allyourbankarebelongtous, 2022-02-28:
Background: upgraded to 1.7.2 from 1.7.1, re-enabled vpn and hybrid mode, and was trying to redo letsencrypt for btcpayserver. Every time I tried to enable letsencrypt it errored out and returned to the raspiblitz main screen very quickly. A screenshot of the error showed:

# running: python /home/admin/config.scripts/blitz.subscriptions.letsencrypt.py create-ssh-dialog
   File "/home/admin/config.scripts/blitz.subscriptions.letsencrypt.py", line 70
       print(*args, file-sys.stderr, **kwargs)
                ^
SyntaxError: invalid syntax
# redis-cli del ln_default_locked:out
(integer) 0

There was more but the screenshot didn't capture it. The problem is that calling the letsencrypt.py script using python didn't default to using python3 and the script has python3 specific functions in it (print as a function, for one). I exited to the command line and ran the same command that is run from the gui but with python3 instead of python and it worked fine. Recommend in addition to separating letsencrypt tls.cert and tls.key from the other nginx services which are not using letsencrypt (i.e., all the other services) calling the letsencrypt script via python3 explicitly.

rootzoll, 2022-03-08:
just checked on my 1.7.2 testnode: `python --version` --> `Python 3.9.2`

Will add to v1.8 milestone to recheck the letsencrypt routine.

rootzoll, 2024-05-29:
With #4579 the letsencrypt ACME was updated and tested. Will be part of v1.11.1rc1 ... reopen if problem persist after that release.

closed 2024-05-29 by rootzoll