raspiblitz/raspiblitz #3007: LNbits on c-lightning not working after RaspiBlitz reboot

Surenic, 2022-03-02:
I am currently on 1.7.2, freshly installed. Everything is running fine so far, but several adjustments are forcing me to reboot from time to time.

So I soon realized that after reboot LNbits' Web Interface is showing me a 502 Bad Gateway.

Un- and reinstalling does help. By keeping the database on uninstall I don't lose any accounts as well.

Any idea on how to fix this? Is it possible to run a command as a short fix? "sudo systemctl restart/start lnbits" does not help though

Surenic, 2022-03-04:
Meanwhile I heard that this issue is not c-lightning only

BitcoinQnA, 2022-03-04:
Also experiencing this issue.

cesarcrisp, 2022-03-05:
I have the same problem. #3010

openoms, 2022-03-06:
Will try to reproduce.
In the meantime check your logs:

`sudo journalctl -u lnbits -n30`

Surenic, 2022-03-06:
Thank you! Here are logs after fresh reinstall and some invoice creations and payments:

-- Journal begins at Sun 2022-03-06 00:09:18 CET, ends at Sun 2022-03-06 13:28:49 CET. --
Mar 06 13:15:41 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:15:41 raspberrypi sh[1730570]: adding sse listener <Queue maxsize=0>
Mar 06 13:15:41 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments HTTP/1.0" 200 OK
Mar 06 13:15:42 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:15:45 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /wallet?usr=f39865dc28a04ada9faae978d995cd18&wal=7f36e35aa64c404490393>
Mar 06 13:15:46 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/currencies HTTP/1.0" 200 OK
Mar 06 13:15:46 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:15:46 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments HTTP/1.0" 200 OK
Mar 06 13:15:46 raspberrypi sh[1730570]: adding sse listener <Queue maxsize=0>
Mar 06 13:15:46 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:15:50 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /wallet?usr=f39865dc28a04ada9faae978d995cd18&wal=d2b5ab4728fb47d7b87c2>
Mar 06 13:15:51 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/currencies HTTP/1.0" 200 OK
Mar 06 13:15:51 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:15:51 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments HTTP/1.0" 200 OK
Mar 06 13:15:51 raspberrypi sh[1730570]: adding sse listener <Queue maxsize=0>
Mar 06 13:15:51 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:16:11 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/sse?api-key=2635e1518f2f427d8cef03613dd9e52f HTTP/1.0>
Mar 06 13:16:15 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "POST /api/v1/payments HTTP/1.0" 201 Created
Mar 06 13:16:20 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/8bb61bc51e7b8fa9b6b8a301ceaf6bca4e7ec9e6ff9b9847af546>
Mar 06 13:16:25 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/8bb61bc51e7b8fa9b6b8a301ceaf6bca4e7ec9e6ff9b9847af546>
Mar 06 13:16:30 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/8bb61bc51e7b8fa9b6b8a301ceaf6bca4e7ec9e6ff9b9847af546>
Mar 06 13:16:35 raspberrypi sh[1730570]: > got a payment notification lbl0.5684637380767649
Mar 06 13:16:35 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/8bb61bc51e7b8fa9b6b8a301ceaf6bca4e7ec9e6ff9b9847af546>
Mar 06 13:16:35 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:16:35 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments HTTP/1.0" 200 OK
Mar 06 13:16:35 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/wallet HTTP/1.0" 200 OK
Mar 06 13:21:34 raspberrypi sh[1730570]: adding sse listener <Queue maxsize=0>
Mar 06 13:21:34 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/sse?api-key=2635e1518f2f427d8cef03613dd9e52f HTTP/1.0>
Mar 06 13:26:57 raspberrypi sh[1730570]: adding sse listener <Queue maxsize=0>
Mar 06 13:26:57 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/sse?api-key=2635e1518f2f427d8cef03613dd9e52f HTTP/1.0>

and these are logs after changing clconf, restart lightnind and trying to create an invoice in lnbits. (Transaction history vanishes, no QR Code on invoice creation)

-- Journal begins at Sun 2022-03-06 00:09:18 CET, ends at Sun 2022-03-06 13:34:50 CET. --
Mar 06 13:34:48 raspberrypi sh[1730570]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 52, in app
Mar 06 13:34:48 raspberrypi sh[1730570]:     response = await func(request)
Mar 06 13:34:48 raspberrypi sh[1730570]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/fastapi/routing.py", line 226, in app
Mar 06 13:34:48 raspberrypi sh[1730570]:     raw_response = await run_endpoint_function(
Mar 06 13:34:48 raspberrypi sh[1730570]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/fastapi/routing.py", line 159, in run_>
Mar 06 13:34:48 raspberrypi sh[1730570]:     return await dependant.call(**values)
Mar 06 13:34:48 raspberrypi sh[1730570]:   File "./lnbits/core/views/api.py", line 369, in api_payment
Mar 06 13:34:48 raspberrypi sh[1730570]:     await check_invoice_status(payment.wallet_id, payment_hash)
Mar 06 13:34:48 raspberrypi sh[1730570]: AttributeError: 'NoneType' object has no attribute 'wallet_id'
Mar 06 13:34:49 raspberrypi sh[1730570]: caught exception in background task: [Errno 13] Permission denied
Mar 06 13:34:49 raspberrypi sh[1730570]: Traceback (most recent call last):
Mar 06 13:34:49 raspberrypi sh[1730570]:   File "./lnbits/tasks.py", line 36, in catch_everything_and_restart
Mar 06 13:34:49 raspberrypi sh[1730570]:     await func()
Mar 06 13:34:49 raspberrypi sh[1730570]:   File "./lnbits/tasks.py", line 79, in invoice_listener
Mar 06 13:34:49 raspberrypi sh[1730570]:     async for checking_id in WALLET.paid_invoices_stream():
Mar 06 13:34:49 raspberrypi sh[1730570]:   File "./lnbits/wallets/clightning.py", line 140, in paid_invoices_stream
[... cut ...]
Mar 06 13:34:49 raspberrypi sh[1730570]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/lightning/lightning.py", line 217, in >
Mar 06 13:34:49 raspberrypi sh[1730570]:     sock.connect(self.socket_path)
Mar 06 13:34:49 raspberrypi sh[1730570]: PermissionError: [Errno 13] Permission denied
Mar 06 13:34:49 raspberrypi sh[1730570]: will restart the task in 5 seconds.
Mar 06 13:34:49 raspberrypi sh[1730570]: INFO:     10.0.0.25:0 - "GET /api/v1/payments/sse?api-key=2635e1518f2f427d8cef03613dd9e52f HTTP/1.0>

openoms, 2022-03-06:
The CL socket `/home/bitcoin/.lightning/bitcoin${clrpcsubdir}/lightning-rpc` is not readable by the `lndbits` user.

Check with:
```
sudo ls -la /home/bitcoin/.lightning/bitcoin/lightning-rpc
```
And run:
```
sudo chmod 660 /home/bitcoin/.lightning/bitcoin/lightning-rpc
```

Now confirm that it is readable by the lnbits user:
```
sudo -u lnbits ls -la /home/bitcoin/.lightning/bitcoin/lightning-rpc
```
Restart lnbits:
```
sudo systemctl restart lnbits
```

openoms, 2022-03-06:
Found the solution in the lightningd config:
```
lightningd -h

--rpc-file-mode <arg>                             Set the file mode (permissions) for the JSON-RPC socket
                                                   (default: "0600")
```

Should persist the setting after restarts if you put this option in the C-ligthning config file:
```
rpc-file-mode=0660
```

openoms, 2022-03-06:
So the quickest persistent fix (can paste these 3 lines):
```
echo "rpc-file-mode=0660" | sudo tee -a /home/bitcoin/.lightning/config
sudo systemctl restart lightningd
sudo systemctl restart lnbits
```

Surenic, 2022-03-06:
**yes! Thank you! this did the fix on restarting lightningd, still on reboot I have the 502 Bad Gateway, here the logs**

-- Journal begins at Sun 2022-03-06 05:15:13 CET, ends at Sun 2022-03-06 14:02:46 CET. --
Mar 06 14:01:57 raspberrypi sh[13939]:     WALLET = wallet_class()
Mar 06 14:01:57 raspberrypi sh[13939]:   File "./lnbits/wallets/clightning.py", line 65, in __init__
Mar 06 14:01:57 raspberrypi sh[13939]:     invoices = self.ln.listinvoices()
Mar 06 14:01:57 raspberrypi sh[13939]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/lightning/lightning.py", line 630, in listinvoices
Mar 06 14:01:57 raspberrypi sh[13939]:     return self.call("listinvoices", payload)
Mar 06 14:01:57 raspberrypi sh[13939]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/lightning/lightning.py", line 217, in call
Mar 06 14:01:57 raspberrypi sh[13939]:     sock.connect(self.socket_path)
Mar 06 14:01:57 raspberrypi sh[13939]: ConnectionRefusedError: [Errno 111] Connection refused
Mar 06 14:01:57 raspberrypi systemd[1]: lnbits.service: Main process exited, code=exited, status=1/FAILURE
Mar 06 14:01:57 raspberrypi systemd[1]: lnbits.service: Failed with result 'exit-code'.
Mar 06 14:01:57 raspberrypi systemd[1]: lnbits.service: Consumed 2.331s CPU time.
Mar 06 14:02:28 raspberrypi systemd[1]: lnbits.service: Scheduled restart job, restart counter is at 2.
Mar 06 14:02:28 raspberrypi systemd[1]: Stopped lnbits.
Mar 06 14:02:28 raspberrypi systemd[1]: lnbits.service: Consumed 2.331s CPU time.
Mar 06 14:02:28 raspberrypi systemd[1]: Starting lnbits...
Mar 06 14:02:28 raspberrypi bonus.lnbits.sh[18189]: # Running: 'bonus.lnbits.sh prestart'
Mar 06 14:02:28 raspberrypi bonus.lnbits.sh[18189]: ## lnbits.service PRESTART CONFIG
Mar 06 14:02:28 raspberrypi bonus.lnbits.sh[18189]: # --> /home/lnbits/lnbits/.env
Mar 06 14:02:28 raspberrypi bonus.lnbits.sh[18189]: # LNBitsFunding(cl) --> network(bitcoin) chain(main) lightning(cl)
Mar 06 14:02:28 raspberrypi bonus.lnbits.sh[18189]: # everything looks OK for lnbits config on c-lightning on mainnet
Mar 06 14:02:28 raspberrypi bonus.lnbits.sh[18189]: # OK: prestart finished
Mar 06 14:02:28 raspberrypi systemd[1]: Started lnbits.
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:     Started server process [18198]
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:uvicorn.error:Started server process [18198]
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:     Waiting for application startup.
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:uvicorn.error:Waiting for application startup.
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:     Application startup complete.
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:uvicorn.error:Application startup complete.
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:     Uvicorn running on http://127.0.0.1:5000 (Press CTRL+C to quit)
Mar 06 14:02:32 raspberrypi sh[18198]: INFO:uvicorn.error:Uvicorn running on http://127.0.0.1:5000 (Press CTRL+C to quit)

openoms, 2022-03-06:
Can refresh the page or try another browser? Your logs show it is running ok now.

Surenic, 2022-03-06:
Ok wow, that is embaressing 🙈 yes, it works. Sorry for that! 

Thank you so much for the quick help!

SpencerPearson, 2022-03-21:
I'm having the same issue but I'm running LND

logs for lnbits:
-- Journal begins at Mon 2022-03-21 15:01:35 GMT, ends at Mon 2022-03-21 22:38:55 GMT. --
Mar 21 22:38:47 raspberrypi bonus.lnbits.sh[435142]: # --> /home/lnbits/lnbits/.env
Mar 21 22:38:47 raspberrypi bonus.lnbits.sh[435142]: # LNBitsFunding(lnd) --> network(bitcoin) chain(main) lightning(lnd)
Mar 21 22:38:47 raspberrypi bonus.lnbits.sh[435142]: # setting lnd config fresh ...
Mar 21 22:38:47 raspberrypi bonus.lnbits.sh[435142]: # Updating LND TLS & macaroon data fresh for LNbits config ...
Mar 21 22:38:47 raspberrypi bonus.lnbits.sh[435142]: # OK: prestart finished
Mar 21 22:38:47 raspberrypi systemd[1]: Started lnbits.
Mar 21 22:38:52 raspberrypi sh[435161]: INFO:     Started server process [435161]
Mar 21 22:38:52 raspberrypi sh[435161]: INFO:     Waiting for application startup.
Mar 21 22:38:52 raspberrypi sh[435161]: ./lnbits/app.py:89: RuntimeWarning:   × The backend for LndRestWallet isn't working properly: '{"code":2, "message":"cannot determine data format of binary-encoded macaroon", "details":[]}'
Mar 21 22:38:52 raspberrypi sh[435161]:   warnings.warn(
Mar 21 22:38:52 raspberrypi sh[435161]: ERROR:    Traceback (most recent call last):
Mar 21 22:38:52 raspberrypi sh[435161]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 540, in lifespan
Mar 21 22:38:52 raspberrypi sh[435161]:     async for item in self.lifespan_context(app):
Mar 21 22:38:52 raspberrypi sh[435161]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 481, in default_lifespan
Mar 21 22:38:52 raspberrypi sh[435161]:     await self.startup()
Mar 21 22:38:52 raspberrypi sh[435161]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 516, in startup
Mar 21 22:38:52 raspberrypi sh[435161]:     await handler()
Mar 21 22:38:52 raspberrypi sh[435161]:   File "./lnbits/app.py", line 94, in check_wallet_status
Mar 21 22:38:52 raspberrypi sh[435161]:     sys.exit(4)
Mar 21 22:38:52 raspberrypi sh[435161]: SystemExit: 4
Mar 21 22:38:52 raspberrypi sh[435161]: ERROR:    Application startup failed. Exiting.
Mar 21 22:38:52 raspberrypi sh[435161]: Starting LNbits with
Mar 21 22:38:52 raspberrypi sh[435161]:   - git version: 41a9689014e40e2830bce53e519248172e56a53d
Mar 21 22:38:52 raspberrypi sh[435161]:   - site title: LNbits
Mar 21 22:38:52 raspberrypi sh[435161]:   - debug: False
Mar 21 22:38:52 raspberrypi sh[435161]:   - data folder: /mnt/hdd/app-data/LNBits
Mar 21 22:38:52 raspberrypi sh[435161]:   - funding source: LndRestWallet
Mar 21 22:38:52 raspberrypi sh[435161]:   - service fee: 0.0
Mar 21 22:38:52 raspberrypi systemd[1]: lnbits.service: Succeeded.
Mar 21 22:38:53 raspberrypi systemd[1]: lnbits.service: Consumed 5.537s CPU time.

dohabee, 2022-04-16:
I think I am having a similar issue.

-- Journal begins at Sat 2022-04-16 02:27:30 BST, ends at Sat 2022-04-16 14:15:14 BST. --
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # --> /home/lnbits/lnbits/.env
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # LNBitsFunding(lnd) --> network(bitcoin) chain(main) lightning(lnd)Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # setting lnd config fresh ...
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # Updating LND TLS & macaroon data fresh for LNbits config ...
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # OK: prestart finished
Apr 16 14:15:02 raspberrypi systemd[1]: Started lnbits.
Apr 16 14:15:07 raspberrypi sh[69743]: INFO:     Started server process [69743]
Apr 16 14:15:07 raspberrypi sh[69743]: INFO:     Waiting for application startup.
Apr 16 14:15:07 raspberrypi sh[69743]: ./lnbits/app.py:89: RuntimeWarning:   × The backend for LndRestWallet isn't work>Apr 16 14:15:07 raspberrypi sh[69743]:   warnings.warn(
Apr 16 14:15:07 raspberrypi sh[69743]: ERROR:    Traceback (most recent call last):
Apr 16 14:15:07 raspberrypi sh[69743]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.p>Apr 16 14:15:07 raspberrypi sh[69743]:     async for item in self.lifespan_context(app):
Apr 16 14:15:07 raspberrypi sh[69743]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.p>Apr 16 14:15:07 raspberrypi sh[69743]:     await self.startup()
Apr 16 14:15:07 raspberrypi sh[69743]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.p>Apr 16 14:15:07 raspberrypi sh[69743]:     await handler()
Apr 16 14:15:07 raspberrypi sh[69743]:   File "./lnbits/app.py", line 94, in check_wallet_status
Apr 16 14:15:07 raspberrypi sh[69743]:     sys.exit(4)
Apr 16 14:15:07 raspberrypi sh[69743]: SystemExit: 4
Apr 16 14:15:07 raspberrypi sh[69743]: ERROR:    Application startup failed. Exiting.
Apr 16 14:15:07 raspberrypi sh[69743]: Starting LNbits with
Apr 16 14:15:07 raspberrypi sh[69743]:   - git version: 41a9689014e40e2830bce53e519248172e56a53d
Apr 16 14:15:07 raspberrypi sh[69743]:   - site title: LNbits
Apr 16 14:15:07 raspberrypi sh[69743]:   - debug: False
Apr 16 14:15:07 raspberrypi sh[69743]:   - data folder: /mnt/hdd/app-data/LNBits
Apr 16 14:15:07 raspberrypi sh[69743]:   - funding source: LndRestWallet
Apr 16 14:15:07 raspberrypi sh[69743]:   - service fee: 0.0
lines 1-29...skipping...
-- Journal begins at Sat 2022-04-16 02:27:30 BST, ends at Sat 2022-04-16 14:15:14 BST. --
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # --> /home/lnbits/lnbits/.env
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # LNBitsFunding(lnd) --> network(bitcoin) chain(main) lightning(lnd)
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # setting lnd config fresh ...
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # Updating LND TLS & macaroon data fresh for LNbits config ...
Apr 16 14:15:02 raspberrypi bonus.lnbits.sh[69730]: # OK: prestart finished
Apr 16 14:15:02 raspberrypi systemd[1]: Started lnbits.
Apr 16 14:15:07 raspberrypi sh[69743]: INFO:     Started server process [69743]
Apr 16 14:15:07 raspberrypi sh[69743]: INFO:     Waiting for application startup.
Apr 16 14:15:07 raspberrypi sh[69743]: ./lnbits/app.py:89: RuntimeWarning:   × The backend for LndRestWallet isn't working properly: '{"code":2, "message":"cannot determine data format of binary-encoded macaroon", "details":[]}'
Apr 16 14:15:07 raspberrypi sh[69743]:   warnings.warn(
Apr 16 14:15:07 raspberrypi sh[69743]: ERROR:    Traceback (most recent call last):
Apr 16 14:15:07 raspberrypi sh[69743]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 540, in lifespan
Apr 16 14:15:07 raspberrypi sh[69743]:     async for item in self.lifespan_context(app):
Apr 16 14:15:07 raspberrypi sh[69743]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 481, in default_lifespan
Apr 16 14:15:07 raspberrypi sh[69743]:     await self.startup()
Apr 16 14:15:07 raspberrypi sh[69743]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 516, in startup
Apr 16 14:15:07 raspberrypi sh[69743]:     await handler()
Apr 16 14:15:07 raspberrypi sh[69743]:   File "./lnbits/app.py", line 94, in check_wallet_status
Apr 16 14:15:07 raspberrypi sh[69743]:     sys.exit(4)
Apr 16 14:15:07 raspberrypi sh[69743]: SystemExit: 4
Apr 16 14:15:07 raspberrypi sh[69743]: ERROR:    Application startup failed. Exiting.
Apr 16 14:15:07 raspberrypi sh[69743]: Starting LNbits with
Apr 16 14:15:07 raspberrypi sh[69743]:   - git version: 41a9689014e40e2830bce53e519248172e56a53d
Apr 16 14:15:07 raspberrypi sh[69743]:   - site title: LNbits
Apr 16 14:15:07 raspberrypi sh[69743]:   - debug: False
Apr 16 14:15:07 raspberrypi sh[69743]:   - data folder: /mnt/hdd/app-data/LNBits
Apr 16 14:15:07 raspberrypi sh[69743]:   - funding source: LndRestWallet
Apr 16 14:15:07 raspberrypi sh[69743]:   - service fee: 0.0
Apr 16 14:15:08 raspberrypi systemd[1]: lnbits.service: Succeeded.
Apr 16 14:15:08 raspberrypi systemd[1]: lnbits.service: Consumed 5.304s CPU time.

rootzoll, 2022-04-25:
Thanks for the reports .. its on the v1.8 recheck/fix list.

m4rcon1, 2022-04-30:
What can I do as workaround, if I have the smae issue but LND installed?

nicklyk, 2022-05-10:
Can confirm as well that my LND node suffers from this. Maybe similar rpc file permission issue?

nicklyk, 2022-05-18:
@m4rcon1 I managed to solve this by going to CONNECT -> RESET. This recreates the macaroons and tls.cert file.

m4rcon1, 2022-05-18:
Hey, wow @nicklyk, thank you for your solution. It worked for me as well. Very nice. :)

rootzoll, 2022-06-02:
Was not able to reproduce in my 1.8.0 testing ... will push to v1.8.2 and keep open for further reports.

Emzy, 2022-08-09:
I had the same problem. Seems to have to do with the macaroons of LND . Maybe an old format?

I have a backup of the not working macaroons. So If you need them please ping me.