raspiblitz/raspiblitz #3009: [CL] Wallet unlockable after uninstalling CLBOSS

Surenic, 2022-03-02:
Sorry for opening one issue after another. I am new to the game, trying to get experience.

I have a c-lighting RaspiBlitz on 1.7.2 and installed CLBOSS today. It did rebalance right away my channels as it should

But besides not being able to use "lightning-cli clboss-unmanage NODEID TAG" (command unknown) I decided to turn it off via RaspiBlitz Settings. I saw a command line window for not more than a second and got back to the menu. From there I could not access the wallet options again because the wallet tried to unlock. I guess this restarts lightningd but obviously not correctly.

I tried everything then. Decrypting the hsm_secret, reboot etc. Nothing helped. Eventually I did a rescan, deactivated encryption and turned CLBOSS back on, followed by rebooting.

I have access again via RTL now but it is still rescanning (which would not have been necessary I think). 

I'd be happy to provide logs, but don't know how.

Why is clboss-unmanage not working and how could I deactivate it without  that chaos?

Surenic, 2022-03-03:
I put a clboss config into the config file, which prevented the wallet from unlocking while clboss was inactive

Still, unmanage dows not work

rootzoll, 2022-03-08:
@openoms can you take a look?

openoms, 2022-03-08:
Answered here: https://github.com/ZmnSCPxj/clboss/issues/96#issuecomment-1061067956

Needs CLBOSS update when there is a new stable release.

ZmnSCPxj, 2022-04-25:
0.11D just released, might make a 0.12 release "soon" if 0.11D looks like it will not have problems.

openoms, 2022-05-13:
if you run `patch` and reinstall CLBOSS should have the new v0.13A version now which should solve this issue as well.

ZmnSCPxj, 2022-05-13:
Oh, you merged in 0.13A?

openoms, 2022-05-13:
> Oh, you merged in 0.13A?
> 
> (should not be a problem, just surprised since you very recently merged in 0.12, LOL)

Yes, the main complaints I was looking for is people losing money. The change in 0.13A has addressed this issue which made me wait to update above 0.10 by default.
Now it is done, so we go all the way.

closed 2022-06-05 by rootzoll