raspiblitz/raspiblitz #3010: C_Lightning Error

cesarcrisp, 2022-03-02:
Hi everyone,
I am getting an error similar to the one described in my raspiblitz installation.
I used the solutions that are described in the issue but it didn't work.
I am using the Raspiblitz software version 1.7.2.
I attach the log and the lnbits image.
Can you help me solve this problem on my node.
If you need any more information, I am at your disposal.
Thank you in advance.
Cesar Crisp

[lnbits.log](https://github.com/rootzoll/raspiblitz/files/8166051/lnbits.log)
[image]
[raspiblitz.log](https://github.com/rootzoll/raspiblitz/files/8173370/raspiblitz.log)

_Originally posted by @cesarcrisp in https://github.com/rootzoll/raspiblitz/issues/2556#issuecomment-1056066503_

Surenic, 2022-03-03:
Hey cesarcrisp, I am having the same issue as described in #3007 

Uninstall and reinstall does the trick, until you reboot. Thanks for providing a log. I didn't know how

cesarcrisp, 2022-03-04:
@Surenic, 
After the reinstallation everything is working.
Thank you for your help.

cesarcrisp, 2022-03-04:
@Surenic , After I restarted my node the problem occurred again.

Surenic, 2022-03-05:
> @Surenic , After I restarted my node the problem occurred again.

then you have to uninstall and install again and make sure NOT to delete the lnbits database. This happens on every reboot. Still don't have a shorter fix :/

Surenic, 2022-03-06:
Solution:

https://github.com/rootzoll/raspiblitz/issues/3007#issuecomment-1059958079

openoms, 2022-03-06:
Closing this as solved and tracked in #3007

closed 2022-03-06 by openoms