raspiblitz/raspiblitz #3011: Provide credentials to connect Alby browser extension (SSH GUI)

MoritzKa, 2022-03-03:
As a user of RaspiBlitz I would like to have a simple access to the credentials that I need to connect to getalby.com. Currently I have to look them up separately. 

Would propose one screen adjustment and 1 new screen:

1. Add Alby to the ["Connect Options"](https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/99connectMenu.sh) screen: Connect Alby browser wallet 
2. new screen: Describe how to connect to Alby:
          - To install Alby open https://getalby.com/ with your preferred browser and click on the "Add to browser button"
          - Use these credentials to connect your node
          - (Show) "Public node URL" or "TOR REST URL" (depending how RaspiBlitz is configured): ....
          - (Show) Admin macaroon: ...
         - If you run your RaspiBlitz behind TOR download and install the Alby app as described during the setup process within the extension. The Alby app allows you to easily connect to your node even when you are not in the same network.

rootzoll, 2022-05-10:
Compare with: #3031

rootzoll, 2022-05-17:
Also test SSH connect menu with Alby

bumi, 2022-05-17:
updated proposal: https://github.com/rootzoll/raspiblitz/pull/3125

[image]
[image]

rootzoll, 2022-06-01:
OK PR merged - closing issue, will be part of v1.8.0 release :)

closed 2022-06-01 by rootzoll