raspiblitz/raspiblitz #3015: c-lightning data corrupted, I do need help to recover funds.

Tracachang, 2022-03-04:
Hello

I just realized that c-lightning was not working any more, after a reboot, checking logs I saw the following message:
`Backup is out of date, we cannot continue safely. Emergency shutdown.`

So I did a backup of lightningd.sqlite3.backup to my PC and then I've tried cl-rescue using repair options, which makes a backup and then uploads to raspiblitz in order to restore it. After restoring c-lightning does not work at all and no logs when typing clogs.

When manually trying to start lightnind:
`lightningd: /home/bitcoin/.lightning/config line 15: plugin-dir: Failed to register /home/bitcoin/cl-plugins-enabled/backup.py: Success`

I've tried to run it again after disabling cl-plugin.backup.sh:
`Optimistic lock on the database failed. There may be a concurrent access to the database. Aborting since concurrent access is unsafe.
lightningd: FATAL SIGNAL 6 (version v0.10.2)
0x5590f8778f send_backtrace
        common/daemon.c:33
0x5590f8782b crashdump
        common/daemon.c:46
0x7fb26257d7 ???
        ???:0
0x7fb212d2ac ???
        ???:0
0x7fb2119e9f ???
        ???:0
0x5590f4436f fatal
        lightningd/log.c:804
0x5590f6a9ab db_data_version_incr
        wallet/db.c:1094
0x5590f6aa33 db_commit_transaction
        wallet/db.c:1109
0x5590f258bf getrawblockbyheight_callback
        lightningd/bitcoind.c:383
0x5590f60c03 plugin_response_handle
        lightningd/plugin.c:521
0x5590f60d53 plugin_read_json_one
        lightningd/plugin.c:627
0x5590f60f0b plugin_read_json
        lightningd/plugin.c:672
0x5590fd9e5f next_plan
        ccan/ccan/io/io.c:59
0x5590fda437 do_plan
        ccan/ccan/io/io.c:407
0x5590fda4f7 io_ready
        ccan/ccan/io/io.c:417
0x5590fdc3f7 io_loop
        ccan/ccan/io/poll.c:453
0x5590f3db97 io_loop_with_timers
        lightningd/io_loop_with_timers.c:21
0x5590f41f63 main
        lightningd/lightningd.c:1163
0x7fb211a217 ???
        ???:0
0x5590f24d87 ???
        ???:0
Log dumped in crash.log.20220304234058
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Lost connection to the RPC socket.
Aborted
`

openoms, 2022-03-05:
Did you make sure your disk and SDcard are working? Why was  corruption in the first place? Cold reset?
Is there enough free space?
`df -h`

Try to reinitialize the backup plugin:
`/home/bitcoin/cl-plugins-enabled/backup.py on`

Good that you saved the backup, will be able to restore it if the above fails.

Tracachang, 2022-03-05:
Well that was an sdcard and hdd that have been using for long without issues, but this week 2 times happened that when I check raspiblitz it was off...

I've tried to reinitialize the backup plugin and not works.

Finally I've flashed raspiblitz in a new SDCard and I've started a new fresh setup (just keeping blockchain), in c-lightning I have uploaded clrescue file. Bitcoin is synched but I do not see any activity in c-lightning, cllogs are empty.

`Will follow the /home/bitcoin/.lightning/bitcoin/cl.log
running 'sudo tail -n 30 -f /home/bitcoin/.lightning/bitcoin/cl.log'

Press ENTER to continue
use CTRL+C any time to abort .. then use the command 'raspiblitz' to return to the menu
#######################################################################################

`

sudo systemctl status lightningd
`● lightningd.service - c-lightning daemon on mainnet
     Loaded: loaded (/etc/systemd/system/lightningd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sat 2022-03-05 08:06:15 GMT; 24s ago
    Process: 132328 ExecStartPre=/home/admin/config.scripts/cl.check.sh prestart mainnet (code=exited, status=1/FAILURE)
    Process: 132343 ExecStart=/bin/sh -c /usr/local/bin/lightningd                         --conf=/home/bitcoin/.lightning/config                    >
   Main PID: 132343 (code=exited, status=1/FAILURE)`

Tracachang, 2022-03-05:
Solved!

I do stop `config.scripts/cl-plugin.backup.sh off` tried to run lightningd (did not start saying it is unsafe), turned on again `config.scripts/cl-plugin.backup.sh on` and then all is online. I can see all channels active and balance.

Thanks a lot openoms!

closed 2022-03-05 by Tracachang