raspiblitz/raspiblitz #3016: How to -reindex-chainstate

RemoRitzmann, 2022-03-05:
I copied BLOCKCHAIN with "dd" from one disk to another. Then after restart I found "Please restart with -reindex or -reindex-chainstate to recover." in a log file. But I could not figure out how to do it properly until I learned that it needs to be done with the correct user:
**sudo -u bitcoin bitcoind -reindex-chainstate**
Can you suggest something like that in the output of the message above? 
So that anyone having that problem can fix it faster then me.

openoms, 2022-03-05:
Could be good FAQ entry.

openoms, 2022-03-30:
This will wither need a separate script or the tutorial should say:
* stop the bitcoin.service first
```
sudo systemctl stop bitcoind
```
then run:
```
sudo -u bitcoin bitcoind -reindex-chainstate
```
If the terminal window is closed the rescan will continue if the node (or the systemd service) is restarted.

RemoRitzmann, 2022-04-01:
Great, that helped! Thanks a lot.

rootzoll, 2022-08-03:
To be added as an Option in the REPAIR menu in 1.8.1 and then pushed to 1.8.2 to get autodetected in cases like:
https://github.com/rootzoll/raspiblitz/issues/3265#issue-1323553813

NilByte, 2022-08-04:
If a FAQ is made . explaining the difference between running -reindex or -reindex-chainstate might be a good idea.

A good explanation here:

https://bitcoin.stackexchange.com/questions/60709/when-should-i-use-reindex-chainstate-and-when-reindex

rootzoll, 2023-02-03:
Added reindex options to repair menu.

closed 2023-02-03 by rootzoll

RemoRitzmann, 2023-02-06:
Thanks a lot for that nice solution! I see how the menu improved a lot recently.

mislam19, 2025-03-06:
who knew this would help a noob 3 years later 👍

mislam19, 2025-03-06:
I think  I had a sort of unique issue my chainstate file has a stalefile. how to fix this?