raspiblitz/raspiblitz #3017: '/mnt/hdd': Input/output error

yusoy, 2022-03-05:
I have this error in my raspiblitz installation:
ls: reading directory '/mnt/hdd': Input/output error
(standard_in) 1: syntax error
/home/admin/config.scripts/blitz.datadrive.sh: line 430: * 1024: syntax error: operand expected (error token is "* 1024")

Raspiblitz is installed on a raspberry pi 4 4GB of Ram and the version is 1.7.2.

Any idea what to do?

harvhat, 2022-03-05:
Double check that the drive is well connected and working (the light flashes sometimes). 
Which installation method are you using? and do you happen to know the base operating system and or the file system type on the SSD?

yusoy, 2022-03-05:
₿ uname -a
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
₿ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
₿ sudo fdisk -l
/dev/sda1   2048 1953523711 1953521664 931.5G Linux filesystem

To install I downloaded the .img 1.7.1 and burned the mSD with dd.....
After update to 1.7.2...
Right now after installing electrs it seems to be indexing Electrum....

I have the node outside my house and I access by ssh. This happened to him installing electrs, when he was already indexing.
I restarted and it seems that for the moment everything is going well... let's see tomorrow how I find it...

yusoy, 2022-03-07:
After finishing electrs indexing and rebooting I haven't had any more failures.
At the moment everything is fine

closed 2022-03-07 by yusoy