raspiblitz/raspiblitz #3018: Raspiblitz emergency mode, then white screen

goolash761, 2022-03-06:
I have had a raspiblitz running since last week.  Yesterday, it completed syncing with the blockchain, I installed several of the optional software packages, ending with Mempool.  This morning, Mempool finished analyzing, but connecting via tor was giving me incomplete information: only the block history was filling in.  So, I rebooted the r-pi.  It failed to connect with ssh at this point, so I checked the lcd screen and it had a string of messages about being in emergency mode.  I decided to reboot by power cycling (I had no keyboard to ctrl-d with).
Now, it powers on witha blank white screen.  Ssh does not connect, but the LEDs on the board blink seemingly like normal.
I am planning to reflash the SD, but thought to ask on here if there is a better path forward?  Also, will reflashing wipe my wallet/channel info (it is empty anyway), or is that stored on the SSD?
Thanks.

openoms, 2022-03-06:
Likely your SDcard failed. Use a new Endurance model and flash it. All unique data will be restored from the disk.

goolash761, 2022-03-06:
I'll test it and see.  Thanks!

Sent from AT&T Yahoo Mail on Android

closed 2022-07-04 by openoms