raspiblitz/raspiblitz #3024: raspberry 3B+ freezes during synchronization of the blockchain. firmware 1.7.2

tomm2001, 2022-03-08:
please tell me, does raspiblitz work on raspberry 3b+? it has 1 gigabyte of RAM. I have tried both the latest firmware version and the previous two - any of them freezes after switching on and starting synchronization of the blockchain, after about 10 hours. how to find the reason ? I tried changing the sd card - it didn't help.

rootzoll, 2022-03-08:
We didnt tested RP3 for a long time anymore .. but thanks for taking a look into. What kind is the freeze? Just blockchain not further syncing or complete system going bust?

Also for RP3 I would not recommend seld-syncing the bnlockchain ... best would be to copy a alaready synced blockchaion from another RaspiBlitz or Laptop. This `COPY` option should be offered during the setup process: https://github.com/rootzoll/raspiblitz#final-setup

tomm2001, 2022-03-09:
freeze is externally manifested as the flashing of the LED of the network interface with the same time intervals. and during the synchronization of the blockchain, the LED flashes randomly. at the same time, it is impossible to connect to RP3 via SSH protocol. freeze on the old firmware occurred both a week and a day later - it was impossible to predict when the freeze would occur. and on the new firmware, after 10-12 hours of RP3 operation. after disconnecting the RP3 power connector and re-enabling, the blockchain download continues for about 10 hours, after which it freezes again. please tell me in which log file can I see the reason for freezing? maybe I'd better send you the log files to study? my nephew suggested that freezing occurs due to lack of RAM RP3, can it be?

rootzoll, 2022-03-09:
Its hard to speculate ... but if possible try to check log files: /mnt/hdd/bitcoin/debug.log ... if you run your RaspiBlitz with an LCD screen on top (as in hardware list) you should see the RAM use and last update time .. what info do you see there?

tomm2001, 2022-03-10:
what messages in the debug.log file should I pay attention to? unfortunately I don't have a screen, I connect to RP3 via SSH. but when RP3 freezes, it is impossible to connect via SSH(

openoms, 2022-03-13:
It can be that the RPi3 is running out of RAM. 
Test ZRAM to prevent that., here is how to install: https://github.com/rootzoll/raspiblitz/issues/2905#issuecomment-1019860323

During high load also the disk might require higher power and cause freezes. Can try running with a powered USB hub.

tomm2001, 2022-03-16:
understood, thanks, I'll try. I installed zram last night, RP3 worked for almost a day without freezing, I will continue to observe.

openoms, 2022-06-28:
closing as seem to be solved with ZRAM - default in the next release.

closed 2022-06-28 by openoms