raspiblitz/raspiblitz #3025: BTCPayserver Update to 1.4.7 fails

pretyflaco, 2022-03-12:
Raspiblitz v.1.7.2

To reproduce:
Update BTCPayserver from menu 

Expected behaviour:
Update BTCPayserver from 1.4.4 to 1.4.7

Actual Result:

```
# Reset to the latest release tag: v1.4.7
HEAD is now at cde5bd87 update 1.4.7
--2022-03-12 15:26:29--  https://keybase.io/nicolasdorier/pgp_keys.asc
Resolving keybase.io (keybase.io)... 3.211.231.6, 18.211.246.37
Connecting to keybase.io (keybase.io)|3.211.231.6|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12753 (12K) [text/plain]
Saving to: ‘/var/cache/raspiblitz/pgp_keys_nicolasdorier.asc’

/var/cache/raspiblitz/pgp_keys_nicolasdorier.as 100%[=====================================================================================================>]  12.45K  --.-KB/s    in 0s      

2022-03-12 15:26:35 (138 MB/s) - ‘/var/cache/raspiblitz/pgp_keys_nicolasdorier.asc’ saved [12753/12753]

pub   rsa4096 2018-09-05 [SC] [expires: 2034-09-01]
      AB4CFA9895ACA0DBE27F6B346618763EF09186FE
uid                      Nicolas Dorier <nicolas.dorier@gmail.com>
sub   rsa4096 2018-09-05 [E] [expires: 2034-09-01]

pub   rsa4096 2019-05-02 [SC] [expires: 2035-04-28]
      9BAE3BD978B26794ECD5675FD6438E1F94D05292
uid                      Nicolas Dorier <dorier@metaco.com>
sub   rsa4096 2019-05-02 [E] [expires: 2035-04-28]

pub   rsa4096 2019-11-26 [SC] [expires: 2035-11-22]
      015B4C837B245509E4AC8995223FDA69DEBEA82D
uid                      Nicolas Dorier <nicolas.dorier@gmail.com>
sub   rsa4096 2019-11-26 [E] [expires: 2035-11-22]

pub   rsa4096 2019-11-27 [SC] [expires: 2035-11-23]
      7121BDE3555D9BE06BDDC68162FE85647DEDDA2E
uid                      BTCPayServer Vault <nicolas.dorier@gmail.com>
sub   rsa4096 2019-11-27 [E] [expires: 2035-11-23]

gpg: key 6618763EF09186FE: "Nicolas Dorier <nicolas.dorier@gmail.com>" not changed
gpg: key D6438E1F94D05292: "Nicolas Dorier <dorier@metaco.com>" not changed
gpg: key 223FDA69DEBEA82D: "Nicolas Dorier <nicolas.dorier@gmail.com>" not changed
gpg: key 62FE85647DEDDA2E: "BTCPayServer Vault <nicolas.dorier@gmail.com>" not changed
gpg: Total number processed: 4
gpg:              unchanged: 4
# running: git verify-commit cde5bd87

# goodSignature(0)
# correctKey(0)

# !!! BUILD FAILED --> PGP verification not OK / signature(0) verify(0)
# submenu signaled exit code '1' --> forward to outside loop
```

openoms, 2022-03-13:
The release commit was not signed by @Kukks: https://github.com/btcpayserver/btcpayserver/releases/tag/v1.4.7

Comment out these two lines to not break on the verification:
https://github.com/rootzoll/raspiblitz/blob/25c52404b0241d58e78771507bf1f3bca9be3672/home.admin/config.scripts/bonus.btcpayserver.sh#L435

```
#    sudo -u btcpay /home/admin/config.scripts/blitz.git-verify.sh \
#     "${PGPsigner}" "${PGPpubkeyLink}" "${PGPpubkeyFingerprint}" || exit 1
```

allyourbankarebelongtous, 2022-03-14:
Attempted to update btcpay by bypassing signature requirement in the script openoms mentioned above, but that script is not the one that's being run by the update menu. It appears the update menu is running this script: `/home/admin/config.scripts/bonus.btcpayserver.sh`
Editing that script had the desired effect and btcpayserver updates to the latest version.

openoms, 2022-03-14:
@allyourbankarebelongtous thabk you forbthe clarification.
Yes, you need to comment out the lines 435-436 in:
`/home/admin/config.scripts/bonus.btcpayserver.sh`
as was shown above.

Kukks, 2022-03-14:
Hi guys, sorry for the chaos. Will ensure that any releases are do are signed in the future. Although, it seems you are coded to only verify against nicolas. 

Maybe we should start a btcpayserver specific pgp key. Ping @nicolasdorier

pretyflaco, 2022-03-14:
> @allyourbankarebelongtous thank you for the clarification. Yes, you need to comment out the lines 435-436 in: `/home/admin/config.scripts/bonus.btcpayserver.sh` as was shown above.

I must be doing sth wrong? 

I have commented out lines 435-436 in /home/admin/config.scripts/bonus.btcpayserver.sh and saved with changes.

[image]

However, I get the same output as before... am I missing sth?

openoms, 2022-03-14:
@pretyflaco make sure the change is saved and try again. Will push a fix for this.

openoms, 2022-03-14:
OK the fix: https://github.com/rootzoll/raspiblitz/commit/6d3af0cd8c79129d838cc91063e56c9d8c458ad7 is in both `dev` and `v1.7` so can just call `patch` to make the update work.

pretyflaco, 2022-03-17:
Patched through menu, then Update to latest BTCPayserver through menu. Works on my end now.

Kukks, 2022-03-23:
I will start signing commits/releases on Github with https://keybase.io/kukks/pgp_keys.asc

rootzoll, 2022-06-01:
OK install looks good - closing for v1.8.0 release.

closed 2022-06-01 by rootzoll