raspiblitz/raspiblitz #3028: npm problems

ttrueten, 2022-03-15:
Hi there,
Hello all,
I tried to start a recovery with a fresh memory card. However, the recovery always gets stuck at one point:

```
# OK - RTL code copy looks good
# Run: npm install
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated request-promise@4.2.6: request-promise has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
```

I am using the current image with raspberry pi 4 / 8 GB.

Cheers,
Thomas

harvhat, 2022-03-16:
These are WARN (Warnings) not actual errors, so if everything works I wouldn't worry.  *deprecated* warnings are common in nodeJS I've noticed. The developer (of RTL)  will need to update their code eventually.

ttrueten, 2022-03-20:
Thank you for your answers. Since everything else is running, I don't worry about it.

Cheers,
Thomas

closed 2022-03-20 by ttrueten