raspiblitz/raspiblitz #3030: Loop doesn't work

Pixel-yml, 2022-03-17:
Latest version of raspiblitz

When i try to use loop from command line i got this:

```Bash
admin@raspiblitz:~ ₿ loop out --max_swap_routing_fee 1500 250000
[loop] unable to read macaroon path : open /home/admin/.loop/mainnet/loop.macaroon: no such file or directory
```
Using loop from RTL gets me a 500 error

```
ERROR
Error Code
500
Error Message
Unknown error
API URL
./api/loop/out/termsAndQuotes
```

[image]

openoms, 2022-03-18:
Loop is not installed by default, but part of the LiT package. Install Lightning Terminal and see the instructions in the menu to use the command line.
RTL will have Loop working once LiT is installed.

Pixel-yml, 2022-03-18:
> Loop is not installed by default, but part of the LiT package. Install Lightning Terminal and see the instructions in the menu to use the command line. RTL will have Loop working once LiT is installed.

Thanks for your answer!
LIT was installed, trying to reinstall but nothing changed.

openoms, 2022-03-18:
Try to reinstall RTL too and try the lit command-line.

Pixel-yml, 2022-03-18:
> Try to reinstall RTL too and try the lit command-line.

Reinstall RTL and LiT again and tried to make a loop out using the `lit-loop out` command and it finished successfully.
But when I try to make a loop using RTL, it shows the same previous error 500.

senf42, 2022-03-20:
Not sure if it's related, but if it's helpful: Since 1.7.2 when I load the Loop page in terminal.lightning.engineering, I get an error that says "UNABLE TO FETCH SWAPS rpc error: code = Unimplemented desc = unknown service looprpc.SwapClient"

I have not tried any loops since upgrading.

rootzoll, 2024-09-26:
Retest when RTL is updated tu v0.15.0 #4333