raspiblitz/raspiblitz #3034: FailedToGetMacaroonFileFromDefaultLocation starting bos

RemoRitzmann, 2022-03-20:
Hi all, I started here https://github.com/alexbosworth/balanceofsatoshis/issues/96 in short:
Source:
  https://gist.github.com/openoms/823f99d1ab6e1d53285e489f7ba38602
  https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/bonus.bos.sh
  https://github.com/alexbosworth/balanceofsatoshis/blob/5054e0046e933fdb5d2c20b1d919dfe0166b8b6f/test/lnd/test_get_macaroon.js
Error:
  error: 503, FailedToGetMacaroonFileFromDefaultLocation

RemoRitzmann, 2022-03-20:
Ok, my goal was to get bos working on my raspiblitz. But bos only works with LND (am I right?) Since I have C-Lightning working, there is only my C-Lightning wallet working and filled with sats. In the meantime, I have LND wallet set up, but LND is not working because of missing macaroons.
http://"IP":3000/rtl/error
Lnd get info failed due to bad or missing macaroon! please check rtl-config.json to verify the setup!
I don't want to have two wallets. I think I'm messed up with LND and C-Lightning so I sadly cannot use bos yet.

RemoRitzmann, 2022-03-20:
Is it possible to remove the actual LND Wallet and connect it to the same Wallet I have with C-Lightning?

rootzoll, 2022-04-25:
> Ok, my goal was to get bos working on my raspiblitz. But bos only works with LND (am I right?) 

Correct. BOS is just an LND tool and will not work if you have just c-lightning running.

> Is it possible to remove the actual LND Wallet and connect it to the same Wallet I have with C-Lightning?

You can deactivate LND completly again in MAINMENU > SETTINGS and decheck the "LND LIGHTNING LABS NODE". But you will not be able to use a c-lightning wallet with LND somehow - the wallets of the different implementations are not interchangeable.

closed 2022-06-07 by RemoRitzmann